introlab/rtabmap_ros #262: Two Kinect playback odometry issue

frushton, 2018-07-11:
I am trying to modify the two Kinect launch file so that I can play bagfiles and have rtabmap run on the topics published by the bagfile. I have it working for when I do not try to republish /tf and /tf_static, so I can replay bagfiles as long as the calibration (of the extrinsics) was correct at the time of the recording. However, I have a bagfile recorded that does not have the correct /tf and /tf_static, but it does not work when I try to modify them. When I play the bagfile, I use the following command:

rosbag play -d 8 --clock /camera1/rgb/image_rect_color/compressed:=/compressed1/rgb/image_rect_color/compressed /camera2/rgb/image_rect_color/compressed:=/compressed2/rgb/image_rect_color/compressed /tf:=/foo1 /tf_static:=/foo2 [file.bag]

Here is the launchfile (it has a .txt extension so I could upload it, trying to copy it here was giving me trouble, and half of the code was not appearing). Note that I have commented out the rosbag play section in the launch file. It throws an error for some reason within the launch file, so I just manually do the same command in another terminal:

[working_demo_two_kinects.txt](https://github.com/introlab/rtabmap_ros/files/2185480/working_demo_two_kinects.txt)

Here is the output from the console (I get the same output when I make sure the bagfile is already playing before launching the launch file as well):

... logging to /home/e4e/.ros/log/760957d8-8527-11e8-b35d-9cb6d0177589/roslaunch-e4e-GS63VR-6RF-10936.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://e4e-GS63VR-6RF:46443/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /rtabmap/rgbd_odometry/Odom/FillInfoData: true
 * /rtabmap/rgbd_odometry/Odom/Strategy: 0
 * /rtabmap/rgbd_odometry/OdomF2M/MaxSize: 1000
 * /rtabmap/rgbd_odometry/Vis/CorNNType: 3
 * /rtabmap/rgbd_odometry/Vis/EstimationType: 0
 * /rtabmap/rgbd_odometry/Vis/FeatureType: 6
 * /rtabmap/rgbd_odometry/Vis/InlierDistance: 0.02
 * /rtabmap/rgbd_odometry/Vis/MaxDepth: 4.0
 * /rtabmap/rgbd_odometry/Vis/MinInliers: 20
 * /rtabmap/rgbd_odometry/frame_id: base_link
 * /rtabmap/rgbd_odometry/rgbd_cameras: 2
 * /rtabmap/rgbd_odometry/subscribe_rgbd: True
 * /rtabmap/rgbd_odometry/wait_for_transform: True
 * /rtabmap/rtabmap/Grid/FromDepth: false
 * /rtabmap/rtabmap/Vis/EstimationType: 0
 * /rtabmap/rtabmap/Vis/InlierDistance: 0.02
 * /rtabmap/rtabmap/Vis/MinInliers: 10
 * /rtabmap/rtabmap/frame_id: base_link
 * /rtabmap/rtabmap/gen_scan: True
 * /rtabmap/rtabmap/map_negative_poses_ignored: False
 * /rtabmap/rtabmap/map_negative_scan_empty_ray_tracing: False
 * /rtabmap/rtabmap/rgbd_cameras: 2
 * /rtabmap/rtabmap/subscribe_depth: False
 * /rtabmap/rtabmap/subscribe_rgbd: True
 * /rtabmap/rtabmap/wait_for_transform: True
 * /rtabmap/rtabmapviz/frame_id: base_link
 * /rtabmap/rtabmapviz/rgbd_cameras: 2
 * /rtabmap/rtabmapviz/subscribe_depth: False
 * /rtabmap/rtabmapviz/subscribe_odom_info: True
 * /rtabmap/rtabmapviz/subscribe_rgbd: True
 * /rtabmap/rtabmapviz/wait_for_transform: True
 * /use_sim_time: True

NODES
  /camera2/
    rgbd_sync (nodelet/nodelet)
  /camera1/
    rgbd_sync (nodelet/nodelet)
  /rtabmap/
    rgbd_odometry (rtabmap_ros/rgbd_odometry)
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)
  /
    base_to_camera1_tf (tf/static_transform_publisher)
    base_to_camera2_tf (tf/static_transform_publisher)
    decompress_image_1 (image_transport/republish)
    decompress_image_2 (image_transport/republish)
[... 119 lines cut ...]
[ WARN] [1531327724.721028239, 1526756966.924824337]: odometry: Could not get transform from base_link to camera1_rgb_optical_frame (stamp=1526756966.792272) after 0.100000 seconds ("wait_for_transform_duration"=0.100000)! Error="canTransform: source_frame camera1_rgb_optical_frame does not exist.. canTransform returned after 0.100936 timeout was 0.1."
   /camera2/rgbd_image
[ WARN] [1531327724.822213536, 1526756967.025693101]: odometry: Could not get transform from base_link to camera1_rgb_optical_frame (stamp=1526756966.892008) after 0.100000 seconds ("wait_for_transform_duration"=0.100000)! Error="canTransform: source_frame camera1_rgb_optical_frame does not exist.. canTransform returned after 0.100869 timeout was 0.1."
[ WARN] [1531327724.932850553, 1526756967.136606740]: odometry: Could not get transform from base_link to camera1_rgb_optical_frame (stamp=1526756966.992579) after 0.100000 seconds ("wait_for_transform_duration"=0.100000)! Error="canTransform: source_frame camera1_rgb_optical_frame does not exist.. canTransform returned after 0.100848 timeout was 0.1."
[ WARN] [1531327725.034156441, 1526756967.237455604]: odometry: Could not get transform from base_link to camera1_rgb_optical_frame (stamp=1526756967.092714) after 0.100000 seconds ("wait_for_transform_duration"=0.100000)! Error="canTransform: source_frame camera1_rgb_optical_frame does not exist.. canTransform returned after 0.100849 timeout was 0.1."
[ WARN] [1531327725.155098766, 1526756967.358800119]: odometry: Could not get transform from base_link to camera1_rgb_optical_frame (stamp=1526756967.228996) after 0.100000 seconds ("wait_for_transform_duration"=0.100000)! Error="canTransform: source_frame camera1_rgb_optical_frame does not exist.. canTransform returned after 0.101151 timeout was 0.1."
[ WARN] [1531327725.256280047, 1526756967.459864567]: odometry: Could not get transform from base_link to camera1_rgb_optical_frame (stamp=1526756967.325903) after 0.100000 seconds ("wait_for_transform_duration"=0.100000)! Error="canTransform: source_frame camera1_rgb_optical_frame does not exist.. canTransform returned after 0.101064 timeout was 0.1."
[ WARN] [1531327725.357578480, 1526756967.560793143]: odometry: Could not get transform from base_link to camera1_rgb_optical_frame (stamp=1526756967.426281) after 0.100000 seconds ("wait_for_transform_duration"=0.100000)! Error="canTransform: source_frame camera1_rgb_optical_frame does not exist.. canTransform returned after 0.100929 timeout was 0.1."
[ WARN] [1531327725.458884494, 1526756967.661851430]: odometry: Could not get transform from base_link to camera1_rgb_optical_frame (stamp=1526756967.527545) after 0.100000 seconds ("wait_for_transform_duration"=0.100000)! Error="canTransform: source_frame camera1_rgb_optical_frame does not exist.. canTransform returned after 0.101058 timeout was 0.1."
^C[decompress_image_2-9] killing on exit
[decompress_image_1-8] killing on exit
[rtabmap/rtabmapviz-7] killing on exit
[rtabmap/rtabmap-6] killing on exit
[rtabmap/rgbd_odometry-5] killing on exit
[camera2/rgbd_sync-4] killing on exit
[camera1/rgbd_sync-3] killing on exit
[base_to_camera2_tf-2] killing on exit
[base_to_camera1_tf-1] killing on exit
[ INFO] [1531327725.535212394, 1526756967.732628221]: rtabmapviz: ctrl-c catched! Exiting Qt app...
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
rtabmap: Saving database/long-term memory... (located at /home/e4e/.ros/rtabmap.db)
shutting down processing monitor...
... shutting down processing monitor complete
done

matlabbe, 2018-07-11:
It looks like the TF for the cameras are not there. For example, you may try to launch [kinect_frames.launch](https://github.com/ros-drivers/rgbd_launch/blob/indigo-devel/launch/kinect_frames.launch) for each camera.

frushton, 2018-07-12:
Thank you for responding. It turns out the problem was that I was suppressing both /tf and /tf_static from the bag file. I suppressed only /tf, and now the playback is working as the new /tf is being published, and the old /tf_static is gving the transforms for the cameras' rgb and depth.

closed 2018-07-12 by frushton