introlab/rtabmap_ros #263: rtabmap_ros for ROS Melodic on Ubuntu 18.04

UditSinghParihar, 2018-07-13:
Sir, I would like to know that whether I could build rtabmap_ros from source for ROS Melodic on Ubuntu 18.04?
Would there be any issues? Since it seems to be that default ROS versions are Lunar/Kinetic or earlier.

matlabbe, 2018-07-13:
Yes, you should be able to build it. All required dependencies should be available from ros binaries. Let me know if catkin_make doesn't work.

cheers,
Mathieu

UditSinghParihar, 2018-07-17:
Sir, I was able to correctly installed rtabmap_ros on Ubuntu 18.04, but I am having difficulty in exporting Point Cloud or view point cloud. Whenever I do [File -> Export 3D Cloud](url) or  [Edit -> View Point Clouds](url), then rtabmapviz collapses. I even try doing the database viewer.
`$ rtabmap ~/.ros/rtabmap.db`
But still unable to export 3D cloud and got segmentation fault.
I am using [rgbdslam_datasets.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/rgbdslam_datasets.launch)
and also using the bagfile: rgbd_dataset_freiburg3_long_office_household.bag. I could generate trajectory and view point cloud in rtabmapviz but unable to export Point cloud.
The error output when rtabmap_ros collapses when trying to export point cloud is:
```
[rtabmap/rtabmapviz-5] process has died [pid 650, exit code -11, cmd /home/udit/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz -d /home/udit/catkin_ws/src/rtabmap_ros/launch/config/rgbd_gui.ini rgb/image:=/camera/rgb/image_color depth/image:=/camera/depth/image rgb/camera_info:=/camera/rgb/camera_info odom:=vis_odom __name:=rtabmapviz __log:=/home/udit/.ros/log/c04a815a-89be-11e8-807c-90489a430961/rtabmap-rtabmapviz-5.log].
log file: /home/udit/.ros/log/c04a815a-89be-11e8-807c-90489a430961/rtabmap-rtabmapviz-5*.log

```
Thanks

matlabbe, 2018-07-23:
Thank you for all the info. `ros-melodic-rtabmap-ros` is indeed not available yet, some dependencies are still missing on melodic to be built on the buildfarm, see https://github.com/introlab/rtabmap_ros/issues/241. When you install/unsintall `ros-melodic-rtabmap`, you should have most of the required dependencies to build `rtabmap_ros` from source (move_base/octomap related topics won't be available though).

For the actual exporting bug, is it crashing when opening the Export dialog or when clicking on Save in the Export dialog? It is maybe related to a Qt bug (http://official-rtab-map-forum.67519.x6.nabble.com/Calibration-tool-td2738.html or http://official-rtab-map-forum.67519.x6.nabble.com/RTab-Map-closes-unexpectedly-when-I-m-trying-to-save-database-td1149.html#a1312).

cheers,
Mathieu

UditSinghParihar, 2018-07-23:
Sir, the error occurs when clicking the save dialog, which was similar to the link you provided (http://official-rtab-map-forum.67519.x6.nabble.com/RTab-Map-closes-unexpectedly-when-I-m-trying-to-save-database-td1149.html#a1312).
Though I have moved back to the Ubuntu16.04, since opencv also have issues with image_view as I mentioned earlier(https://github.com/ros-perception/image_pipeline/issues/201).

For me it's too early to move to Ubuntu 18.04 as everything seems to be development phase.
We could closed this issue as of now.

Thanks

closed 2019-04-21 by UditSinghParihar