introlab/rtabmap_ros #265: How big is the error range?

SSTTTAY, 2018-08-01:
Hello,I am trying to navigate indoor with turtlebot and kinect .

Have you tried to measure this method(rtabmap_ros) on indoor navigation and How big is the error range?

matlabbe, 2018-08-01:
The trajectory error would be dependent on the odometry drift when there are no loop closures. You may evaluate what is your odometry error.

SSTTTAY, 2018-08-02:
Thank you for replying!
I'm using turtlebot and kinect 1 for testing but there are some error about their data flow . Should I change kinect 1 to kinect 2 ? Because I heard that most of people use kinect 2 .
By the way, what about the trajectory error would be when there are loop closures?I just wonder if you test your model on somewhere ? and how about the error range ？

SSTTTAY, 2018-08-03:
another two questions:
1.when I put the turtlebot on a random place , I found it always locate on where I start building the map. It seems it's accessible when the map is just one room , but when it comes to a big area, it goes wrong.
2. Loading 3D map raise this error: **Map received with code error 1!**

SSTTTAY, 2018-08-03:
another two questions:
1.when I put the turtlebot on a random place , I found it always start locating on where I start building the map. It seems it's feasible when the map is just one room , but when it comes to a big area, it goes wrong.
2. Loading 3D map raise this error: Map received with code error 1!

matlabbe, 2018-08-03:
Kinect 2 depth image is more accurate (over 3 meters) than the one of kinect 1, so yes loop closure detections will be more precise. If you use visual odometry, it will be also more accurate with kinect v2. The problem with kinect v2 is that it uses quite a lot of computing resources and USB3 is required. 

If you are using turtlebot's wheel odometry, then the error is dependent of its accuracy. If you are using visual odometry, you can look at this [post](http://official-rtab-map-forum.67519.x6.nabble.com/How-to-process-RGBD-SLAM-datasets-with-RTAB-Map-td939.html) for example. On the [last](http://official-rtab-map-forum.67519.x6.nabble.com/How-to-process-RGBD-SLAM-datasets-with-RTAB-Map-tp939p4840.html) post, results show a 2 cm error.

For your two other questions:
1. Odometry is always restarted from 0. RTAB-Map creates a new map when this happens, so it is not a problem. When a loop closure will happen with a location of a previous map, the maps will be transformed accordingly.

2. What exactly are the steps to reproduce the error? Is it in rviz or rtabmapviz? from the same or remote computer?

kissmaidou, 2018-08-06:
hi,matlabbe.I have the same problem.  Useing RGB-D Handheld Mapping ,when i finished the map and starting to localization , I found the position is always displayed at the begining of start mapping .It is still inaccurate after reset odometry.

kissmaidou, 2018-08-06:
hi,matlabbe.I have the same problem.Use RGB-D Handheld Mapping, start positioning after the map is built, The kinect position  always displayed at the place where the construction started. After  reset odometry, it is still wrong.