introlab/rtabmap_ros #267: Got g2o Fatal error when using robot own odometry

ruifanwu, 2018-08-10:
Hi,

I was trying to use robot odometry in rtabmap, and I got g2o error everytime I ran it:

[FATAL] (2018-08-10 16:34:16.674) OptimizerG2O.cpp:537::optimize() Condition (optimizer.verifyInformationMatrices()) not met!

I found the similar error in the official rtambap forum. 
http://official-rtab-map-forum.67519.x6.nabble.com/g2o-error-on-loop-closure-td4707.html#a4725 

I checked the covariance of the twist and the values in diagonal positions are all less than 10e-3 or around. When I use visual odometry, this error doesn't show up. 

Thanks in advance,

matlabbe, 2018-08-14:
Are there off-diagonal values? This issue happens when one of the conditions [here](https://github.com/RainerKuemmerle/g2o/blob/8564e1e365d4e719dceca6269a0ba203f7f43dec/g2o/core/optimizable_graph.cpp#L981-L1013) is not valid.

ruifanwu, 2018-08-14:
Yes, there are off-diagonal values at last three rows of the matrix. Their values are all less than 10e-3. When this problem happened, I realize I did not install GSTAM. After I installed it, everything worked fine. But I still don't know what cause it.

matlabbe, 2018-08-14:
Without GTSAM, g2o is used for graph optimization. g2o looks more sensible to covariance format than GTSAM. If you have the latest version of rtabmap, the off-diagonal values are ignored by default (`Mem/CovOffDiagIgnored=true`) to avoid this kind of problem with g2o.

closed 2018-08-14 by matlabbe

IngenuityFV, 2025-12-16:
Hi

[FATAL] (2025-12-16 12:45:03.310) OptimizerG2O.cpp:1925::optimizeBA() Condition (optimizer.verifyInformationMatrices()) not met

I have used  the option Mem/CovOffDiagIgnored=true as  suggested and I have tried g2o and gtsam but I cannot solve this situation.

Any advices?

Thank you in advance