introlab/rtabmap_ros #268: double free or corruption (out)

embadmath, 2018-08-12:
Hi,
i'm trying to use rtabmap on ubuntu 18.04 with ros and even standalone. 
In standalone mode when i compile i get this error:
`[ 38%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/toro3d/treeoptimizer2.cpp.o
[ 38%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/OctoMap.cpp.o
[ 38%] Linking CXX shared library ../../../bin/librtabmap_core.so
/usr/bin/x86_64-linux-gnu-ld: cannot find -lQt4::QtSql
collect2: error: ld returned 1 exit status
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:2377: recipe for target '../bin/librtabmap_core.so.0.17.4' failed
make[2]: *** [../bin/librtabmap_core.so.0.17.4] Error 1
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
`
while i can get 100% compilation with catkin_make but when i try to use it i get this error:
`[ INFO] [1534117231.463109225]: rviz version 1.13.1
[ INFO] [1534117231.463161503]: compiled against Qt version 5.9.5
[ INFO] [1534117231.463174286]: compiled against OGRE version 1.9.0 (Ghadamon)
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[ INFO] [1534117231.660057253]: Stereo is NOT SUPPORTED
[ INFO] [1534117231.661253198]: OpenGl version: 3 (GLSL 1.3).
libpng warning: iCCP: known incorrect sRGB profile
double free or corruption (out)
Aborted (core dumped)
`
I'm using kinect v2.

Any suggestions?

thanks

matlabbe, 2018-08-14:
It is strange it tries to link on Qt4 libraries. Wiith 18.04, Qt5 should be used by default. See the [build logs](http://build.ros.org/job/Mbin_uB64__rtabmap__ubuntu_bionic_amd64__binary/12/consoleFull) of rtabmap on melodic: 
```
20:51:54 -- --------------------------------------------
20:51:54 -- Info :
20:51:54 --   Version : 0.17.1
20:51:54 --   CMAKE_INSTALL_PREFIX = /opt/ros/melodic
20:51:54 --   CMAKE_BUILD_TYPE =     None
20:51:54 --   CMAKE_INSTALL_LIBDIR = lib
20:51:54 --   BUILD_APP =            ON
20:51:54 --   BUILD_TOOLS =          ON
20:51:54 --   BUILD_EXAMPLES =       ON
20:51:54 --   BUILD_SHARED_LIBS =    ON
20:51:54 --   CMAKE_CXX_FLAGS = -g -O2 -fdebug-prefix-map=/tmp/binarydeb/ros-melodic-rtabmap-0.17.1=. -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2 -fmessage-length=0  -fopenmp -std=c++11
20:51:54 --   PCL_DEFINITIONS =  -march=native -msse4.2 -mfpmath=sse ;-DDISABLE_ENSENSO;-DDISABLE_DAVIDSDK;-DDISABLE_DSSDK;-DDISABLE_PCAP;-DDISABLE_PNG;-DDISABLE_LIBUSB_1_0;-DFLANN_STATIC;-Dqh_QHpointer;-DDISABLE_RSSDK
20:51:54 --   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = NO (not found, License: BSD)
20:51:54 --   With Freenect             = YES (License: Apache v2 and/or GPLv2)
20:51:54 --   With OpenNI2              = YES (License: Apache v2)
20:51:54 --   With Freenect2            = NO (libfreenect2 not found)
20:51:54 --   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
20:51:54 --   With dc1394               = YES (License: LGPL)
20:51:54 --   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
20:51:54 --   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
20:51:54 --   With g2o                  = YES (License: BSD)
20:51:54 --   With GTSAM                = NO (GTSAM not found)
20:51:54 --   With VERTIGO              = YES (License: GPLv3)
20:51:54 --   With cvsba                = NO (cvsba not found)
20:51:54 --   With libpointmatcher      = NO (libpointmatcher not found)
20:51:54 --   With ZED                  = NO (ZED sdk not found)
20:51:54 --   With RealSense            = NO (librealsense not found)
20:51:54 --   With OCTOMAP              = YES (License: BSD)
20:51:54 --   With CPUTSDF              = NO (CPUTSDF not found)
20:51:54 --   With OpenChisel           = NO (open_chisel not found)
20:51:54 --   With libfovis             = NO (libfovis not found)
20:51:54 --   With libviso2             = NO (libviso2 not found)
20:51:54 --   With dvo_core             = NO (dvo_core not found)
20:51:54 --   With okvis                = NO (okvis not found)
20:51:54 --   With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
20:51:54 --   With Qt5                  = YES (License: Open Source or Commercial)
20:51:54 -- --------------------------------------------
20:51:54 -- Configuring done
20:51:59 -- Generating done
```

> when i try to use it
What exactly are your commands?

embadmath, 2018-08-14:
Hi,
thanks for reply, i simply clone rtabmap from github in myhome:
cd myhome
git clone https://github.com/introlab/rtabmap.git
cd rtabmap
mkdir build
cd build
cmake ..
make -j3
and i get "lQt4::QtSql collect2: error: ld returned 1 exit status".
i install libqt4-dev but nothing change i get same error.

matlabbe, 2018-08-14:
What is the output of `cmake ..` ?

embadmath, 2018-08-15:
Hi,
i'm sorry, i installed ubuntu 16.04 and now it's working 100%.

Thanks

closed 2018-08-16 by matlabbe