introlab/rtabmap_ros #269: How to  eliminate  dynamic object from map while mapping

zoumaguanxin, 2018-08-15:
Hi,
I has been fusing the odometry, 2d laser range finder and depth camera to reconstruct surroundings by rtabmap. After that, I can get a 2d grid map for navigation, but I found that it built dynamic object as static obstacle. For example, my body was treated as obstacle because I always followed my robot while mapping, it is very terrible for robot path planning. 

To my knowledge, there should be a threshold so that the space where occupancy odds is less than it will not be treat as obstacle. I found a parameter named "GridGlobal/OctoMapOccupancyThr" in rtabmap, which seem to work only when we built map in octomap style.  Is that it? Are there other parameters that may be helpful for me?

if not, can you give me some advice to cope with my situation?

Thanks

matlabbe, 2018-08-15:
For the 2D occupancy map, there are no intermediate levels between free (value=0) and obstacle (value=100). You could use [LaserScanAngularBoundsFilter](http://wiki.ros.org/laser_filters#LaserScanAngularBoundsFilter) to ignore scans behind the robot (where you are standing). In general, dynamic obstacles would be cleared after they moved out of the field of view of the sensor. However, when someone is following the robot (or staying in field of view of the robot while moving), it will indeed add a trail of obstacles. I can take a look if it is possible to add a parameter to not have more levels of occupancy.

zoumaguanxin, 2018-08-16:
Hi,
I use LaserScanAngularBoundsFilter to ignore scans behind the robot, the trails of my body have been cleared from the built map. Of course, reducing of FOV of the lidar increases accumulative error of motion tracking, and global optimization can't fix it well. The used lidar is a low cost product, so  it has low rotation resolution，narrow range from 0.5 to 8m, and low rotation frequency 5~15Hz. It will be very grateful if you add some robust functions to reduce the compact of the dynamic objects in mapping process.

Thanks a lot

matlabbe, 2018-08-16:
I implemented the logodds approach from OctoMap library (see section 3.2 from their [paper](http://www2.informatik.uni-freiburg.de/~hornunga/pub/hornung13auro.pdf) for more info) to the standard map created from rtabmap (version 0.17.5).
Related parameters are:
```
Param: GridGlobal/OccupancyThr = "0.5"                     [Occupancy threshold (value between 0 and 1).]
Param: GridGlobal/ProbClampingMax = "0.971"                [Probability clamping maximum (value between 0 and 1).]
Param: GridGlobal/ProbClampingMin = "0.1192"               [Probability clamping minimum (value between 0 and 1).]
Param: GridGlobal/ProbHit = "0.7"                          [Probability of a hit (value between 0.5 and 1).]
Param: GridGlobal/ProbMiss = "0.4"                         [Probability of a miss (value between 0 and 0.5).]
```

The `grid_map` topic gives the same format as before, setting 0 for free and 100 for obstacle (Figure 1), but based on `GridGlobal/OccupancyThr` (setting `GridGlobal/OccupancyThr` to 0 will give exactly the same results than before). I added a new topic called `grid_prob_map`, which provides the grid map with probabilities between 0 and 100 (Figure 2).

1. [image]

2. [image]

zoumaguanxin, 2018-08-20:
I set the "GridGlobal/OccupancyThr" equal to  "0.65", the trails of dynamic objects were removed purely. Pretty good!

Thanks very much

closed 2018-08-20 by matlabbe