introlab/rtabmap_ros #270: How does the laser scan work in localization mode

zoumaguanxin, 2018-08-27:
Hi,

Setting the parameter Mem/IncrementalMemory" as false, I make the rtabmap work in localization mode. I got terrible result when the ego trajectory was different from the mapping trajectory although I provided it with a good initialpose . I found that the tabmap seemly don't take advantage of the laser scan well. In general, the robot can achieve robust motion tracking with low drift by computing the transform from the current scan to 2d map. How does the rtabmap make use of the laser scan in localization mode? Any suggestion will be glad to receive.

regards

matlabbe, 2018-08-29:
What kind of lidar do you use and what are the parameters used? Rtabmap does localization by loop closure detection (visual transform refined by laser scan if available) or by proximity detection (current scan is aligned with other close scans in the map). Accordingly to this paper : "[Long-Term Online Multi-Session Graph-Based SPLAM with
Memory Management](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/8/87/LabbeAURO2017.pdf)", see section 3.2 and 3.3 for a description of the loop closure detection approach and the proximity detection approach respectively.

If you have a lidar with large field of view (>=275 degrees), the robot would be able to localize in almost any orientations against the map using the proximity detection approach. For visual localization, the robot should indeed see the environment with the same orientation than in the mapping phase.

matlabbe, 2018-08-30:
I suggest to modify the following parameters with your setup:
```
<param name="Reg/Strategy" 	             type="string" value="1"/>
<param name="RGBD/ProximityPathMaxNeighbors" type="string" value="10"/>
```

When `RGBD/ProximityPathMaxNeighbors` is 0, the proximity detection using only laser scans is disabled. `Reg/Strategy` to 1 would be slightly faster without losing accuracy.

zoumaguanxin, 2018-08-31:
EXCELLENT. Your suggestions work.

By the way, another questions need your help. 

Whether the robot equipped with laser, RGBD camera and encoder is best for rtabmap to realize navigation? 

rtabmap has a excellent memory mangagement and provides much choices in different modules of  slam. In your experience, Is there a best  assembly for the robot navigation system ? for example, GFTT/ORB+pnp+PM+F2F+g2o. Is it best to be composed of the best?

Best,
Shawn

matlabbe, 2018-08-31:
The default parameters should do fine. To realize navigation indoor on a plane (2d slam), a 2D lidar with large field of view will give the best results (avoid using fake laser scan from kinect). The longer the range of the lidar is, the better is the robustness against empty spaces.

PM with Point2Plane enabled can help to be more robust to simple geometric areas, like long corridors where the scan sees only two parallel lines. We tested with many configurations RTAB-Map on the [MIT Stata Center data set](http://projects.csail.mit.edu/stata/), here we used Icp/VoxelSize = "0.05". The paper is not published yet (see preprint [here](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf)), but here some results related to this discussion using a lidar:
[image]

S2S and S2M are actually F2F and F2M approaches respectively when Reg/Strategy=1. WheelIMU is the use of encoder/IMU odometry directly in rtabmap. WheelIMU_refined means that RGBD/NeighborLinkRefining is true. WheelIMU->S2M/S2S means that [icp_odometry](https://github.com/introlab/rtabmap_ros/blob/master/src/nodelets/icp_odometry.cpp) node is used with WheelIMU as odometry [guess](https://github.com/introlab/rtabmap_ros/blob/8be9fdaaaf5811e0ae1d9481dfb0d5b55efb32b8/src/OdometryROS.cpp#L145-L147). The odometry guess from encoder is particularly important when having a short-range lidar to avoid large drift in corridor-like environments. Note also that rtabmap has been built with GTSAM, which is used by default for graph optimization.

zoumaguanxin, 2018-09-03:
Thanks a lot

closed 2020-08-24 by matlabbe