introlab/rtabmap_ros #271: Integration of ZED camera and IMU data

OgnjenMilicevic, 2018-08-28:
Hello!
I have found excellent examples how to integrate RTAB-MAP with ZED camera, but the results are somewhat imprecise for my application. I would like to see if the data can be improved by using an IMU (in contrast to extrapolating the movement from ZED camera).

Is there a way to integrate an IMU with this setup so this data is used in addition or instead of ZED camera position data?

matlabbe, 2018-08-29:
If you have a ZED Mini, you may try [their VIO approach](https://www.stereolabs.com/blog/introducing-zed-mini/). What you are looking for is a tightly coupled visual inertial odometry approach working with ZED instead of a loosely coupled sensor fusion using [robot_localization](http://docs.ros.org/melodic/api/robot_localization/html/index.html). The difficulty is to get a hardware setup for which the **camera and IMU are synchronized**. It is often not possible to do with an external IMU attached to a camera without buying a camera already coming with an IMU (like the ZED mini). Currently [MSCKF_VIO](https://github.com/KumarRobotics/msckf_vio) and [OKVIS](https://github.com/ethz-asl/okvis) are integrated in rtabmap (when built with these dependencies), but you can also use directly their ros packages as odometry input to rtabmap_ros, thus you can see before integrating to rtabmap if their approaches work ok for you. Rtabmap with those VIO approaches has only being tested on the [EuRoC](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets) dataset, which uses a VI-SENSOR (stereo synchronized with IMU sensor).

OgnjenMilicevic, 2018-09-17:
Thank you for a great answer!