introlab/rtabmap_ros #273: catkin_make error

koshish, 2018-09-12:
While doing `catkin_make` i am getting the  below error `Invoking "make -j4 -l4" failed`.  Please let me know what am I missing?

I am using  ROS Kinetic , Ubuntu 16.04, ZED Camera.

`hercules@hercules:~/robot_operating_system/first_workspace/catkin_ws/src$ git clone https://github.com/introlab/rtabmap_ros.git
hercules@hercules:~/robot_operating_system/first_workspace/catkin_ws/src$ cd rtabmap_ros/
hercules@hercules:~/robot_operating_system/first_workspace/catkin_ws/src/rtabmap_ros$ git tag -l
hercules@hercules:~/robot_operating_system/first_workspace/catkin_ws/src/rtabmap_ros$ git checkout tags/0.17.0-kinetic -b kinetic
hercules@hercules:~/robot_operating_system/first_workspace/catkin_ws$ catkin_make --pkg rtabmap_ros
Base path: /home/hercules/robot_operating_system/first_workspace/catkin_ws
Source space: /home/hercules/robot_operating_system/first_workspace/catkin_ws/src
Build space: /home/hercules/robot_operating_system/first_workspace/catkin_ws/build
Devel space: /home/hercules/robot_operating_system/first_workspace/catkin_ws/devel
Install space: /home/hercules/robot_operating_system/first_workspace/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/hercules/robot_operating_system/first_workspace/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/hercules/robot_operating_system/first_workspace/catkin_ws/build/rtabmap_ros"
####
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_NodeData
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_ListLabels
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_SetLabel
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_MapGraph
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_RGBDImage
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_MapData
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Link
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Point3f
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_UserData
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_SetGoal
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Goal
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_Point2f
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_ResetPose
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_GetMap
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_KeyPoint
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_GPS
[  0%] Built target _rtabmap_ros_generate_messages_check_deps_OdomInfo
[  1%] Built target rtabmap_ros_gencfg
[  1%] Built target _rtabmap_ros_generate_messages_check_deps_Info
[  1%] Built target std_msgs_generate_messages_lisp
[  1%] Built target _rtabmap_ros_generate_messages_check_deps_PublishMap
[  1%] Built target sensor_msgs_generate_messages_lisp
[  1%] Built target geometry_msgs_generate_messages_lisp
[  1%] Built target std_msgs_generate_messages_cpp
[  1%] Built target geometry_msgs_generate_messages_cpp
[  1%] Built target sensor_msgs_generate_messages_cpp
[  1%] Built target std_msgs_generate_messages_eus
[  1%] Built target sensor_msgs_generate_messages_eus
[  1%] Built target geometry_msgs_generate_messages_eus
[  1%] Built target std_msgs_generate_messages_nodejs
[  1%] Built target geometry_msgs_generate_messages_nodejs
[  1%] Built target sensor_msgs_generate_messages_nodejs
[  1%] Linking CXX executable /home/hercules/robot_operating_system/first_workspace/catkin_ws/devel/lib/rtabmap_ros/icp_odometry
[  1%] Linking CXX executable /home/hercules/robot_operating_system/first_workspace/catkin_ws/devel/lib/rtabmap_ros/stereo_odometry
[  1%] Linking CXX executable /home/hercules/robot_operating_system/first_workspace/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry
[  1%] Linking CXX executable /home/hercules/robot_operating_system/first_workspace/catkin_ws/devel/lib/rtabmap_ros/rtabmap
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreenect.so: undefined reference to `libusb_get_parent'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreenect.so: undefined reference to `libusb_error_name'
collect2: error: ld returned 1 exit status
rtabmap_ros/CMakeFiles/rtabmap.dir/build.make:1349: recipe for target '/home/hercules/robot_operating_system/first_workspace/catkin_ws/devel/lib/rtabmap_ros/rtabmap' failed
make[2]: *** [/home/hercules/robot_operating_system/first_workspace/catkin_ws/devel/lib/rtabmap_ros/rtabmap] Error 1
CMakeFiles/Makefile2:7290: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreenect.so: undefined reference to `libusb_get_parent'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreenect.so: undefined reference to `libusb_error_name'
collect2: error: ld returned 1 exit status
rtabmap_ros/CMakeFiles/icp_odometry.dir/build.make:1349: recipe for target '/home/hercules/robot_operating_system/first_workspace/catkin_ws/devel/lib/rtabmap_ros/icp_odometry' failed
make[2]: *** [/home/hercules/robot_operating_system/first_workspace/catkin_ws/devel/lib/rtabmap_ros/icp_odometry] Error 1
CMakeFiles/Makefile2:6892: recipe for target 'rtabmap_ros/CMakeFiles/icp_odometry.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/icp_odometry.dir/all] Error 2
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreenect.so: undefined reference to `libusb_get_parent'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreenect.so: undefined reference to `libusb_error_name'
collect2: error: ld returned 1 exit status
rtabmap_ros/CMakeFiles/rgbd_odometry.dir/build.make:1349: recipe for target '/home/hercules/robot_operating_system/first_workspace/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry' failed
make[2]: *** [/home/hercules/robot_operating_system/first_workspace/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry] Error 1
CMakeFiles/Makefile2:7221: recipe for target 'rtabmap_ros/CMakeFiles/rgbd_odometry.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rgbd_odometry.dir/all] Error 2
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreenect.so: undefined reference to `libusb_get_parent'
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libfreenect.so: undefined reference to `libusb_error_name'
collect2: error: ld returned 1 exit status
rtabmap_ros/CMakeFiles/stereo_odometry.dir/build.make:1349: recipe for target '/home/hercules/robot_operating_system/first_workspace/catkin_ws/devel/lib/rtabmap_ros/stereo_odometry' failed
make[2]: *** [/home/hercules/robot_operating_system/first_workspace/catkin_ws/devel/lib/rtabmap_ros/stereo_odometry] Error 1
CMakeFiles/Makefile2:7088: recipe for target 'rtabmap_ros/CMakeFiles/stereo_odometry.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/stereo_odometry.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
hercules@hercules:~/robot_operating_system/first_workspace/catkin_ws$
`

matlabbe, 2018-09-13:
On my system I cannot reproduce the error. The thing I see is that on my system it is linking on freenect installed by ros kinetic instead of the system one:
```
~/catkin_ws$ ldd devel/lib/librtabmap_ros.so | grep freenect
	libfreenect.so.0.5 => /opt/ros/kinetic/lib/libfreenect.so.0.5 (0x00007f6304c7f000)
~/catkin_ws$ ldd /opt/ros/kinetic/lib/libfreenect.so.0.5 
	linux-vdso.so.1 =>  (0x00007ffe937ec000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007ff6fa60b000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ff6fa241000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007ff6fabda000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007ff6fa024000)
	/lib64/ld-linux-x86-64.so.2 (0x00007ff6faa32000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007ff6f9e1c000)
```

If I look at the system one:
```
$ ldd /usr/lib/x86_64-linux-gnu/libfreenect.so.0.5
	linux-vdso.so.1 =>  (0x00007ffc6bf4f000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f87951ad000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f8794f90000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f8794bc6000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f879577c000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f87955d5000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f87949be000)
```
They link on same libusb library. The difference is that libfreenect in kinetic path is version 0.5.1 and on system path is 0.5.3. Not sure why on your system it is not linking on libfreenect from kinetic by default, did you source setup.bash of kinetic?

lunarpulse, 2018-11-04:
From Readmd.md of this repository:

9. Install RTAB-Map standalone libraries. Add -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel to cmake command below if you want to install in your Catkin's devel folder without sudo. **### Do not clone in your Catkin workspace.**

```

$ cd ~
$ git clone https://github.com/introlab/rtabmap.git rtabmap
$ cd rtabmap/build
$ cmake ..  [<---double dots included]
$ make
$ sudo make install
```

aguilaair, 2019-07-29:
This is kind of old but according to the next step you have to run catkin_make on you catkin dir after cloning it. How do I fix this?

matlabbe, 2019-07-29:
It is `rtabmap_ros`, not `rtabmap` that you should build in the `catkin_ws`. See steps 2 and 3 here: https://github.com/introlab/rtabmap_ros#build-from-source

CluelessIT, 2020-03-04:
Hi, I encounter this error when im trying to catkin_make -j2 for step 10 building from source for jetson nano. https://github.com/introlab/rtabmap_ros#build-from-source 
*Im doing it for jetson nano using ubuntu 18.04 ros melodic.

Base path: /home/mole/catkin_ws
Source space: /home/mole/catkin_ws/src
Build space: /home/mole/catkin_ws/build
Devel space: /home/mole/catkin_ws/devel
Install space: /home/mole/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/mole/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/mole/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/mole/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 9 packages in topological order:
-- ~~  - image_transport_plugins (metapackage)
-- ~~  - opencv_tests
-- ~~  - vision_opencv (metapackage)
-- ~~  - cv_bridge
-- ~~  - image_geometry
-- ~~  - compressed_depth_image_transport
-- ~~  - compressed_image_transport
-- ~~  - theora_image_transport
-- ~~  - rtabmap_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'image_transport_plugins'
-- ==> add_subdirectory(image_transport_plugins/image_transport_plugins)
-- +++ processing catkin package: 'opencv_tests'
-- ==> add_subdirectory(vision_opencv/opencv_tests)
-- +++ processing catkin metapackage: 'vision_opencv'
-- ==> add_subdirectory(vision_opencv/vision_opencv)
-- +++ processing catkin package: 'cv_bridge'
-- ==> add_subdirectory(vision_opencv/cv_bridge)
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   python
CMake Error at vision_opencv/cv_bridge/CMakeLists.txt:16 (find_package):
  Could not find a configuration file for package "OpenCV" that is compatible
  with requested version "3".

  The following configuration files were considered but not accepted:

    /usr/lib/aarch64-linux-gnu/cmake/opencv4/OpenCVConfig.cmake, version: 4.1.1

-- Configuring incomplete, errors occurred!
See also "/home/mole/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/mole/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:320: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

It seems like it need openCV3 but mine is openCV4. Any idea how i can force it to think that it is openCV3? or any other solutions are welcome! Thanks~

aguilaair, 2020-03-04:
@CluelessIT this is what i found out worked (this is one of my projects) https://learn.todbot.co/set-up-todbot/software#installing-opencv

matlabbe, 2020-03-10:
@CluelessIT You may try changing 3 by 4 here: https://github.com/ros-perception/vision_opencv/blob/0bbebb73bea65eca856e732b919439853a04986b/cv_bridge/CMakeLists.txt#L16

tdam2112, 2020-03-18:
> @CluelessIT You may try changing 3 by 4 here: https://github.com/ros-perception/vision_opencv/blob/0bbebb73bea65eca856e732b919439853a04986b/cv_bridge/CMakeLists.txt#L16

had similar error, changed 3 to 4 as mentioned, it went away but still not finishing successfully.
this is wha it outputs:

`nvidiaagx@nvidiaagx-desktop:~/catkin_ws$ catkin_make -j2
Base path: /home/nvidiaagx/catkin_ws
Source space: /home/nvidiaagx/catkin_ws/src
Build space: /home/nvidiaagx/catkin_ws/build
Devel space: /home/nvidiaagx/catkin_ws/devel
Install space: /home/nvidiaagx/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/nvidiaagx/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/nvidiaagx/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nvidiaagx/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 9 packages in topological order:
-- ~~  - image_transport_plugins (metapackage)
-- ~~  - opencv_tests
-- ~~  - vision_opencv (metapackage)
-- ~~  - cv_bridge
-- ~~  - image_geometry
-- ~~  - compressed_depth_image_transport
-- ~~  - compressed_image_transport
-- ~~  - theora_image_transport
-- ~~  - rtabmap_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'image_transport_plugins'
-- ==> add_subdirectory(image_transport_plugins/image_transport_plugins)
-- +++ processing catkin package: 'opencv_tests'
-- ==> add_subdirectory(vision_opencv/opencv_tests)
-- +++ processing catkin metapackage: 'vision_opencv'
-- ==> add_subdirectory(vision_opencv/vision_opencv)
-- +++ processing catkin package: 'cv_bridge'
-- ==> add_subdirectory(vision_opencv/cv_bridge)
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   python
-- Found OpenCV: /usr (found suitable version "4.1.1", minimum required is "4") found components:  opencv_core opencv_imgproc opencv_imgcodecs 
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython2.7.so (found suitable version "2.7.17", minimum required is "2.7") 
-- +++ processing catkin package: 'image_geometry'
-- ==> add_subdirectory(vision_opencv/image_geometry)
-- Found OpenCV: /usr (found version "4.1.1") 
-- +++ processing catkin package: 'compressed_depth_image_transport'
-- ==> add_subdirectory(image_transport_plugins/compressed_depth_image_transport)
-- +++ processing catkin package: 'compressed_image_transport'
-- ==> add_subdirectory(image_transport_plugins/compressed_image_transport)
-- +++ processing catkin package: 'theora_image_transport'
-- ==> add_subdirectory(image_transport_plugins/theora_image_transport)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- theora_image_transport: 1 messages, 0 services
-- +++ processing catkin package: 'rtabmap_ros'
-- ==> add_subdirectory(rtabmap_ros)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
[... 156 lines cut ...]
/usr/include/opencv4/opencv2/core/mat.hpp:478:18: note: 	virtual void cv::MatAllocator::deallocate(cv::UMatData*) const
     virtual void deallocate(UMatData* data) const = 0;
                  ^~~~~~~~~~
/home/nvidiaagx/catkin_ws/src/vision_opencv/cv_bridge/src/module_opencv2.cpp: In function ‘int convert_to_CvMat2(const PyObject*, cv::Mat&)’:
/home/nvidiaagx/catkin_ws/src/vision_opencv/cv_bridge/src/module_opencv2.cpp:233:11: error: ‘class cv::Mat’ has no member named ‘refcount’
         m.refcount = refcountFromPyObject(o);
           ^~~~~~~~
/home/nvidiaagx/catkin_ws/src/vision_opencv/cv_bridge/src/module_opencv2.cpp: In function ‘PyObject* pyopencv_from(const cv::Mat&)’:
/home/nvidiaagx/catkin_ws/src/vision_opencv/cv_bridge/src/module_opencv2.cpp:254:12: error: ‘class cv::Mat’ has no member named ‘refcount’
     if(!p->refcount || p->allocator != &g_numpyAllocator)
            ^~~~~~~~
/home/nvidiaagx/catkin_ws/src/vision_opencv/cv_bridge/src/module_opencv2.cpp:261:36: error: ‘class cv::Mat’ has no member named ‘refcount’
     return pyObjectFromRefcount(p->refcount);
                                    ^~~~~~~~
vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/build.make:86: recipe for target 'vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module_opencv2.cpp.o' failed
make[2]: *** [vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module_opencv2.cpp.o] Error 1
CMakeFiles/Makefile2:1059: recipe for target 'vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/all' failed
make[1]: *** [vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 72%] Linking CXX executable /home/nvidiaagx/catkin_ws/devel/lib/rtabmap_ros/map_optimizer
[ 72%] Built target rtabmap_map_optimizer
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2" failed
`

matlabbe, 2020-04-02:
It is a [vision_opencv](https://github.com/ros-perception/vision_opencv) issue with OpenCV4. Look at this post: https://github.com/ros-perception/vision_opencv/issues/272

You may try this pull request (worked for me): https://github.com/ros-perception/vision_opencv/pull/288

cesarecaputi, 2020-04-17:
Hi, I got this error even though all dependencies have been downloaded.
...
Scanning dependencies of target rtabmap_rgbd_odometry
[ 87%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_rgbd_odometry.dir/src/RGBDOdometryNode.cpp.o
/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp: In function ‘std::map<std::__cxx11::basic_string<char>, float> rtabmap_ros::odomInfoToStatistics(const rtabmap::OdometryInfo&)’:
/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:1139:78: error: ‘const class rtabmap::RegistrationInfo’ has no member named ‘icpStructuralDistribution’; did you mean ‘inliersDistribution’?
 stats.insert(std::make_pair("Odometry/ICPStructuralDistribution/", info.reg.icpStructuralDistribution));
                                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~
                                                                              inliersDistribution
/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:1140:71: error: ‘const class rtabmap::RegistrationInfo’ has no member named ‘icpCorrespondences’
  stats.insert(std::make_pair("Odometry/ICPCorrespondences/", info.reg.icpCorrespondences));
                                                                       ^~~~~~~~~~~~~~~~~~
/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp: In function ‘rtabmap::OdometryInfo rtabmap_ros::odomInfoFromROS(const OdomInfo&)’:
/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:1219:11: error: ‘class rtabmap::RegistrationInfo’ has no member named ‘icpStructuralDistribution’; did you mean ‘inliersDistribution’?
  info.reg.icpStructuralDistribution = msg.icpStructuralDistribution;
           ^~~~~~~~~~~~~~~~~~~~~~~~~
           inliersDistribution
/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:1220:11: error: ‘class rtabmap::RegistrationInfo’ has no member named ‘icpCorrespondences’
  info.reg.icpCorrespondences = msg.icpCorrespondences;
           ^~~~~~~~~~~~~~~~~~
/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp: In function ‘void rtabmap_ros::odomInfoToROS(const rtabmap::OdometryInfo&, rtabmap_ros::OdomInfo&)’:
/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:1277:43: error: ‘const class rtabmap::RegistrationInfo’ has no member named ‘icpStructuralDistribution’; did you mean ‘inliersDistribution’?
  msg.icpStructuralDistribution = info.reg.icpStructuralDistribution;
                                           ^~~~~~~~~~~~~~~~~~~~~~~~~
                                           inliersDistribution
/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:1278:36: error: ‘const class rtabmap::RegistrationInfo’ has no member named ‘icpCorrespondences’
  msg.icpCorrespondences = info.reg.icpCorrespondences;
                                    ^~~~~~~~~~~~~~~~~~
[ 87%] Building CXX object navigation/costmap_2d/CMakeFiles/layers.dir/plugins/voxel_layer.cpp.o
rtabmap_ros/CMakeFiles/rtabmap_ros.dir/build.make:62: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MsgConversion.cpp.o' failed
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MsgConversion.cpp.o] Error 1
make[2]: *** Attesa per i processi non terminati....
Scanning dependencies of target rtabmap
[ 87%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap.dir/src/CoreNode.cpp.o
[ 87%] Linking CXX executable /home/cesare_caputi/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry
[ 87%] Linking CXX executable /home/cesare_caputi/catkin_ws/devel/lib/rtabmap_ros/rtabmap
[ 87%] Built target rtabmap_rgbd_odometry
Scanning dependencies of target rtabmap_icp_odometry
[ 87%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_icp_odometry.dir/src/ICPOdometryNode.cpp.o
[ 87%] Building CXX object navigation/costmap_2d/CMakeFiles/layers.dir/src/observation_buffer.cpp.o
[ 87%] Built target rtabmap
Scanning dependencies of target rtabmap_imu_to_tf
[ 87%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_imu_to_tf.dir/src/ImuToTFNode.cpp.o
CMakeFiles/Makefile2:36670: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all] Error 2
make[1]: *** Attesa per i processi non terminati....
[ 88%] Linking CXX executable /catkin_ws/devel/lib/rtabmap_ros/icp_odometry
[ 88%] Linking CXX executable /catkin_ws/devel/lib/rtabmap_ros/imu_to_tf
[ 88%] Linking CXX shared library /catkin_ws/devel/lib/liblayers.so
[ 88%] Built target layers
[ 88%] Built target rtabmap_imu_to_tf
[ 88%] Built target rtabmap_icp_odometry
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Does anyone of you know why? How could I fix it?

matlabbe, 2020-04-18:
The OdomInfo msg has been updated recently. There is a mismatch between rtabmap_ros and rtabmap versions on your side. Make sure to re-update rtabmap library and reinstall it, then try rebuild rtabmap_ros.