introlab/rtabmap_ros #276: record and play data using rtabmap

zoumaguanxin, 2018-09-25:
Hi,
   I would like to use `rtabmap` to record data as db file, and use `data_player` to play db file because recording rosbag engrosses my disk space. But the message type of the topic named "rgbd_image" is unknown, I don't know which topic I should remap it to. My launch file about camera also really doesn't publish it.  Rtabmap mapping work well even though the topic is not available, so  why to subscribe it  necessarily.   Without that topic, I can not record any db file.  
    In addition, the parameters demonstration about `Rtabmap/* ` seemly are difficult to understand. Would you mind to provide longer comments. Readers usually care which function the parameter contributes to but its define. The type of `Mem/LaserScanNormalRadius` might be `float` while it is `int` now.
Regards,
Shawn

matlabbe, 2018-09-27:
How did you launch [data_recorder.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/data_recorder.launch)? If I launch like the following the required inputs are the default ones (rgb, depth and camera_info):
```bash
$ roslaunch rtabmap_ros data_recorder.launch
...
/data_recorder subscribed to (approx sync):
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info
```
To replicate your case:
```bash
$ roslaunch rtabmap_ros data_recorder.launch subscribe_odometry:=true subscribe_scan:=true subscribe_rgbd:=true
...
/data_recorder subscribed to (approx sync):
   /odom,
   /camera/rgbd_image,
   /scan

```
The rgbd_image is a msg generated by the [rtabmap_ros/rgbd_sync](http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2BAC8-rgbd_sync) nodelet (I just updated the ros wiki with this info). An example of usage is shown in [test_rgbd_image.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/test_rgbd_image.launch).

Can you give a link to the demonstration in question? If you mean the descriptions in the [data_recorder.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/data_recorder.launch) file, they are "hacks" to make rtabmap just record to database without any other processing. For description of the parameters (with your current installed version), do:
```bash
$ rtabmap --params | grep Rtabmap/
Param: Rtabmap/ComputeRMSE = "true"                        [Compute root mean square error (RMSE) and publish it in statistics, if ground truth is provided.]
Param: Rtabmap/CreateIntermediateNodes = "false"           [Create intermediate nodes between loop closure detection. Only used when Rtabmap/DetectionRate>0.]
Param: Rtabmap/DetectionRate = "1"                         [Detection rate (Hz). RTAB-Map will filter input images to satisfy this rate.]
Param: Rtabmap/ImageBufferSize = "1"                       [Data buffer size (0 min inf).]
Param: Rtabmap/ImagesAlreadyRectified = "true"             [Images are already rectified. By default RTAB-Map assumes that received images are rectified. If they are not, they can be rectified by RTAB-Map if this parameter is false.]
Param: Rtabmap/LoopRatio = "0"                             [The loop closure hypothesis must be over LoopRatio x lastHypothesisValue.]
Param: Rtabmap/LoopThr = "0.11"                            [Loop closing threshold.]
Param: Rtabmap/MaxRetrieved = "2"                          [Maximum locations retrieved at the same time from LTM.]
Param: Rtabmap/MemoryThr = "0"                             [Maximum signatures in the Working Memory (ms) (0 means infinity).]
Param: Rtabmap/PublishLastSignature = "true"               [Publishing last signature.]
Param: Rtabmap/PublishLikelihood = "true"                  [Publishing likelihood.]
Param: Rtabmap/PublishPdf = "true"                         [Publishing pdf.]
Param: Rtabmap/PublishRAMUsage = "false"                   [Publishing RAM usage in statistics (may add a small overhead to get info from the system).]
Param: Rtabmap/PublishStats = "true"                       [Publishing statistics.]
Param: Rtabmap/RectifyOnlyFeatures = "false"               [If "Rtabmap/ImagesAlreadyRectified" is false and this parameter is true, the whole RGB image will not be rectified, only the features. Warning: As projection of RGB-D image to point cloud is assuming that images are rectified, the generated point cloud map will have wrong colors if this parameter is true.]
Param: Rtabmap/SaveWMState = "false"                       [Save working memory state after each update in statistics.]
Param: Rtabmap/StartNewMapOnGoodSignature = "false"        [Start a new map only if the first signature is not bad (i.e., has enough features, see Kp/BadSignRatio).]
Param: Rtabmap/StartNewMapOnLoopClosure = "false"          [Start a new map only if there is a global loop closure with a previous map.]
Param: Rtabmap/StatisticLogged = "false"                   [Logging enabled.]
Param: Rtabmap/StatisticLoggedHeaders = "true"             [Add column header description to log files.]
Param: Rtabmap/StatisticLogsBufferedInRAM = "true"         [Statistic logs buffered in RAM instead of written to hard drive after each iteration.]
Param: Rtabmap/TimeThr = "0"                               [Maximum time allowed for the detector (ms) (0 means infinity).]
Param: Rtabmap/WorkingDirectory = ""                       [Working directory.]

```

For `Mem/LaserScanNormalRadius`, the parameter should be indeed a float, this is fixed in this [commit](https://github.com/introlab/rtabmap/commit/3b7453456756361912a66fae7327ba07acbfc4b9). Thx!

cheers,
Mathieu

zoumaguanxin, 2018-09-28:
Very sorry to confuse you. Actually, I open an issue for two class questions, they have a little relation between them. I am using latest Version 0.17.5.
#  For parameter demonstrations
 I mean,  for example
```
Rtabmap/TimeThr = "0"                               [Maximum time allowed for the detector (ms) (0 means infinity)
```
In your [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf), Rtabmap/TimeThr is a fixed time threshold some nodes in WM are transferred to LTM  to limit the size of the EM and decrease the update time. But in here, it means maximum time allowed for the detector . Is the two descriptions equal? I found that the demonstrations of parameters included in 'Rtabmp/*' are given shorter demonstration compared to demonstrations of other parameters. You can consider providing some longer comments for them. Just a suggestion.

# For data_recorder launch
Now, I use the following configure.
```
  <arg name="subscribe_odometry"   default="true"/><!--default false-->
  <arg name="subscribe_depth"        default="true"/>
  <arg name="subscribe_stereo"       default="false"/>
  <arg name="subscribe_rgbd"         default="false"/>
  <arg name="subscribe_scan"         default="true"/><!--default false-->
```
It worked. Later on, I run the follwing command to play db file.
`rosrun rtabmap_ros data_player _database="PATH/my.db"`
At the same, a  mapping launch was started, it started rtabmap. I found that mapping using the recorded db file got terrible result compared to using rosbag, it is noted that I use completely same setting of parameters. Did this situation happen to you? if did, how did you solve it.  
[image]
[image]

zoumaguanxin, 2018-09-30:
It works. It is noted that the parameters of laser scan must be consistent with the recorded scan. In general, these parameters should not be necessary to set when playing a recorded file.

I updated the latest version rtabmap that you updated in yesterday. rtabmapviz now can not show any data, but old 0.17.5 version is ok. Did you have same problem?   
Regards,
Shawn

matlabbe, 2018-09-30:
For the laser scan data, the default parameters should simulate most laser scans. I am supposed to refactor camera calibration data saved in database, I added the refactoring of laser scan to save more parameters, see https://github.com/introlab/rtabmap/issues/313. The `time_increment` should still be null afterwards as the scans are already [corrected](https://github.com/introlab/rtabmap_ros/blob/803be533bcc35457ca4332e4af4a4c573153de42/src/MsgConversion.cpp#L1583) in the database.

Not sure about the rtabmapviz problem, as it seems to work on my system, folliwing example above (after making sure `use_sim_time` is false in `demo_robot_mapping.launch`):
```bash
$ roslaunch rtabmap_ros demo_robot_mapping.launch
$ rosrun rtabmap_ros data_player _database:=~/.ros/output.db _frame_id:=base_footprint odom:=/az3/base_controller/odom scan:=/jn0/base_scan rgb/image:=data_throttled_image depth_registered/image:=data_throttled_image_depth /rgb/camera_info:=data_throttled_camera_info
``` 
[image]