introlab/rtabmap_ros #280: test_prior.launch: "Rejecting all added loop closures" if cov of priors is set to 0.0001

matlabbe, 2018-10-11:
From:
```
$ roslaunch rtabmap_ros test_prior.launch
$ ./test_prior_tf_to_pose.py _cov:=0.0001
$ rosbag play --clock rgbd_dataset_freiburg3_long_office_household_tf_renamed.bag
```

See log:
```
[ WARN] (2018-10-11 19:10:50.508) Rtabmap.cpp:2386::process() Rejecting all added loop closures (2) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 3.470248 (edge 70->71, type=0, abs error=0.055587 m, stddev=0.016018). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 1.000000 of std deviation.
[ WARN] (2018-10-11 19:10:50.508) Rtabmap.cpp:2408::process() Rejecting all added loop closures (2) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 1.302817 (edge 31->32, type=0, abs error=4.891232 deg, stddev=0.065526). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 1.000000 of std deviation.
[ WARN] (2018-10-11 19:10:50.508) Rtabmap.cpp:2418::process() Loop closure 73->6 rejected!
[ WARN] (2018-10-11 19:10:50.508) Rtabmap.cpp:2418::process() Loop closure 73->4 rejected!
[ INFO] [1539299450.512488995, 1341848056.455781460]: rtabmap (73): Rate=1.00s, Limit=0.000s, RTAB-Map=0.0974s, Maps update=0.0018s pub=0.0001s (local map=73, WM=73)
[ INFO] [1539299450.514897447, 1341848056.465837518]: Odom: quality=427, std dev=0.017854m|0.058373rad, update time=0.051639s
[ INFO] [1539299450.607815966, 1341848056.556369688]: Odom: quality=198, std dev=0.031962m|0.071095rad, update time=0.043878s
[ INFO] [1539299450.724124755, 1341848056.667434888]: Odom: quality=390, std dev=0.019982m|0.060540rad, update time=0.058571s
[ INFO] [1539299450.826715499, 1341848056.778309324]: Odom: quality=315, std dev=0.028033m|0.070438rad, update time=0.069808s
[ INFO] [1539299450.902697482, 1341848056.848846447]: Odom: quality=236, std dev=0.037114m|0.077212rad, update time=0.046104s
[ INFO] [1539299451.011063082, 1341848056.959862955]: Odom: quality=501, std dev=0.014369m|0.058466rad, update time=0.053889s
[ INFO] [1539299451.134397233, 1341848057.080591264]: Odom: quality=345, std dev=0.020451m|0.071026rad, update time=0.074918s
[ INFO] [1539299451.207607736, 1341848057.151122443]: Odom: quality=357, std dev=0.018418m|0.057552rad, update time=0.048457s
[ INFO] [1539299451.311047584, 1341848057.261897166]: Odom: quality=254, std dev=0.025349m|0.059396rad, update time=0.050998s
[ INFO] [1539299451.449480118, 1341848057.392826885]: Odom: quality=530, std dev=0.012807m|0.051257rad, update time=0.087971s
[ INFO] [1539299451.512894862, 1341848057.463695944]: Odom: quality=455, std dev=0.015080m|0.053404rad, update time=0.050944s
[ WARN] (2018-10-11 19:10:51.589) Rtabmap.cpp:2386::process() Rejecting all added loop closures (1) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 3.469422 (edge 70->71, type=0, abs error=0.055574 m, stddev=0.016018). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 1.000000 of std deviation.
[ WARN] (2018-10-11 19:10:51.589) Rtabmap.cpp:2408::process() Rejecting all added loop closures (1) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 1.302817 (edge 31->32, type=0, abs error=4.891232 deg, stddev=0.065526). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 1.000000 of std deviation.
[ WARN] (2018-10-11 19:10:51.589) Rtabmap.cpp:2418::process() Loop closure 74->6 rejected!
[ INFO] [1539299451.593845400, 1341848057.544175730]: rtabmap (74): Rate=1.00s, Limit=0.000s, RTAB-Map=0.0698s, Maps update=0.0018s pub=0.0001s (local map=74, WM=74)
[ INFO] [1539299451.610245947, 1341848057.554236452]: Odom: quality=359, std dev=0.018216m|0.060164rad, update time=0.048323s
[ INFO] [1539299451.707471341, 1341848057.655051179]: Odom: quality=319, std dev=0.019998m|0.053844rad, update time=0.044828s
[ INFO] [1539299451.825982032, 1341848057.775918749]: Odom: quality=251, std dev=0.019333m|0.055048rad, update time=0.060998s
[ INFO] [1539299451.929874531, 1341848057.876674231]: Odom: quality=396, std dev=0.017779m|0.063089rad, update time=0.065215s
[ INFO] [1539299452.029654182, 1341848057.978510926]: Odom: quality=353, std dev=0.021403m|0.058648rad, update time=0.062337s
[ INFO] [1539299452.111432794, 1341848058.059120225]: Odom: quality=242, std dev=0.022616m|0.065384rad, update time=0.044225s
[ INFO] [1539299452.211939146, 1341848058.159802142]: Odom: quality=348, std dev=0.027895m|0.067146rad, update time=0.044410s
[ INFO] [1539299452.335302335, 1341848058.280796671]: Odom: quality=227, std dev=0.031481m|0.073385rad, update time=0.073525s
[ INFO] [1539299452.418834338, 1341848058.362004618]: Odom: quality=304, std dev=0.021849m|0.071784rad, update time=0.057228s
[ INFO] [1539299452.506434781, 1341848058.452734194]: Odom: quality=358, std dev=0.025200m|0.059756rad, update time=0.045431s
[ WARN] (2018-10-11 19:10:52.575) Rtabmap.cpp:2386::process() Rejecting all added loop closures (1) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 3.469600 (edge 70->71, type=0, abs error=0.055576 m, stddev=0.016018). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 1.000000 of std deviation.
[ WARN] (2018-10-11 19:10:52.575) Rtabmap.cpp:2408::process() Rejecting all added loop closures (1) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 1.302817 (edge 31->32, type=0, abs error=4.891232 deg, stddev=0.065526). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 1.000000 of std deviation.
[ WARN] (2018-10-11 19:10:52.575) Rtabmap.cpp:2418::process() Loop closure 75->6 rejected!
[ INFO] [1539299452.579384476, 1341848058.523068474]: rtabmap (75): Rate=1.00s, Limit=0.000s, RTAB-Map=0.0670s, Maps update=0.0017s pub=0.0001s (local map=75, WM=75)
[ INFO] [1539299452.604149741, 1341848058.553466831]: Odom: quality=202, std dev=0.027846m|0.070698rad, update time=0.043349s
[ INFO] [1539299452.710622554, 1341848058.654177934]: Odom: quality=454, std dev=0.017754m|0.051435rad, update time=0.048019s
[ INFO] [1539299452.825444207, 1341848058.775059389]: Odom: quality=443, std dev=0.023754m|0.061119rad, update time=0.062084s
[ INFO] [1539299452.913371420, 1341848058.865729774]: Odom: quality=342, std dev=0.021971m|0.057180rad, update time=0.049269s
[ INFO] [1539299453.008694425, 1341848058.956249041]: Odom: quality=247, std dev=0.017206m|0.067567rad, update time=0.044322s
[ INFO] [1539299453.115323817, 1341848059.067300621]: Odom: quality=462, std dev=0.018577m|0.060758rad, update time=0.050421s
[ INFO] [1539299453.220336318, 1341848059.168020255]: Odom: quality=451, std dev=0.022806m|0.070069rad, update time=0.054023s
[ INFO] [1539299453.321328195, 1341848059.268636450]: Odom: quality=448, std dev=0.015168m|0.057103rad, update time=0.055136s
[ INFO] [1539299453.418432494, 1341848059.369380064]: Odom: quality=326, std dev=0.021187m|0.069263rad, update time=0.051473s
[ INFO] [1539299453.519777374, 1341848059.470203367]: Odom: quality=432, std dev=0.017840m|0.055719rad, update time=0.051549s
[ INFO] [1539299453.623979697, 1341848059.571326818]: Odom: quality=357, std dev=0.018818m|0.058583rad, update time=0.055291s
[ WARN] (2018-10-11 19:10:53.671) Rtabmap.cpp:1925::process() Rejected loop closure 6 -> 76: Not enough inliers 17/20 (matches=25) between 6 and 76
[ INFO] [1539299453.679556704, 1341848059.631684941]: rtabmap (76): Rate=1.00s, Limit=0.000s, RTAB-Map=0.0485s, Maps update=0.0018s pub=0.0001s (local map=76, WM=76)
[ INFO] [1539299453.719795790, 1341848059.671920471]: Odom: quality=391, std dev=0.016994m|0.055212rad, update time=0.049467s
[ INFO] [1539299453.819988270, 1341848059.772659139]: Odom: quality=348, std dev=0.025093m|0.052812rad, update time=0.049174s
[ INFO] [1539299453.917984876, 1341848059.863452205]: Odom: quality=307, std dev=0.026648m|0.058784rad, update time=0.046917s
[ INFO] [1539299454.008288229, 1341848059.954052336]: Odom: quality=237, std dev=0.022764m|0.068618rad, update time=0.046047s
[ INFO] [1539299454.111376334, 1341848060.054831108]: Odom: quality=432, std dev=0.019479m|0.055523rad, update time=0.048660s
[... cut ...]
[ WARN] (2018-10-11 19:11:06.635) Rtabmap.cpp:2408::process() Rejecting all added loop closures (1) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 1.302817 (edge 31->32, type=0, abs error=4.891232 deg, stddev=0.065526). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 1.000000 of std deviation.
[ WARN] (2018-10-11 19:11:06.635) Rtabmap.cpp:2418::process() Loop closure 84->71 rejected!
```
The resulting map doesn't have any loop closures.

TSC21, 2018-10-12:
@matlabbe I can't reproduce this issue. I tried with the covariances set to `0.0001` and I do get loop closures.

TSC21, 2018-10-13:
actually I just confirmed this on my side. here's the resulting g2o file:
```
PARAMS_SE3OFFSET 0 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 1.000000

VERTEX_SE3:QUAT 1 -0.705456 2.679623 1.812445 0.241809 0.018992 -0.028851 0.969709
VERTEX_SE3:QUAT 2 -0.613120 2.710330 1.807182 0.238696 0.014159 -0.041716 0.970095
VERTEX_SE3:QUAT 3 -0.479260 2.706206 1.814073 0.243767 0.013114 -0.077323 0.966658
VERTEX_SE3:QUAT 4 -0.042863 2.709086 1.806556 0.232914 0.002067 -0.117209 0.965406
VERTEX_SE3:QUAT 5 0.263171 2.455487 1.714021 0.252617 -0.011402 -0.130302 0.958685
VERTEX_SE3:QUAT 6 0.393091 2.101152 1.705947 0.225543 -0.026509 -0.214131 0.950040
VERTEX_SE3:QUAT 7 0.455115 1.677541 1.689781 0.240667 -0.063619 -0.303251 0.919821
VERTEX_SE3:QUAT 8 0.591847 1.362139 1.574901 0.198777 -0.054712 -0.387258 0.898624
VERTEX_SE3:QUAT 9 0.789383 1.210625 1.557745 0.202991 -0.076415 -0.416963 0.882665
VERTEX_SE3:QUAT 10 1.101432 1.128568 1.616405 0.205120 -0.058394 -0.439925 0.872343
VERTEX_SE3:QUAT 11 1.352802 0.985675 1.618760 0.195778 -0.108726 -0.484859 0.845436
VERTEX_SE3:QUAT 12 1.472688 0.816619 1.651581 0.189201 -0.108386 -0.565040 0.795730
VERTEX_SE3:QUAT 13 1.658940 0.538405 1.649871 0.176924 -0.127231 -0.625585 0.749102
VERTEX_SE3:QUAT 14 1.721208 0.237765 1.668717 0.165853 -0.134530 -0.678848 0.702538
VERTEX_SE3:QUAT 15 1.799733 -0.099256 1.661383 0.146044 -0.155668 -0.743856 0.633337
VERTEX_SE3:QUAT 16 1.789868 -0.265478 1.659595 0.129917 -0.178291 -0.782994 0.581596
VERTEX_SE3:QUAT 17 1.814389 -0.573805 1.650047 0.123312 -0.176242 -0.819666 0.530924
VERTEX_SE3:QUAT 18 1.655201 -0.944235 1.657961 0.132670 -0.175428 -0.826467 0.518242
VERTEX_SE3:QUAT 19 1.517099 -1.251205 1.625652 -0.079937 0.191493 0.877004 -0.433365
VERTEX_SE3:QUAT 20 1.223326 -1.285523 1.600216 -0.043039 0.239222 0.922197 -0.300789
VERTEX_SE3:QUAT 21 0.946966 -1.257109 1.577344 -0.028420 0.229539 0.957249 -0.173720
VERTEX_SE3:QUAT 22 0.687812 -1.142961 1.524131 -0.019929 0.260428 0.960760 -0.093383
VERTEX_SE3:QUAT 23 0.435285 -1.089067 1.482701 -0.075618 0.234814 0.953024 -0.175754
VERTEX_SE3:QUAT 24 0.219242 -0.948806 1.397651 -0.136521 0.262199 0.905245 -0.305198
VERTEX_SE3:QUAT 25 0.113005 -0.988345 1.378798 -0.162107 0.248048 0.867316 -0.399945
VERTEX_SE3:QUAT 26 0.012247 -1.090996 1.390853 -0.138205 0.219572 0.853695 -0.451545
VERTEX_SE3:QUAT 27 -0.129920 -1.342199 1.446897 -0.074915 0.254636 0.909787 -0.319117
VERTEX_SE3:QUAT 28 -0.308471 -1.476139 1.477816 -0.042215 0.278724 0.938841 -0.197760
VERTEX_SE3:QUAT 29 -0.522275 -1.448336 1.435386 -0.034638 0.296239 0.943449 -0.144733
VERTEX_SE3:QUAT 30 -0.609107 -1.247333 1.460200 -0.009451 0.316294 0.947110 -0.053402
VERTEX_SE3:QUAT 31 -0.612885 -1.120820 1.433336 0.010491 0.336815 0.941345 0.017756
VERTEX_SE3:QUAT 32 -0.666844 -0.939950 1.381907 -0.091632 0.341482 0.930562 -0.095117
VERTEX_SE3:QUAT 33 -0.819240 -0.964797 1.324682 -0.195311 0.306116 0.871834 -0.328714
VERTEX_SE3:QUAT 34 -1.038169 -1.040930 1.278803 0.251562 -0.262940 -0.771480 0.521917
VERTEX_SE3:QUAT 35 -1.073511 -1.071354 1.282044 0.265907 -0.202693 -0.648766 0.683603
VERTEX_SE3:QUAT 36 -0.957552 -1.211772 1.328398 0.199608 -0.211251 -0.718667 0.631702
VERTEX_SE3:QUAT 37 -0.841041 -1.379648 1.360147 0.133796 -0.231686 -0.808233 0.524576
VERTEX_SE3:QUAT 38 -0.818449 -1.682181 1.396162 -0.120375 0.227832 0.861784 -0.436955
VERTEX_SE3:QUAT 39 -0.901957 -1.949372 1.404234 -0.058569 0.231675 0.929653 -0.280430
VERTEX_SE3:QUAT 40 -1.154948 -2.135177 1.433617 -0.018932 0.222696 0.964881 -0.138030
VERTEX_SE3:QUAT 41 -1.473692 -2.081204 1.502514 0.021281 0.215303 0.975508 0.039701
VERTEX_SE3:QUAT 42 -1.717822 -1.888983 1.485601 0.044179 0.248983 0.939892 0.229474
VERTEX_SE3:QUAT 43 -1.828656 -1.648299 1.476387 0.095480 0.263960 0.896385 0.343077
VERTEX_SE3:QUAT 44 -2.021514 -1.487557 1.457492 0.129080 0.265069 0.857533 0.421561
VERTEX_SE3:QUAT 45 -2.196759 -1.290168 1.427191 0.163575 0.260537 0.820081 0.482526
VERTEX_SE3:QUAT 46 -2.460700 -1.132343 1.431570 0.146094 0.234693 0.834676 0.476332
VERTEX_SE3:QUAT 47 -2.592310 -0.830643 1.453456 0.156433 0.228519 0.829400 0.485185
VERTEX_SE3:QUAT 48 -2.775915 -0.666094 1.444167 0.175478 0.217039 0.789058 0.547257
VERTEX_SE3:QUAT 49 -2.910952 -0.434886 1.442387 0.219379 0.184340 0.722539 0.629150
VERTEX_SE3:QUAT 50 -3.058592 -0.247149 1.467589 0.224875 0.164449 0.666566 0.691432
VERTEX_SE3:QUAT 51 -3.157283 -0.022298 1.484235 0.224524 0.139759 0.636045 0.724916
VERTEX_SE3:QUAT 52 -3.217215 0.166241 1.513951 0.236083 0.092516 0.552360 0.794106
VERTEX_SE3:QUAT 53 -3.278665 0.410877 1.491243 0.252203 0.086549 0.522944 0.809588
VERTEX_SE3:QUAT 54 -3.222443 0.613594 1.504034 0.274785 0.066326 0.476295 0.832609
VERTEX_SE3:QUAT 55 -3.158677 0.972411 1.549903 0.216502 0.027078 0.445833 0.868117
VERTEX_SE3:QUAT 56 -2.969926 1.231718 1.547295 0.229878 0.036182 0.368728 0.899937
VERTEX_SE3:QUAT 57 -2.882047 1.517073 1.547289 0.234809 0.028721 0.312946 0.919839
VERTEX_SE3:QUAT 58 -2.704277 1.673668 1.551697 0.255536 0.000817 0.251221 0.933589
VERTEX_SE3:QUAT 59 -2.432626 1.769823 1.531066 0.273722 -0.037800 0.086670 0.957150
VERTEX_SE3:QUAT 60 -2.142562 1.742403 1.523721 0.286211 -0.091685 -0.034564 0.953143
VERTEX_SE3:QUAT 61 -1.885826 1.648340 1.515176 0.299263 -0.115179 -0.120717 0.939469
VERTEX_SE3:QUAT 62 -1.704136 1.491055 1.494772 0.291772 -0.146928 -0.204706 0.922701
VERTEX_SE3:QUAT 63 -1.518070 1.262186 1.458277 0.303788 -0.143287 -0.199742 0.920481
[... cut ...]
EDGE_SE3:QUAT 81 82 -0.227970 -0.062732 0.035505 -0.002285 0.023254 0.051322 0.998409 1805.878660 0.000000 0.000000 0.000000 0.000000 0.000000 1805.878660 0.000000 0.000000 0.000000 0.000000 1805.878660 0.000000 0.000000 0.000000 198.641995 0.000000 0.000000 198.641995 0.000000 198.641995
EDGE_SE3:QUAT 82 83 -0.187883 -0.107268 0.054025 0.002308 0.056475 0.075146 0.995569 1050.978726 0.000000 0.000000 0.000000 0.000000 0.000000 1050.978726 0.000000 0.000000 0.000000 0.000000 1050.978726 0.000000 0.000000 0.000000 219.606998 0.000000 0.000000 219.606998 0.000000 219.606998
```