introlab/rtabmap_ros #281: Could we change odometry parameters on runtime (e.g., enabling Odometry panel in rtabmapviz)?

matlabbe, 2018-10-12:


mattiasmar, 2024-07-18:
Is this feature enabled? Can one update rtabmap parameters dynamically using ros2 messages now? If yes, how?

matlabbe, 2024-07-21:
For rtabmap node, you can update parameters directly with `set_parameters` ros service. Under the hood, the RTAB-Map's parameters will be handled:
https://github.com/introlab/rtabmap_ros/blob/57dd787a589ab40c75b43f1ed72084858440f107/rtabmap_slam/src/CoreWrapper.cpp#L855-L911

For odometry nodes, there is also a `set_parameters` that exists like all ros2 nodes, but updating the parameters after the node is initialized won't change anything. Note that even in standalone application, we don't support changing odometry parameters after it is initialized. The Odometry object has to be recreated as it cannot be reconfigured after being constructed:
https://github.com/introlab/rtabmap/blob/master/corelib/include/rtabmap/core/Odometry.h

Under ROS2, an option 3 could be to keep in cache the parameters set on initialization, and when `set_parameters` is called, we re-initialized from zero the node by re-creating a new Odometry:
https://github.com/introlab/rtabmap_ros/blob/57dd787a589ab40c75b43f1ed72084858440f107/rtabmap_odom/src/OdometryROS.cpp#L348

Note however that it is exactly like re-launching the node with different parameters.