introlab/rtabmap_ros #284: `rgbd_odometry`  node crash

BrunoXL, 2018-11-08:
On running `rgbd_odometry` I get the following error below:

    terminate called after throwing an instance of 'cv_bridge::Exception'
    what():  Image is wrongly formed: height * step != size  or  480 * 1920 != 110592

cv_bridge: 1.12.8
opencv: 3.3.1
Machine: Ubuntu 16.04

matlabbe, 2018-11-08:
That cv_bridge assert is about a wrong image format received when converting [sensor_msgs/Image](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/Image.html) to OpenCV image. What is the format of the input images you are streaming? Make sure all fields are correctly set in the topic:
```
# This message contains an uncompressed image
# (0, 0) is at top-left corner of image
#

Header header        # Header timestamp should be acquisition time of image
                     # Header frame_id should be optical frame of camera
                     # origin of frame should be optical center of camera
                     # +x should point to the right in the image
                     # +y should point down in the image
                     # +z should point into to plane of the image
                     # If the frame_id here and the frame_id of the CameraInfo
                     # message associated with the image conflict
                     # the behavior is undefined

uint32 height         # image height, that is, number of rows
uint32 width          # image width, that is, number of columns

# The legal values for encoding are in file src/image_encodings.cpp
# If you want to standardize a new string format, join
# ros-users@lists.sourceforge.net and send an email proposing a new encoding.

string encoding       # Encoding of pixels -- channel meaning, ordering, size
                      # taken from the list of strings in include/sensor_msgs/image_encodings.h

uint8 is_bigendian    # is this data bigendian?
uint32 step           # Full row length in bytes
uint8[] data          # actual matrix data, size is (step * rows)
```

BrunoXL, 2018-11-08:
I'm publishing two topics: one for rgb image in `bgr8`  and another for the depth image in `32FC1` encoding

matlabbe, 2018-11-08:
Can you record a small rosbag of **rgb**, **depth** and **camera_info** topics?

BrunoXL, 2018-11-09:
Here are the rosbag outputs: [depth](https://files.fm/f/yzy6ahht), [rgb](https://files.fm/f/czsztxp9), [camera_info](https://files.fm/f/a9ny78nk)

matlabbe, 2018-11-10:
I reproduced the error with `rqt_image_view`:
```
Depth:
ImageView.callback_image() while trying to convert image from '32FC1' to 'rgb8' an exception was thrown (Image is wrongly formed: height * step != size  or  480 * 2560 != 147456)
RGB:
ImageView.callback_image() while trying to convert image from 'bgr8' to 'rgb8' an exception was thrown (Image is wrongly formed: height * step != size  or  480 * 1920 != 110592)
```
The header seems okay, but the **data** field in the topics is too small. It should be at least 480 * 1920 = 921600 bytes for RGB and 480 * 2560=1228800 bytes for depth. You would need to fix that in your node publishing the images.

Another note, when recording rosbag, you can record multiple topics in the same rosbag, it is a lot more convenient to replay them:
```
$ rosbag record /camera/depth_registered/image_raw /camera/rgb/camera_info /camera/rgb/image_rect_color
$ rosbag play mybag.bag
```

closed 2018-11-12 by BrunoXL