introlab/rtabmap_ros #288: Failed to load nodelet [/obstacles_detection] of type [rtabmap_ros/obstacles_detection] even after refreshing the cache: Failed to load library /opt/ros/kinetic/lib//librtabmap_ros.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = librtabmap_core.so.0.11: cannot open shared object file: No such file or directory)

B-06Rxs, 2018-12-04:
hi,
I have this problem after install some more dependencies in my turtle-bot.

**LOGS**
... logging to /home/ubuntu/.ros/log/50b67944-f7bd-11e8-ada0-00044b66a1f0/roslaunch-tegra-ubuntu-9677.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.19.113.126:58699/

SUMMARY
========

CLEAR PARAMETERS
 * /ekf_se_odom/

PARAMETERS
 * /amcl/base_frame_id: base_footprint
 * /amcl/gui_publish_rate: 10.0
 * /amcl/initial_pose_a: 0
 * /amcl/initial_pose_x: 0
 * /amcl/initial_pose_y: 0
 * /amcl/kld_err: 0.01
 * /amcl/kld_z: 0.99
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 60
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.15
 * /amcl/laser_z_hit: 0.95
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.05
 * /amcl/laser_z_short: 0.1
 * /amcl/max_particles: 500
 * /amcl/min_particles: 100
 * /amcl/odom_alpha1: 1.5
 * /amcl/odom_alpha2: 0.5
 * /amcl/odom_alpha3: 0.5
 * /amcl/odom_alpha4: 1.5
 * /amcl/odom_alpha5: 0.5
 * /amcl/odom_frame_id: /odom
 * /amcl/odom_model_type: diff
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 1
 * /amcl/tf_broadcast: True
 * /amcl/transform_tolerance: 1.0
 * /amcl/update_min_a: 0.1
 * /amcl/update_min_d: 0.1
 * /amcl/use_map_topic: False
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/driver/auto_exposure: True
 * /camera/driver/auto_white_balance: True
 * /camera/driver/bootorder: 0
 * /camera/driver/color_depth_synchronization: False
 * /camera/driver/depth_camera_info_url: 
 * /camera/driver/depth_frame_id: /camera_depth_opt...
 * /camera/driver/depth_registration: False
 * /camera/driver/device_id: #1
 * /camera/driver/devnums: 1
 * /camera/driver/rgb_camera_info_url: 
 * /camera/driver/rgb_frame_id: /camera_rgb_optic...
 * /ekf_se_odom/base_link_frame: base_link
 * /ekf_se_odom/debug: False
 * /ekf_se_odom/frequency: 30
 * /ekf_se_odom/imu0: mobile_base/senso...
 * /ekf_se_odom/imu0_config: [False, False, Fa...
 * /ekf_se_odom/imu0_differential: False
[... 258 lines cut ...]
[INFO] [1543925317.850874]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1543925317.857808]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[ INFO] [1543925318.541476739]: Done initializing likelihood field model.
[INFO] [1543925319.243800]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1543925319.253677]: Note: subscribe buffer size is 1024 bytes
[INFO] [1543925319.255547]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1543925319.278525]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1543925319.295845]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1543925319.308380]: Setup subscriber on reset [std_msgs/Empty]
[ WARN] [1543925320.830039563]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100798 timeout was 0.1.
[INFO] [1543925320.865141]: Setup TF on Odometry [odom]
[INFO] [1543925320.872042]: Setup TF on IMU [imu_link]
[INFO] [1543925320.877228]: Setup TF on MagneticField [mag_link]
[INFO] [1543925320.881288]: Setup TF on JointState [base_link]
[INFO] [1543925320.890176]: --------------------------
[INFO] [1543925320.894398]: Connected to OpenCR board!
[INFO] [1543925320.898051]: This core(v1.2.3) is compatible with TB3 Waffle or Waffle Pi
[INFO] [1543925320.901342]: --------------------------
[INFO] [1543925320.904578]: Start Calibration of Gyro
[INFO] [1543925323.424689]: Calibration End
**[ INFO] [1543925324.800743567]: Using plugin "static_layer"
terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
  what():  Failed to load library /opt/ros/kinetic/lib//librtabmap_ros.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = librtabmap_core.so.0.11: cannot open shared object file: No such file or directory)**
[move_base-25] process has died [pid 9967, exit code -6, cmd /opt/ros/kinetic/lib/move_base/move_base scan:=/laser_scan map:=/map __name:=move_base __log:=/home/ubuntu/.ros/log/50b67944-f7bd-11e8-ada0-00044b66a1f0/move_base-25.log].
log file: /home/ubuntu/.ros/log/50b67944-f7bd-11e8-ada0-00044b66a1f0/move_base-25*.log

B-06Rxs, 2018-12-04:
I could not find the librtabmap_core.so.0.11 under lib at all. what I found is librtabmap_core.so.0.17 instead using the locate command. hopefully adding this information might help.

matlabbe, 2018-12-05:
Yes it is strange it is searching for 0.11. The default binaries are on 0.17. Can you do `$ ldd /opt/ros/kinetic/lib//librtabmap_ros.so`?

Try reinstall ros-kinetic-rtabmap and ros-kinetic-rtabmap-ros:
```bash
$ sudo apt remove ros-kinetic-rtabmap
$ sudo apt install ros-kinetic-rtabmap-ros
```

B-06Rxs, 2018-12-05:
thank you, it worked now after reinstall the ros-kinetic-rtabmap.

abdulahad01, 2022-10-13:
I am facing the same issues in ros noetic, I tried reinstalling rtab.
I have raised a separate issue here, please look into it [link to issue](https://github.com/introlab/rtabmap_ros/issues/823)