introlab/rtabmap_ros #289: Client [/rtabmap/rtabmapviz] wants topic /rtabmap/info

zoumaguanxin, 2018-12-21:
roslaunch rtabmap_ros rgbd_mapping.launch

ERROR] [1545374306.163100566]: Client [/rtabmap/rtabmapviz] wants topic /rtabmap/info to have datatype/md5sum [rtabmap_ros/Info/a3319209ed98ef442877eb716ab8963b], but our version has [rtabmap_ros/Info/f9cc2e215f6cae05161c74d102ffc592]. Dropping connection.

My rtabmap standalone and rtabmap_ros are both latest version. 
 
WITH_LOAM=OFF
WITH_GTSAM=OFF

matlabbe, 2018-12-21:
Is it on the same computer or rtabmapviz is launched on a remote computer? That error is just that the rtabmapviz version launched is not the same than the rtabmap version of the node. If it is on the same computer, make sure you uninstalled the binaries version of rtabmap if you built rtabmap from source.

zoumaguanxin, 2018-12-24:
Hi, 
it works. There are too many history versions  of RTABMap installed, it is very difficult to purge them. So I used the following command to assure that unnecessary versions were cleared.
`locate rtabmap`
`locate librtabmap`
 `sudo updatedb` 
Especially, I don't like  using `catkin clean` so that some `librtabmap_ros.so` in other workspaces are easily forgotten to deleted manually.
Thanks very much.

closed 2018-12-24 by zoumaguanxin