introlab/rtabmap_ros #291: RTABMAP make error////////usr/bin/ld: cannot find -lBoost::timer

ndinarvand, 2019-02-15:
Hello everyone,
I want to install rtabmap from scratch but in make step I've got bellow error:
[  5%] Built target rtabmap_utilite
[  6%] Built target res_tool
[  6%] Linking CXX shared library ../../../bin/librtabmap_core.so
/usr/bin/ld: cannot find -lBoost::timer
collect2: error: ld returned 1 exit status
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:3045: recipe for target '../bin/librtabmap_core.so.0.18.3' failed
make[2]: *** [../bin/librtabmap_core.so.0.18.3] Error 1
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

I'm using ubuntu 16.04 with installed ROS kinetic and opencv 3.3

THanks

matlabbe, 2019-02-15:
Can you show outputs of `cmake ..` and `make VERBOSE=1`?

matlabbe, 2019-02-15:
May be related to https://github.com/introlab/rtabmap/pull/357

ndinarvand, 2019-02-28:
> Can you show outputs of `cmake ..` and `make VERBOSE=1`?

Sorry for my late reply...
*******`cmake .. `output:
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Found OpenMP
-- Found OpenCV: /opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-6.2;/usr/include/jsoncpp;/usr/include/x86_64-linux-gnu;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/include/hdf5/openmpi;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/python2.7;/usr/include/libxml2;/usr/include/tcl
-- Found ZLIB: /usr/include
-- VTK_RENDERING_BACKEND=OpenGL
-- Found Freenect: /usr/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- Found octomap 1.8.1: /opt/ros/kinetic/include
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.18.3
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
--   FLANN_KDTREE_MEM_OPT = OFF
--   PCL_DEFINITIONS = -DEIGEN_USE_NEW_STDVECTOR;-DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET;-DFLANN_STATIC;-Dqh_QHpointer
-- Optional dependencies ('*' affects some default parameters) :
--  *With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With external SQLite3     = YES (License: Public Domain)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--  *With g2o                  = NO (WITH_G2O=OFF)
--  *With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (cvsba not found)
--  *With libpointmatcher      = NO (libpointmatcher not found)
--   With loam_velodyne        = NO (loam_velodyne not found)
--   With ZED                  = NO (ZED sdk not found)
[... 34 lines cut ...]
make[2]: Leaving directory '/home/nvd/rtabmap/build'
make -f utilite/resource_generator/CMakeFiles/res_tool.dir/build.make utilite/resource_generator/CMakeFiles/res_tool.dir/build
make[2]: Entering directory '/home/nvd/rtabmap/build'
make[2]: Nothing to be done for 'utilite/resource_generator/CMakeFiles/res_tool.dir/build'.
make[2]: Leaving directory '/home/nvd/rtabmap/build'
[  6%] Built target res_tool
make -f corelib/src/CMakeFiles/rtabmap_core.dir/build.make corelib/src/CMakeFiles/rtabmap_core.dir/depend
make[2]: Entering directory '/home/nvd/rtabmap/build'
cd /home/nvd/rtabmap/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/nvd/rtabmap /home/nvd/rtabmap/corelib/src /home/nvd/rtabmap/build /home/nvd/rtabmap/build/corelib/src /home/nvd/rtabmap/build/corelib/src/CMakeFiles/rtabmap_core.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/nvd/rtabmap/build'
make -f corelib/src/CMakeFiles/rtabmap_core.dir/build.make corelib/src/CMakeFiles/rtabmap_core.dir/build
make[2]: Entering directory '/home/nvd/rtabmap/build'
[  6%] Linking CXX shared library ../../../bin/librtabmap_core.so
cd /home/nvd/rtabmap/build/corelib/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/rtabmap_core.dir/link.txt --verbose=1
[... cut ...]
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:2869: recipe for target '../bin/librtabmap_core.so.0.18.3' failed
make[2]: *** [../bin/librtabmap_core.so.0.18.3] Error 1
make[2]: Leaving directory '/home/nvd/rtabmap/build'
CMakeFiles/Makefile2:287: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make[1]: Leaving directory '/home/nvd/rtabmap/build'
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

matlabbe, 2019-02-28:
`-lBoost::timer` may come from the installed `/usr/local/lib/cmake/GTSAM/GTSAM-exports-release.cmake` file. Try to modify this file to replace `-lBoost::timer` by its full path (e.g., 
`/usr/lib/x86_64-linux-gnu/libboost_timer.so`) like the other boost libraries. See this [post](https://github.com/introlab/rtabmap/pull/357#issuecomment-461515840). 

cheers,
Mathieu

shoelick, 2019-03-25:
Just wanted to jump in and say the substitution of `Boost::timer` to `/usr/lib/x86_64-linux-gnu/libboost_timer.so` solved this problem for me, but in the file `/usr/local/lib/cmake/GTSAM/GTSAM-exports.cmake` on line 72.

```
INTERFACE_LINK_LIBRARIES "Boost::serialization;Boost::system;Boost::filesystem;Boost::thread;Boost::date_time;Boost::regex;Boost::timer;Boost::chrono;tbb;tbbmalloc;metis" 
```
to 
```  
INTERFACE_LINK_LIBRARIES "Boost::serialization;Boost::system;Boost::filesystem;Boost::thread;Boost::date_time;Boost::regex;/usr/lib/x86_64-linux-gnu/libboost_timer.so;Boost::chrono;tbb;tbbmalloc;metis" 
```

closed 2019-04-04 by ndinarvand

zengletian1491, 2019-12-25:
> Just wanted to jump in and say the substitution of `Boost::timer` to `/usr/lib/x86_64-linux-gnu/libboost_timer.so` solved this problem for me, but in the file [...]
The substitution of Boost::timer to /usr/lib/x86_64-linux-gnu/libboost_timer.so solved this problem for me, but in the file
/usr/local/lib/cmake/GTSAM/GTSAM-exports-release.cmake on line 41

TINY-KE, 2021-09-13:
Add  " find_package(Boost REQUIRED COMPONENTS timer) "  into the CmakeLists.txt.
I solve this problem by this way.

arjunskumar, 2022-02-18:
@TINY-KE Thanks, this worked