introlab/rtabmap_ros #292: unmet dependencies for ros-melodic-rtabmap on Ubuntu 18.04

swiss-knight, 2019-02-21:
I'm trying to follow this tutorial : http://wiki.ros.org/rtabmap_ros/Tutorials/Tango%20ROS%20Streamer with a Phab 2 Pro from Lenovo and a Ubuntu 18.04 64bit based laptop which hosts the ROS master.
The Tango light of the App is still red.

I guessed that I was missing the "rtabmap_ros" package (i.e. `ros-melodic-rtabmap`).

So I tried to install it but it fails;

```
$ sudo apt-get install ros-melodic-rtabmap
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-melodic-rtabmap : Depends: libpcl-dev but it is not going to be installed
                       Depends: libvtk6-qt-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

```

I already have `libvtk7-dev`. If I try to installed the dependencies from the dependencies from the dependencies from the (...), I end up on this:

```
$ sudo apt-get install libvtk6-jni
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libarmadillo-dev libarpack2-dev libdap-dev libdapserver7v5 libepsilon-dev
  libfreexl-dev libfyba-dev libgdal-dev libgeotiff-dev libgif-dev libgl2ps-dev
  libglew-dev libhdf5-mpi-dev libhdf5-openmpi-dev libjson-c-dev libjsoncpp-dev
  libkml-dev libkmlconvenience1 libkmlregionator1 libkmlxsd1 libminizip-dev
  libnetcdf-cxx-legacy-dev libogdi3.2-dev libogg-dev libpoppler-dev
  libpoppler-private-dev libproj-dev libsuperlu-dev libtheora-dev
  liburiparser-dev libwebp-dev libxerces-c-dev libxml2-dev python3-autobahn
  python3-cbor python3-lz4 python3-mpi4py python3-qrcode python3-snappy
  python3-trie python3-txaio python3-u-msgpack python3-ubjson python3-vtk7
  python3-wsaccel qttools5-private-dev tcl-dev tcl-vtk7 tcl8.6-dev tk-dev
  tk8.6-dev
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libvtk6.3-qt
The following packages will be REMOVED:
  libvtk7-dev libvtk7-java libvtk7-jni libvtk7-qt-dev
The following NEW packages will be installed:
  libvtk6-jni libvtk6.3-qt
0 upgraded, 2 newly installed, 4 to remove and 0 not upgraded.
Need to get 4'592 kB of archives.
After this operation, 33.0 MB disk space will be freed.
Do you want to continue? [Y/n] n
Abort.
```

I don't want to mess up with these libraries.
Is there an easy way to fix that or the only way is to proceed with this removal?

Computer information;
Ubuntu 18.04 4.15.0-45-generic x86_64 GNU/Linux
/rosdistro: melodic
/rosversion: 1.14.3

Phone: Android 6.0.1 (Lenovo Phab2 pro) with both rtabmap and Tango Ros Streamer apps installed from the playstore.

matlabbe, 2019-02-23:
[libpcl-dev](https://packages.ubuntu.com/bionic/libpcl-dev) depends on vtk6. Do you really need vtk7? If I were you, I would uninstall it and install vtk6 instead. Otherwise, **the much longer way** is to rebuild from source [PCL](http://pointclouds.org/) so that the PCL links on vtk7 instead. If you do so, you would have to rebuild also from [source](https://github.com/introlab/rtabmap_ros#build-from-source) rtabmap and rtabmap_ros.

swiss-knight, 2019-03-04:
Okay, thanks.
It's working with VTK6 under the hood.