introlab/rtabmap_ros #293: Unable to build rtabmap_ros for trusty

vityafx, 2019-02-27:
I can't build rtabmap_ros for ubuntu trusty (from the `indigo-devel` branch). I have previously built `rtabmap` and installed it into the system. Whenever I try to build rtabmap_ros, I get [this (800 lines error)](https://pastebin.com/zZaXni2P):

```
/home/ubuntu/catkin_ws/src/rtabmap_ros/src/nodelets/rgbd_odometry.cpp:176:25:   required from here
/opt/ros/indigo/include/message_filters/synchronizer.h:358:14: error: invalid operands of types '<unresolved overloaded function type>' and 'int' to binary 'operator<'
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/nodelets/rgbd_odometry.cpp.o] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all] Error 2
make: *** [all] Error 2
```

matlabbe, 2019-02-27:
The indigo-devel branch and the master branch should build on Indigo. I cannot reproduce the error on my Indigo laptop (14.04). Make sure ROS is all updated:
```
$ sudo apt-get upgrade ros-indigo-message-filters
```

vityafx, 2019-02-28:
ROS is fully updated and rebooted, that did not help. Also, tried to build from the `master` branch and that did not help either.

matlabbe, 2019-02-28:
Look at the buildfarm: https://travis-ci.org/introlab/rtabmap_ros, it is on Trusty-Indigo. gcc version is `gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4`
It is maybe a compiler version problem.

vityafx, 2019-03-14:
Unfortunately, I was not able to continue to work on this project, so I close this issue. Thank you for your response.

closed 2019-03-14 by vityafx