introlab/rtabmap_ros #295: Tuning visual odometry parameters from rtabmap.launch - param placement?

luhcforgh, 2019-03-11:
Hi!

RTAB-MAP seems like a great software so far. We're trying to optimize performance for use with an _Intel RealSense D435i_ and _Aaeon Up Board_ by following the guide at: http://wiki.ros.org/rtabmap_ros/Tutorials/Advanced%20Parameter%20Tuning

Sections 3.1.1, 3.1.2, and 3.1.2 all seem very interesting - but our problem is that we can't figure out where to place the <param> tags in the rtabmap.launch file (/opt/ros/kinetic/share/rtabmap_ros/launch/rtabmap.launch). We can edit this file by running "sudo gedit". 

On the Wiki page, it is suggested to "use <param> tag": 

```
 For exmaple, you can put this under rgbd_odometry node
 <param name="Odom/Strategy" value="0"/>
```

Does this mean that we should add the line, right before the </node> line? This however places the parameter within the name space of /rtabmap/rgbd_odometry/Odom/Strategy, when running "rosparam list | grep "Odom/Str", while there is already a parameter called /rtabmap/Odom/Strategy. 

```
<!-- RGB-D Odometry -->
        <node unless="$(arg stereo)" pkg="rtabmap_ros" type="rgbd_odometry" name="rgbd_odometry" output="$(arg output)" args="$(arg rtabmap_args) $(arg odom_args)" launch-prefix="$(arg launch_prefix)">
          <remap from="rgb/image"       to="$(arg rgb_topic_relay)"/>
          <remap from="depth/image"     to="$(arg depth_topic_relay)"/>
          <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
          <remap     if="$(arg rgbd_sync)" from="rgbd_image" to="$(arg rgbd_topic)"/>
          <remap unless="$(arg rgbd_sync)" from="rgbd_image" to="$(arg rgbd_topic)_relay"/>
          <remap from="odom"            to="$(arg odom_topic)"/>
      
          <param name="frame_id"                    type="string" value="$(arg frame_id)"/>
          <param name="odom_frame_id"               type="string" value="$(arg vo_frame_id)"/>
          <param name="ground_truth_frame_id"       type="string" value="$(arg ground_truth_frame_id)"/>
          <param name="ground_truth_base_frame_id"  type="string" value="$(arg ground_truth_base_frame_id)"/>
          <param name="wait_for_transform_duration" type="double" value="$(arg wait_for_transform)"/>
          <param name="approx_sync"                 type="bool"   value="$(arg approx_sync)"/>
          <param name="config_path"                 type="string" value="$(arg cfg)"/>
          <param name="queue_size"                  type="int"    value="$(arg queue_size)"/>
          <param name="subscribe_rgbd"              type="bool"   value="$(arg subscribe_rgbd)"/>
          <param name="guess_frame_id"              type="string" value="$(arg odom_guess_frame_id)"/>
          <param name="guess_min_translation"       type="double" value="$(arg odom_guess_min_translation)"/>
          <param name="guess_min_rotation"          type="double" value="$(arg odom_guess_min_rotation)"/>

 **ADD LINE HERE????**

        </node>
```
I was able to successfully change a single parameter by specifying an argument when including the rtabmap.launch file, but I would like to change all three settings at once. 

```
    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
        <arg name="args" value="--delete_db_on_start"/>
        <arg name="compressed" value="false"/> <!--If you want to subscribe to compressed image topics--> 
        <arg name="rgb_topic" value="/camera/color/image_raw"/>
        <arg name="depth_topic" value="/camera/aligned_depth_to_color/image_raw"/>
        <arg name="camera_info_topic" value="/camera/color/camera_info"/>
        <arg name="depth_camera_info_topic" value="/camera/depth/camera_info"/>
        <arg name="rtabmapviz" value="false"/>
        <arg name="rviz" value="true"/>
**<arg name="rtabmap_args" default="--Odom/Strategy 1"/>** 	
    </include>
```

Thanks for the help! :)

luhcforgh, 2019-03-13:
Although not quite what we wanted, we found that we can supply multiple arguments to the rtabmap.launch file by doing this (note that Odom/Strategy2 is just a placeholder):

`    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
        <arg name="args" value="--delete_db_on_start"/>
        <arg name="compressed" value="false"/>  
        <arg name="rgb_topic" value="/camera/color/image_raw"/>
        <arg name="depth_topic" value="/camera/aligned_depth_to_color/image_raw"/>
        <arg name="camera_info_topic" value="/camera/color/camera_info"/>
        <arg name="depth_camera_info_topic" value="/camera/depth/camera_info"/>
        <arg name="rtabmapviz" value="false"/>
        <arg name="rviz" value="true"/>
**<arg name="rtabmap_args" default="--Odom/Strategy 1 --Odom/Strategy2 2"/>** 	
    </include>`

matlabbe, 2019-03-22:
When using rtabmap.launch, you can indeed use `args` (`rtabmap_args` for older version) argument to set parameters related to RTAB-Map library, like `--Odom/Strategy 1`. Well all the parameters shown by `$ rtabmap --params`.
```bash
$ roslaunch rtabmap_ros rtabmap.launch args:="--delete_db_on_start --Odom/Strategy 1"
```

Using param tag is more used when adding directly rtabmap's ndoes to a custom launch file (without using rtabmap.launch), then the parameter should be set inside the node definition:
```xml
<node pkg="rtabmap_ros" type="rgbd_odometry" name="rgbd_odometry">
   ...
   <param name="Odom/Stratregy" value="1"/>
</node>
```
or the following would also work too:
```xml
<node pkg="rtabmap_ros" type="rgbd_odometry" name="rgbd_odometry" args:="--Odom/Strategy 1">
   ...
</node>
```