introlab/rtabmap_ros #297: Wrong calibration?

abylikhsanov, 2019-03-14:
Whenever I start the rtabmap mapping using the stereo camera I get this error:

```
[ INFO] [1552591533.217995979]: rtabmap 0.19.0 started...
[ERROR] (2019-03-14 20:25:33.621) Memory.cpp:4017::createSignature() Camera calibration not valid, calibrate your camera!
[ERROR] (2019-03-14 20:25:33.621) Memory.cpp:786::update() Failed to create a signature...
```

I wonder what could it be, as my calibration is more or less correct.

This is the camera_info message I have:

```
header: 
  seq: 39
  stamp: 
    secs: 1552591459
    nsecs: 648231030
  frame_id: "camera_fisheye1_optical_frame"
height: 800
width: 848
distortion_model: "plumb_bob"
D: [1.0, 0.0, 0.0, 0.0, 0.0]
K: [287.759288, 0.0, 421.361583, 0.0, 286.221648, 397.829615, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [287.759288, 0.0, 421.361583, 0.0, 0.0, 286.221648, 397.829615, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---

```

closed 2019-03-15 by abylikhsanov

matlabbe, 2019-03-22:
Were you using a single camera? RGB-D or stereo cameras are required with rtabmap.

bahartman, 2019-06-25:
I am having a similar issue. My camera_info looks like this:

```
header: 
  seq: 4
  stamp: 
    secs: 1561487248
    nsecs: 380853891
  frame_id: "camera_link"
height: 1080
width: 1920
distortion_model: "plum_bob"
D: [-0.000591, 0.000519, 1e-06, -3e-05, 0.0]
K: [959.661112, 0.0, 959.385435, 0.0, 959.693478, 539.53336, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [959.779968, 0.0, 959.290331, 0.0, 0.0, 959.867798, 539.535675, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
```

I am using both an RGB and a Depth camera (this is the camera info for the RGB camera).
How were you able to resolve this issue (i'm guessing it was resolved since you closed the topic)?

matlabbe, 2019-06-25:
This [error](https://github.com/introlab/rtabmap/blob/9d8fd364e36cb9f51e6895c60d6fa1a99cb7fce6/corelib/src/Memory.cpp#L4011-L4018) happens if:

- the depth is empty,
- the calibration is wrong (seems ok here) or
- the pose received is null (if you have also the error: `RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided.`).

Can you show the depth image? (`rqt_image_view`)

mirellameelo, 2020-10-14:
@matlabbe I'm having this error. From your previous comment, 2 and 3 I think is fine, but I could not understand 1, regarding the depth since I'm using a stereo camera. Could you please, please take a look at my publisher's? Thaanks in advance!

### /right/camera_info 
```
---
header:
  stamp:
    sec: 1602715218
    nanosec: 899329939
  frame_id: camera_link
height: 480
width: 640
distortion_model: plumb_bob
d: [0.0, 0.0, 0.0, 0.0, 0.0]
k: [772.868570835972, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 6.94825075394135e-310, 2.5e-323]
r: [0.9985404059825475, -3.103795276460111e-05, 0.004420071389006859, 0.0003203935697372317, 0.9985087763742083, 5.0e-324, 0.0, 4.6881566600091e-310, 4.6881566600091e-310]
p: [778.572438721727, 0.0, 0.0, 0.0, 1.0, 0.0, 2.0e-323, 4.67890871e-315, 0.09897682798532774, 4.7993333914e-314, 0.0, 0.0]
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false
---
```

### /left/camera_info
```
---
header:
  stamp:
    sec: 1602715249
    nanosec: 467410351
  frame_id: camera_link
height: 480
width: 640
distortion_model: plumb_bob
d: [0.0, 0.0, 0.0, 0.0, 0.0]
k: [772.572438721727, 0.0, 0.0, 0.0, 1.0, 6.94818012665477e-310, 6.9482508620052e-310, 0.0, 0.0]
r: [0.9985404059825475, -3.103795276460111e-05, 0.004420071389006859, 0.0003203935697372317, 0.9985087763742083, 0.0, 0.0, 0.0, 0.0]
p: [778.572438721727, 0.0, 0.0, 0.0, 1.0, 0.0, 2.673714697234e-312, 2.716154613063e-312, 6.94817349624005e-310, 6.94817349624005e-310, 7.0e-323, 7.0e-323]
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false
---
```

### /right/image_rect
```
---
header:
  stamp:
    sec: 1602715328
    nanosec: 898418742
  frame_id: camera_link
height: 480
width: 640
encoding: bgr8
is_bigendian: 0
step: 1920
data: [23, 32, 54, 30, 41, 64, 38, 51, 75, 45, 59, 86, 51, 67, 96, 59, 77, 108, 63, 83, 115, 62, 83, 116, 60, 82, 116, 63, 86, 120, 64, 88, 122, 64, 90, 124, 64, 90, 124, 64, 91, 124, 65, 92, 125, 65, 93, 125, 65, 93, 126, 64, 92, 126, 63, 91, 126, 63, 91, 126, 61, 90, 126, 60, 89, 125, 60, 89, 125, 60, 89, 125, 60, 88, 125, 61, 90, 126, 62, 91, 128, 62, 92, 128, 61, 91, 127, 60, 89, 126, 61, 90, 126, 63, 91, 126, 63, 91, 126, 62, 90, 126, 62, 90, 125, 62, 90, 126, 63, 92, 127, 64, 93, 129, 64, 93, 129, 63, 92, 127, 63, 92, 127, 64, 93, 128, 65, 93, '...']
---
```

### /left/image_rect
```
---
header:
  stamp:
    sec: 1602715292
    nanosec: 460720176
  frame_id: camera_link
height: 480
width: 640
encoding: bgr8
is_bigendian: 0
step: 1920
data: [35, 12, 17, 50, 26, 32, 51, 26, 33, 49, 24, 31, 45, 20, 26, 32, 7, 14, 34, 9, 17, 50, 25, 32, 50, 25, 32, 49, 24, 31, 45, 21, 28, 37, 13, 19, 34, 11, 16, 44, 19, 26, 48, 23, 30, 53, 27, 34, 53, 27, 34, 38, 12, 19, 29, 3, 10, 34, 8, 15, 47, 21, 28, 52, 26, 33, 45, 19, 25, 35, 10, 17, 31, 6, 13, 32, 8, 14, 46, 21, 27, 52, 27, 33, 48, 23, 29, 42, 18, 24, 43, 19, 25, 48, 25, 31, 50, 26, 32, 44, 20, 26, 33, 9, 15, 40, 16, 22, 34, 11, 16, 24, 1, 6, 32, 8, 14, 44, 19, 27, 54, 29, 37, 47, 24, 31, 46, 23, '...']
---
```

### /odom
```
---
header:
  stamp:
    sec: 1602715359
    nanosec: 609015909
  frame_id: camera_link
child_frame_id: base_link
pose:
  pose:
    position:
      x: -2.9503552343425676
      y: 0.0
      z: 1.3238682980507146
    orientation:
      x: 0.5111583718471847
      y: 0.6355218643324685
      z: -0.4639515797989502
      w: 0.34580053563865004
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist:
  twist:
    linear:
      x: 0.0
      y: 0.0
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: 0.0
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
```

mhmd-mst, 2024-03-13:
Hello,
I have the following errors:
`[ERROR] (2024-03-13 12:31:51.840) util3d.cpp:1185::cloudsRGBFromSensorData() Camera model 0 is invalid
[ERROR] (2024-03-13 12:31:51.870) RegistrationVis.cpp:1566::computeTransformationImpl() Calibrated camera required. Id=22 Models=1 StereoModels=0 weight=0
[ WARN] (2024-03-13 12:31:51.871) Rtabmap.cpp:2944::process() Rejected loop closure 6 -> 22: `
I have the following message format for camera_info:
`header: 
  seq: 116381
  stamp: 
    secs: 1707840038
    nsecs: 726655006
  frame_id: "camera"
height: 720
width: 1280
distortion_model: "plumb_bob"
D: []
K: [923.0411376953125, 0.0, 350.8592529296875, 0.0, 923.5192260742188, 656.9656982421875, 0.0, 0.0, 1.0]
R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
P: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False`
Using data recorded from realsense D415 RGB-D camera, what is it a calibration problem?

matlabbe, 2024-03-16:
Rtabmap expects P to be set. In your case, P would look like this:
```
P: [923.0411376953125, 0.0, 350.8592529296875, 0.0, 
    923.5192260742188, 656.9656982421875, 0.0, 0.0, 
    0.0, 0.0, 1.0, 0.0]
```
Normally the realsense2 driver should be already providing R and P in the camera_info.

ADITYA97-CODER, 2024-06-13:
I am having the same error. Is this issue resolved?

matlabbe, 2024-06-16:
> I am having the same error. Is this issue resolved?

With realsense camera? What is your system? ROS1/ROS2? A quick check on my side (ros2 humble), and R is identity and P is set like I said in my previous post. Here an example for the color camera_info (D435i camera):
```
ros2 launch realsense2_camera rs_launch.py
ros2 topic echo /camera/color/camera_info
header:
  stamp:
    sec: 1718500617
    nanosec: 444867328
  frame_id: camera_color_optical_frame
height: 720
width: 1280
distortion_model: plumb_bob
d:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
k:
- 923.28076171875
- 0.0
- 636.6947631835938
- 0.0
- 923.6041870117188
- 376.4419860839844
- 0.0
- 0.0
- 1.0
r:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
p:
- 923.28076171875
- 0.0
- 636.6947631835938
- 0.0
- 0.0
- 923.6041870117188
- 376.4419860839844
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false
---

```

ADITYA97-CODER, 2024-06-16:
I fixed it.it was camera info error. I subscribed to wrong topic . My bad 😬.