introlab/rtabmap_ros #299: rtabmap update stops in between

aaravrav142, 2019-03-19:
Hi, 

I have recorded a small dataset using `data_throttle.launch` on a robot equipped with Lidar and Asus Xtion RGBD camera. The messages recorded are as follows when recording the bag. 
```
<launch>
  <node pkg="nodelet" type="nodelet" name="data_throttle" args="standalone rtabmap_ros/data_throttle">
    <param name="max_rate" type="double" value="20.0"/>
   
    <remap from="rgb/image_in"       to="camera/rgb/image_rect_color"/>
    <remap from="depth/image_in"     to="camera/depth_registered/image_raw"/>
    <remap from="rgb/camera_info_in" to="camera/depth_registered/camera_info"/>
    
    <remap from="rgb/image_out"       to="data_throttled_image"/>
    <remap from="depth/image_out"     to="data_throttled_image_depth"/>
    <remap from="rgb/camera_info_out" to="data_throttled_camera_info"/>
  </node>
</launch>
```

`rosbag record /data_throttled_image/compressed /data_throttled_image_depth/compressedDepth /data_throttled_camera_info /tf /odom /scan`

But when playing later offline the rosbag with `demo_robot_mapping.launch`, the bag file runs ok for few seconds and then the map update stops. I am getting few warning as below 
`[ WARN] [1552973823.283927610]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.`
and 
`[ INFO] [1552973884.893767963]: Creating 1 swatches
`

Can you describe if I am doing something wrong? 
Thanks

matlabbe, 2019-03-23:
Strange, make sure you started the bag with "--clock". You may have to modify demo_robot_mapping.launch too to remap the correct topics.

aaravrav142, 2019-03-25:
Hi, 

The --clock parameter is used when playing the bag.
Can you have a look at the[ bag file ](https://www.dropbox.com/s/yhit6x94pjezo7s/2019-03-18-21-16-22.bag?dl=0)if you can find something. 
Thanks

matlabbe, 2019-03-31:
I tried your bag with [demo_robot_mapping.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_robot_mapping.launch). Only changes I've made are replacing `/az3/base_controller/odom` by `/odom` and all `/jn0/base_scan` by `/scan`. I cannot reproduce your problem with:
```bash
$ roslaunch rtabmap_ros demo_robot_mapping.launch 
$ rosbag play --clock 2019-03-18-21-16-22.bag
```
[image]

Some issues:
1) Your wheel odometry seems to drift a lot in rotation. In this case the odometry is corrected by the laser scans, but in some cases scan matching cannot correct all odometry errors. Just launch your robot and open RVIZ, look at the TF when rotating on-place, the robot looks rotating more or less than in reality.

2) I noticed the TF of the RGB-D camera is wrong. There is a tilt error (ground not straight) and the ground plane doesn't match the base_footprint (ground cloud from the camera appears floating around 10-15 cm over the ground). We can easily see that when launching with RVIZ:
```bash
$ roslaunch rtabmap_ros demo_robot_mapping.launch rviz:=true
```
[image]

aaravrav142, 2019-04-01:
Hi 
Thanks for the reply. Strange on my PC the rosbag stops playing after a while. I will report back after rebuilding the rtabmap library or try on a different PC altogether to check if the problem persists. 
Thanks for your time.

closed 2019-04-15 by aaravrav142