introlab/rtabmap_ros #300: How to increase the frequency of topic /rtabmap/localization_pose with rtabmap_ros？

TsaiYC73, 2019-03-20:
Hi,
I have created a map with RTAB-map.
Now, I would like using sensor_fusion.launch to do indoor localization with EKF2 on Px4.
But the frequency of this topic is 1hz on average. (As the following figure)

[image]

That will lead to a bad result of EKF2(use visual position fusion).
How could I increase the frequency ?
Or if there's any ways to improve the performance of localize?

Thanks!

matlabbe, 2019-03-23:
You may use the visual odometry pose for px4 instead, it should be published faster (referring to /odom -> /base_link). There, it is the localization pose sent by rtabmap only when there is a localization or a loop closure (/map -> /base_link). You can increase the detection rate with **Rtabmap/DetectionRate** (default 1 Hz) parameter. However, if no localizations or loop closures occur, that pose is not published. I think you should use this message like a GPS received at low rate and not constantly published, not sure about the px4 EKF2 assumptions about the required input rate of the different topics.

cheers,
Mathieu

TsaiYC73, 2019-03-28:
@matlabbe Thank you for the suggest.
I am still working on it. I will keep you updated if there is anything new coming up.

matlabbe, 2023-02-08:
Update: from this [commit](https://github.com/introlab/rtabmap/commit/dcedd8bb6b9f6e9a8e884a271fcfff6671ee3cbf), localization_pose is always published in localization mode and covariance would increase when not re-localized, otherwise covariance would shrink when re-localized.