introlab/rtabmap_ros #301: Export saved 2D map, Edit the map and Import 2D map to RTAB-Map Database

vitorhirozawa, 2019-03-22:
Hi, 

The option in rtabmap-databaseViewer that allow to export saved 2D Map (File- > Export saved 2D map) and Import 2D map of a specific database is not available for me.  A watermarked option is shown that is not possible to select.
Is there an specific configuration to enable this option?

I figured out about this feature looking to this comment:
Using commits above, the grid map is automatically saved to database on exit. We can use DatabaseViewer -> File- > Export saved 2D map... to export the map in pgm image file. That file can be edited using a image editor to remove/add obstacles. Then with File -> Import 2D map, the map can be re-imported and overwriting the one saved in the database. When relaunching the robot in **localization** mode, the edited map will be used.

_Originally posted by @matlabbe in https://github.com/introlab/rtabmap_ros/issues/213#issuecomment-378775919_

The rtabmap version I am using is 0.17.6.
I use one rgbd camera only.
I can see the topic /proj_map (nav_msgs/OccupancyGrid) on rviz as a 2d map when executing mapping.
I do CTRL+C on mapping process and the database is saved. But it looks that the 2d map is not automatically saved in database.

I appreciate any answer about my problem.

matlabbe, 2019-03-23:
Can you share the database? With 0.17.6, the map should be saved on exit:
https://github.com/introlab/rtabmap_ros/blob/803be533bcc35457ca4332e4af4a4c573153de42/src/CoreWrapper.cpp#L678

It could be that there is no one subscribed to grid_map or proj_map topic when closing rtabmap, so that the grid wasn't updated and then cannot be saved.

vitorhirozawa, 2019-03-25:
Hi @matlabbe, 

When I was mapping with rtabmap, there was no one subscribed to grid_map or proj_map topic. That was the problem.

I did some tests. To save the 2d map, grid_map or proj_map topic should be subscribed at least one time. I did the subscription in rviz. Even if I subscribe and unsubscribe to this topic, it saves the 2d map in database. After pressing CTRL+C, the terminal prints the following message: "rtabmap: 2D occupancy grid map saved."
Then it is possible to export saved 2d map and import 2d map using rtabmap-databaseViewer.

Thank you very much matlabbe!

closed 2019-03-31 by matlabbe