introlab/rtabmap_ros #304: Add custom ground/obstacle detection system to rtabmap_ros

NeilNie, 2019-04-10:
Hello, 

After I experimented with rtabmap_ros, I noticed that it was incorrectly labeling certain positions on the grid as obstacles. This creates problems in the navigation stack. 

[image]

[image]

I developed my own ground detection system, that labels points in a point cloud as ground. I am wondering if it's possible to integrate my custom ground detection system to rtabmap, so that the `grid_map` and `costmap_2d` more accurately reflects the actual traversable space and obstacles.

If modifying the rtabmap is not possible. How would I tune the parameters to improve ground classification accuracy? 

Thank you very much for your time!

matlabbe, 2019-04-13:
Do you have a `rtabmap.db` to share? What is your approach? I could tell you if you can reproduce something similar by tuning some parameters. To see all grid related parameters:
```bash
$ rtabmap --params | grep Grid/
```

cheers,
Mathieu

NeilNie, 2019-04-18:
Hello @matlabbe, unfortunately, I don't have a `rtabmap.db`. I was trying to follow the `Stereo Outdoor Navigation` tutorial. I was unable to reproduce the good performance that was shown in the video. Your outdoor map looks very clean, free of random obstacles. However, as you saw, my map has lots of noise. 

My mapping launch file:

```
<launch>

    <arg name="zed_namespace"		    	default="zed" />

    <arg name="svo_file"              	default="" /> <!-- <arg name="svo_file" default="path/to/svo/file.svo"> -->

    <arg name="camera_model"         	default="1" /> <!-- 0=ZED, 1=ZEDM-->
    <arg name="serial_number"        	default="0" />
    <arg name="verbose"              	default="true" />

    <arg name="rgb_topic"               	default="rgb/image_rect_color" />
    <arg name="depth_topic"             	default="depth/depth_registered" />
    <arg name="camera_info_topic"       	default="rgb/camera_info" />
    <arg name="depth_camera_info_topic" 	default="depth/camera_info" />
    <arg name="camera_frame"              default="zed_camera_center" />

    <!-- RTAB-map Node-->
    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">

        <arg name="rtabmap_args"		value="--delete_db_on_start" />
        <arg name="rgb_topic"               	value="/$(arg zed_namespace)/$(arg rgb_topic)" />
        <arg name="depth_topic"             	value="/$(arg zed_namespace)/$(arg depth_topic)" />
        <arg name="camera_info_topic"       	value="/$(arg zed_namespace)/$(arg camera_info_topic)" />
        <arg name="depth_camera_info_topic" 	value="/$(arg zed_namespace)/$(arg depth_camera_info_topic)" />
        <arg name="frame_id"                  value="$(arg camera_frame)" />
        <arg name="approx_sync"               value="false" />
        <arg name="visual_odometry"           value="false" />
        <arg name="odom_topic"                value="/$(arg zed_namespace)/odom" />

        <!-- When sending goals on /rtabmap/goal topic, use actionlib to communicate with move_base -->
        <param name="use_action_for_goal" type="bool" value="true"/>
        <remap from="move_base"            to="/move_base"/>

        <!-- output -->
        <remap from="grid_map" to="/map"/>

        <!-- RTAB-Map's parameters: do "rosrun rtabmap rtabmap (double-dash)params" to see the list of available parameters. -->
        <param name="RGBD/ProximityBySpace"        type="string" value="true"/>   <!-- Local loop closure detection (using estimated position) with locations in WM -->
        <param name="RGBD/OptimizeFromGraphEnd"    type="string" value="false"/>  <!-- Set to false to generate map correction between /map and /odom -->
        <param name="Kp/MaxDepth"                  type="string" value="4.0"/>
        <param name="Reg/Strategy"                 type="string" value="0"/>      <!-- Loop closure transformation: 0=Visual, 1=ICP, 2=Visual+ICP -->
        <param name="Icp/CorrespondenceRatio"      type="string" value="0.3"/>
        <param name="Vis/MinInliers"               type="string" value="15"/>      <!-- 3D visual words minimum inliers to accept loop closure -->
        <param name="Vis/InlierDistance"           type="string" value="0.1"/>    <!-- 3D visual words correspondence distance -->
        <param name="RGBD/AngularUpdate"           type="string" value="0.1"/>    <!-- Update map only if the robot is moving -->
        <param name="RGBD/LinearUpdate"            type="string" value="0.1"/>    <!-- Update map only if the robot is moving -->
        <param name="RGBD/ProximityPathMaxNeighbors" type="string" value="0"/>
        <param name="Rtabmap/TimeThr"              type="string" value="0"/>
        <param name="Mem/RehearsalSimilarity"      type="string" value="0.30"/>
        <param name="Reg/Force3DoF"                type="string" value="true"/>
        <param name="GridGlobal/MinSize"           type="string" value="20"/>

    </include>

    <!-- Create point cloud for the local planner -->
    <node pkg="nodelet" type="nodelet" name="obstacles_detection" args="load rtabmap_ros/obstacles_detection stereo_nodelet">

        <remap from="cloud" to="/zed/point_cloud/cloud_registered"/>
<!--        <remap from="obstacles" to="/planner_cloud"/>-->

        <param name="frame_id" type="string" value="base_link"/>
        <param name="map_frame_id" type="string" value="map"/>
        <param name="min_cluster_size" type="int" value="10"/>
        <param name="max_obstacles_height" type="double" value="4.0"/>
    </node>
[... 70 lines cut ...]

  global_frame: map
  robot_base_frame: base_link
  update_frequency: 5.0
  publish_frequency: 5.0
  static_map: false
  rolling_window: true
  width: 10.0
  height: 12.0
  resolution: 0.05
  origin_x: 0.0
  origin_y: 0.0

  observation_sources: point_cloud_sensor

  # assuming receiving a cloud from rtabmap_ros/obstacles_detection node
  point_cloud_sensor: {
    sensor_frame: zed_camera_center,
    data_type: PointCloud2,
    topic: /point_cloud/ground_segmentation,
    expected_update_rate: 10.0,
    marking: false,
    clearing: true
  }
```

matlabbe, 2019-04-18:
What kind of environment you are navigating? Is it indoor or indoor? Based on the screenshots, it is difficult to see which points in the point clouds are segmented as obstacle. If the rtabmap.db cannot be shared, we would need to see at least the point cloud of the map to give tips.

cheers,
Mathieu

NeilNie, 2019-04-18:
I am navigating in an outdoor environment. I was able to recreate the `.db` file. [Here it is](https://drive.google.com/file/d/1IF0kZQ0h4pYJlyvDThaWoSRgZkahCFP9/view?usp=sharing). The frame rate is very low because of poor `rosbag` performance. Here are some screenshot of the point cloud. 

Thank you very much!

Best regards,

Neil

[image]
[image]
[image]

matlabbe, 2019-04-18:
Thx for the database, it is easier to debug (`rtabmap-databaseViewer` is used below). First the TF between base frame and camera frame is wrong, the ground appears tilted in the point of view of car. This can cause obstacle segmentation issues. To debug TF, you can show up the point cloud of the zed camera and set fixed frame in global option to base frame you are using, the cloud of the ground should match the RVIZ x-y plane. 
[image]

Here is the current result of segmentation (red are obstacles that create black cells in the map):
[image]

The hood of the car seems seen by the camera, causing disturbed point cloud at the bottom of the image, causing the obstacles right in front of the car. To fix this, we can filter the bottom of the depth image by setting `Grid/DepthRoiRatios` to `"0.0 0.0 0.0 0.1" ` (filtering 10% of the image bottom):
[image]

Because the camera is tilted, the noise in on the ground causes vertical obstacles. If TF of the camera is fixed, you may not need to touch the following parameter I used to fix this. Here I set `Grid/MaxGroundAngle` to `60` deg (default 45).
[image]
[image]

By setting `Grid/MaxGroundAngle` higher, the side effect is that little obstacles are considered as ground. Here we cannot detect the right side of the road, and because the cloud is tilted we cannot segment as obstacle the plane detected over the car on left:
[image]

Here are other parameters I've been playing with:
```ini
Grid/CellSize=0.1
Grid/ClusterRadius=0.2
Grid/RangeMax=6
```

You can modifying them using the Core Parameters panel in rtabmap-databaseViewer, show the 3D View, select a frame, then do Edit->Regenerate local grid maps... to re-update grids with current parameters. Press "2" in 3D View to see green/red colors like above when showing the Grid (checkbox below 3D view).

cheers,
Mathieu

NeilNie, 2019-04-19:
@matlabbe Thank you very much for your thorough response! `rtabmap-databaseViewer` is so useful! 

I am just a bit confused about the tf issue. (I am sorry, I am a newbie to ROS tf). In the screenshots above, did you perform some transformation `rtabmap-databaseViewer`? 

You mentioned that "you can show up the point cloud of the zed camera and set fixed frame in global option to base frame you are using". You suggest that I change the fixed frame in rviz, right? 

Thank you very much for your help!

Neil

nathangeorge1, 2019-04-22:
@NeilNie If you're interested in using rtabmap with ZED in ROS, my software does that.

https://www.newmindrobotics.com/

matlabbe, 2019-04-24:
@NeilNie The TF transform between `base_link` and the zed frame (`zed_camera_center `?) should be modified. Where `base_link` -> `zed_camera_center` frame is defined? Is it? This can be created like this:
```xml
<node pkg="tf" type="static_transform_publisher" name="camera_base_link"
        args="0 0 1 0 0.3 0 base_link zed_camera_center 100" /> 
```
In this example the frame `zed_camera_center` is one meter over `base_link` and tilting towards the ground. In rtabmap launch file, set `frame_id` to `base_link` so that the camera cloud is correctly transformed in `base_link` frame:
```xml
<arg name="frame_id"                  value="base_link" />
```
In RVIZ, in global top left options, set the frame fo `base_link`, then show the ZED cloud. The ground in the cloud should match the xy grid if `base_link` -> `zed_camera_center` is correctly set.

cheers,
Mathieu

NeilNie, 2019-05-02:
@matlabbe Thank you for the comment. Most of my issues are solved. Closing for now.

closed 2019-05-02 by NeilNie