introlab/rtabmap_ros #306: How to make it with CUDA  in jetson Tx2 with Mynteye Stereo Camera or Kinect v1.

pamzerbhu, 2019-04-14:
I build Rtabmap in my tx2, but it is so slow. It seems that GPU not work. How to use GPU to
accelerate it?

matlabbe, 2019-04-17:
What is the frame rate of odometry? What is frame rate of the image topics? What is the resolution of the images? The visual odometry implementation of RTAB-Map doesn't have GPU optimizations.

rahulsharma11, 2021-06-02:
Hi,
I am using L515 with Tx2.

while running roslaunch realsense2_ros rs_camera.launch, i am getting continuous warnings like-

 02/06 10:23:48,065 WARNING [140645050664704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 10:23:49,317 WARNING [140645050664704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 10:23:50,569 WARNING [140645050664704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 10:23:50,961 WARNING [140645050664704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 10:23:51,821 WARNING [140645050664704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 10:23:53,074 WARNING [140645050664704] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

and output in rtabmap is also not there.
Any suggestions for efficient use of L515 with TX2?

matlabbe, 2021-06-03:
I would suggest to ask this question on https://github.com/IntelRealSense/realsense-ros