introlab/rtabmap_ros #307: Fuse the lidar 2d grid map and depth camera 3d point cloud map

Atian971, 2019-04-22:
Hi,I'm trying to fuse the lidar and depth camera map into the rviz,
I run this command : [roslaunch openni_launch openni.launch depth_registration:=true] 
to open my XBOX360 depth camera, and then [ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" rviz:=true rtabmapviz:=false] to show the 3d point cloud map in rviz successfully, I can also run gmapping using lidar on my turtlebot, then any ideas how should I fuse the two map into one rviz window?
Thanks in advance.

matlabbe, 2019-04-24:
To see the two maps, they should be linked in TF tree. For example, you may add a static transform between map frame of gmapping and the map frame from rtabmap. The maps will be aligned at start, but as soon there is a loop closure, the maps won't match anymore.

Note that you can only use rtabmap to get 2d map from lidar and the 3d point cloud from camera aligned together. See this example: http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot#Kinect_.2B-_Odometry_.2B-_2D_laser

Having 2 SLAM approaches running at the same time is not trivial, choose one or the other.

cheers,
Mathieu

Atian971, 2019-04-27:
Thank you for your reply. I'm trying to run your demo_hector_mapping.launch using kinect1,rplidar and turtlebot2, the result I want is like thie video:{ https://www.youtube.com/watch?v=MlrcTyXy5No }
how should I change the file to zchieve that result?

matlabbe, 2019-04-30:
What is the TF tree you have after bring up the robot? also which topics are published? There would be only some topic remapping needed in [demo_hector_mapping.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_hector_mapping.launch).