introlab/rtabmap_ros #308: param Optimizer/Slam2D exist?

nathangeorge1, 2019-04-25:
Hello, 

In https://wiki.ros.org/rtabmap_ros/Tutorials/Advanced%20Parameter%20Tuning#Three_DOF_mapping,

It says to do 

```
<node pkg="rtabmap_ros" type="rtabmap" name="rtabmap">
        <param name="Reg/Force3DoF"    value="true" />
        <param name="Optimizer/Slam2D" value="true" />
<node/>
```

But i don't see `Optimizer/Slam2D` in `rosparam list`.

Does it still exist?

matlabbe, 2019-04-26:
`Optimizer/Slam2D` has been removed and replaced by `Reg/Force3DoF`. For backward compatibility, rtabmap should still detect `Optimizer/Slam2D` and convert it to `Reg/Force3DoF` [automatically](https://github.com/introlab/rtabmap/blob/195f6147ad5918a373e7e40644395b7aa2247117/corelib/src/Parameters.cpp#L266).

I tried your launch:
```xml
<launch>
<node pkg="rtabmap_ros" type="rtabmap" name="rtabmap" output="screen">
        <param name="Optimizer/Slam2D"    value="true" />
</node>
</launch>
```
and with:
```bash
$ roslaunch test.launch
...
[ WARN] [1556321766.003320022]: Rtabmap: Parameter name changed: "Optimizer/Slam2D" -> "Reg/Force3DoF". Please update your launch file accordingly. Value "true" is still set to the new parameter name.
...
$ rosparam get /rtabmap/Reg/Force3DoF 
'true'
```

If rtabmap node is under a namespace like this:
```xml
<launch>
<group ns="rtabmap">
   <node pkg="rtabmap_ros" type="rtabmap" name="rtabmap" output="screen">
        <param name="Reg/Force3DoF"    value="true" />
   </node>
</group>
</launch>
```
you should query the parameter like this instead:
```bash
$ roslaunch test.launch
$ rosparam get /rtabmap/Reg/Force3DoF 
'true'
```

Do `$ rosparam list` to see all parameters.

nathangeorge1, 2019-04-29:
Thanks.

This helped me realize that because I'm using pkg zed_rtabmap_example in my parameter yaml file I need to do 

```
rtabmap:
    rtabmap:
        Reg:
            Force3DoF: true
```

not sure why 2 are needed but probably a namespace things

closed 2019-04-29 by nathangeorge1