introlab/rtabmap_ros #309: DBDriverSqlite3.cpp:398::connectDatabaseQuery() Opened database version (0.19.2) is more recent than rtabmap installed version (0.17.6). Please update rtabmap to new version!

jinfagang, 2019-04-26:
I cloned both rtabmap and rtabmap_ros from master, but it says out of date? How do I check which one is out of date?

matlabbe, 2019-04-27:
master is 0.19 right know. In the title, you are starting rtabmap 0.17 and trying to open a 0.19 database. If you have built 0.19, uninstall 0.17 binaries to make sure you are starting the right rtabmap version (`$ which rtabmap`).

Atian971, 2019-04-27:
What command should I use to uninstall 0.17 vision?

jinfagang, 2019-04-28:
@matlabbe  I installed an older version from apt, but rebuilt ros package using git clone master branch, however after purged apt installed one, it still shows the error, even I deleted bin file under /usr/local/bin

matlabbe, 2019-04-29:
To uninstall binaries (0.17):
```bash
$ sudo apt-get remove ros-kinetic-rtabmap
# or if you are on ROS melodic:
$ sudo apt-get remove ros-melodic-rtabmap
```

rtabmap library and rtabmap_ros should have the same version. Building rtabmap_ros against an older rtabmap version may create incompatibilities. The version in `/usr/local` prefix is the one built from source.

See **Build from source** section on the main rtabmap_ros page: https://github.com/introlab/rtabmap_ros#build-from-source

cheers,
Mathieu