introlab/rtabmap_ros #311: Failing to bring lidar to work with Kinect v2 and wheel odometry

Prcuvu, 2019-05-21:
rtabmap works fine until I added lidar into it, failing to output any data with a message:
```
[ WARN] [1558463774.770100255]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /odom_ekf,
   /rtabmap/rgbd_image,
   /scan
```
I tried to check out the topics, they are all subscribed. I have to emphasize that it worked fine before /scan joined the subscription list, and /scan's publish rate seems fine:
```
$ rostopic hz /odom_ekf /rtabmap/rgbd_image /scan
subscribed to [/odom_ekf]
subscribed to [/rtabmap/rgbd_image]
subscribed to [/scan]
       topic           rate   min_delta   max_delta   std_dev    window
=======================================================================
/odom_ekf             10.01   0.09706     0.1053      0.002679   10    
/rtabmap/rgbd_image   4.897   0.1365      0.2994      0.05961    10    
/scan                 10.22   0.08949     0.1212      0.01331    5     

       topic           rate   min_delta   max_delta   std_dev    window
=======================================================================
/odom_ekf             9.985   0.09673     0.1053      0.002543   20    
/rtabmap/rgbd_image   5.623   0.07435     0.2994      0.06122    20    
/scan                 10.11   0.08876     0.1212      0.01304    11    

       topic           rate   min_delta   max_delta   std_dev    window
=======================================================================
/odom_ekf             10.01   0.09161     0.1053      0.002997   30    
/rtabmap/rgbd_image   5.655   0.07435     0.2994      0.06137    30    
/scan                 10.09   0.08846     0.1212      0.01319    17
```
My launch file:
```
<launch>
  <!-- Rotate the camera -->
  <arg name="pi/2" value="1.5707963267948966"/>
  <arg name="optical_rotate" value="0 0 0 -$(arg pi/2) 0 -$(arg pi/2)" />
  <node pkg="tf" type="static_transform_publisher" name="kinect2_base_link"
       args="$(arg optical_rotate) base_link kinect2_link 100" />

  <group ns="rtabmap">

    <!-- Use RGBD synchronization -->
    <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="standalone rtabmap_ros/rgbd_sync" output="screen">
      <remap from="rgb/image"       to="/kinect2/qhd/image_color_rect"/>
      <remap from="depth/image"     to="/kinect2/qhd/image_depth_rect"/>
      <remap from="rgb/camera_info" to="/kinect2/qhd/camera_info"/>
      <remap from="rgbd_image"      to="rgbd_image"/> <!-- output -->
      
      <param name="queue_size" type="int" value="10"/>
      <!-- Should be true for not synchronized camera topics 
           (e.g., false for kinectv2, zed, realsense, true for xtion, kinect360)-->
      <param name="approx_sync"       value="false"/>
    </node>

    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start">
          <param name="frame_id" type="string" value="base_link"/>

          <param name="subscribe_depth" type="bool" value="false"/>
          <param name="subscribe_rgbd" type="bool" value="true"/>
          <param name="subscribe_scan" type="bool" value="true"/>

          <remap from="rgb/image" to="/kinect2/qhd/image_color_rect"/>
          <remap from="depth/image" to="/kinect2/qhd/image_depth_rect"/>
          <remap from="rgb/camera_info" to="/kinect2/qhd/camera_info"/>
          <remap from="odom" to="/odom_ekf"/>
          <remap from="scan" to="/scan"/>
          <remap from="rgbd_image" to="rgbd_image"/>

          <param name="queue_size" type="int" value="10"/>
          <param name="approx_sync" type="bool" value="true"/>

          <!-- RTAB-Map's parameters -->
          <param name="RGBD/NeighborLinkRefining" type="string" value="true"/>
          <param name="RGBD/ProximityBySpace"     type="string" value="true"/>
          <param name="RGBD/ProximityPathMaxNeighbors" type="string" value="10"/>
          <param name="RGBD/AngularUpdate"        type="string" value="0.4"/>
          <param name="RGBD/LinearUpdate"         type="string" value="0.3"/>
          <param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/>
          <param name="Grid/FromDepth"            type="string" value="false"/> <!-- occupancy grid from lidar -->
          <param name="Reg/Force3DoF"             type="string" value="true"/>
          <param name="Reg/Strategy"              type="string" value="1"/> <!-- 1=ICP -->
          
          <!-- ICP parameters -->
          <param name="Icp/VoxelSize"                 type="string" value="0.05"/>
          <param name="Icp/MaxCorrespondenceDistance" type="string" value="0.1"/>
          
          <param name="GFTT/BlockSize" type="string" value="5"/>
          <param name="GFTT/MinDistance" type="string" value="5"/>
    </node>
  </group>
</launch>
```

matlabbe, 2019-05-21:
Hi,

If switching off the scan is working by setting "subscribe_scan" to false, it means it is a synchronization problem. As all topics are published, there could be wrong timestamps somewhere, maybe the scan timestamp is off a couple of seconds from the camera topic. Are they published by the same computer? If not, make sure the clocks of the computers are synchronized (with `ntpdate`).

cheers,
Mathieu

Prcuvu, 2019-05-22:
I `rostopic echo`ed /scan topic and noticed that the lidar driver failed to publish timestamp for unknown reasons, while there is nothing wrong with the source code. It gets fixed by simply recompiling the driver, and everything starts working.

closed 2019-05-22 by Prcuvu