introlab/rtabmap_ros #312: make error: /usr/bin/ld: cannot find -lvtkGUISupportQt

hachbani2019, 2019-05-23:
,
I'm trying to install rtabmap_ros for a 3D mapping object using a LiDAR. I followed the instructions explained in the readme file. When I execute the make command, I get this error:

```
hamza@hamza-Latitude-5590:~/rtabmap/build$ cmake ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
-- Found eigen: /usr/include/eigen3  
-- Checking for module 'libopenni'
--   Found libopenni, version 1.5.4.0
-- Found openni: /usr/lib/libOpenNI.so  
-- Found libusb-1.0: /usr/include  
-- Checking for module 'flann'
--   Found flann, version 1.8.4
-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
-- Found qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- Found PCL_COMMON: /usr/local/lib/libpcl_common.so  
-- Found PCL_OCTREE: /usr/local/lib/libpcl_octree.so  
-- Found PCL_IO: /usr/local/lib/libpcl_io.so  
-- Found PCL_KDTREE: /usr/local/lib/libpcl_kdtree.so  
-- Found PCL_SEARCH: /usr/local/lib/libpcl_search.so  
-- Found PCL_SURFACE: /usr/local/lib/libpcl_surface.so  
-- Found PCL_FILTERS: /usr/local/lib/libpcl_filters.so  
-- Found PCL_2D: /usr/local/include/pcl-1.8  
-- Found PCL_FEATURES: /usr/local/lib/libpcl_features.so  
-- Found PCL_REGISTRATION: /usr/local/lib/libpcl_registration.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/local/lib/libpcl_sample_consensus.so  
-- Found PCL_GEOMETRY: /usr/local/include/pcl-1.8  
-- Found PCL_ML: /usr/local/lib/libpcl_ml.so  
-- Found PCL_SEGMENTATION: /usr/local/lib/libpcl_segmentation.so  
-- Found PCL_VISUALIZATION: /usr/local/lib/libpcl_visualization.so  
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Found OpenMP
-- Found OpenCV: /opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv
-- Found PCL: /usr/local/include/pcl-1.8;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/local/include/vtk-7.1
-- Found ZLIB: /usr/include
-- VTK_RENDERING_BACKEND=OpenGL2
-- Found Freenect: /usr/include
-- Found DC1394: /usr/include/dc1394
-- Found CSPARSE: /usr/include/suitesparse  
-- Old g2o version detected with c++03 interface (config file: /opt/ros/kinetic/include/g2o/config.h).
-- Found g2o: /opt/ros/kinetic/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
[... 188 lines cut ...]
[ 36%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/opencv/solvepnp.cpp.o
[ 36%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/toro3d/posegraph3.cpp.o
[ 36%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/toro3d/treeoptimizer3_iteration.cpp.o
[ 37%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/toro3d/treeoptimizer3.cpp.o
[ 37%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/toro3d/posegraph2.cpp.o
[ 37%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/toro3d/treeoptimizer2.cpp.o
[ 38%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/g2o/edge_se3_xyzprior.cpp.o
[ 38%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/edge_se2Switchable.cpp.o
[ 38%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/edge_se3Switchable.cpp.o
[ 39%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/edge_switchPrior.cpp.o
[ 39%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/types_g2o_robust.cpp.o
[ 39%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/vertex_switchLinear.cpp.o
[ 40%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/OctoMap.cpp.o
[ 40%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/gtsam/GravityFactor.cpp.o
[ 40%] Linking CXX shared library ../../../bin/librtabmap_core.so
/usr/bin/ld: cannot find -lvtkGUISupportQt
collect2: error: ld returned 1 exit status
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:3041: recipe for target '../bin/librtabmap_core.so.0.19.2' failed
make[2]: *** [../bin/librtabmap_core.so.0.19.2] Error 1
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```
Any help please ?

matlabbe, 2019-05-23:
I see `/usr/local/include/vtk-7.1`, make sure VTK is built with [Qt module](https://github.com/Kitware/VTK/tree/v7.1.1/GUISupport/Qt):
```bash
$ cd VTK/build
$ cmake -DVTK_Group_Qt=ON -DVTK_QT_VERSION=5 ..
$ make 
$ make install
```
cheers,
Mathieu

hachbani2019, 2019-05-24:
Hi, 
Thanks your for your response. Though, I don't understand your instructions since Im a ROS beginner, Do I have to download the QT module ? when Im in the rtabmap folder, there's no VTK folder, so the cd VTK/build doesn't work.

matlabbe, 2019-05-29:
For a beginner, I don't recommend to build VTK and PCL from source. There is no Qt module to download, it should be built. From the log you shown, you have built VTK from source somewhere (as it is installed in `/usr/local`). When building VTK, we should enable the options I've shown in my previous post so that `vtkGUISupportQt` library is built. Not sure which Qt version is installed on your computer, but set "`-DVTK_QT_VERSION=4`" instead if it is Qt4.

Another approach (the normal approach) is to uninstall your PCL and VTK versions installed in `/usr/local` to use the binary ones instead (for which the vtk Qt module is already built).