introlab/rtabmap_ros #313: Mapping and Navigation with Turtlebot with realsense D435i

MattixLee, 2019-05-29:
@matlabbe  hello,

Thank you for your work, I want to try to finish the work in the Mapping and Navigation with Turtlebot（http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot#If_you_don.27t_have_the_robot）.

I do not have the robot, but I have a realsense d435i,
when I executed ,roslaunch rtabmap_ros demo_turtlebot_mapping.launch args:="--delete_db_on_start" rgbd_odometry:=true, the error was occured.
（No matching device found.... waiting for devices. Reason: std::__cxx11::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /tmp/binarydeb/ros-kinetic-openni2-camera-0.4.2/src/openni2_driver.cpp @ 737 : Invalid device number 1, there are 0 devices connected.）

I think it is a problem of 3d seneor-driven, but I did not find seneor-driven of relasense d435i.
Can you help me solve it? Thank you !

cheers!

matlabbe, 2019-05-30:
Hi,

The included [3dsensor.launch](https://github.com/turtlebot/turtlebot/blob/kinetic/turtlebot_bringup/launch/3dsensor.launch) from [demo_turtlebot_mapping.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_turtlebot_mapping.launch) includes one of the sensor in this directory: https://github.com/turtlebot/turtlebot/tree/kinetic/turtlebot_bringup/launch/includes/3dsensor

There is no D435i file, you would have to create one, maybe based on the R200 one.

But as you don't have the robot, did you try the hand-held tutorial? https://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping
There is an example with D435.

cheers,
Mathieu

MattixLee, 2019-05-31:
@matlabbe 

Thank you for your reply! I have finished mapping with the hand-held tutorial, a map was very good.
I did navigation in gazebo, and I'm going to use Kinect to complete the course.

cheers,
Lee