introlab/rtabmap_ros #314: How to use/read the stereo camera(ZED) pose information in mapping mode

JoeyLeeNPU, 2019-06-03:
We try to use RTAB-map for UAV navigation in a GPS-denied environment, and we use a stereo to provide images as "input" to rtab-map.
Now we can see many data has been published on the topics without any pose data.
We don't want to use move_base pkg, hence, only pose and map information is enough.

Thanks for response very much!

matlabbe, 2019-06-03:
Look at this page: http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping
There is an example with ZED cameras, using their odometry approach or the one from rtabmap.

cheers,
Mathieu

JoeyLeeNPU, 2019-06-04:
Thank you very much!
We have tested the method in that toturial and run "tf" to get the "pose" between /map to /base_link, but the /map to /odom is not published by /rtab-map.
As we want to input ZED's /odom and obtain "pose" after rtab-map's optimation, we set FALSE on "visual_odometry",  "ZED/odom" on "odom_topic",  TRUE on ZED's "tracking/publish_tf", FALSE on ZED's "tracking/publish_map_tf". Then we listen TF tree between /map to /base_link to obtain "pose" that I want, the TF tree show /map published by /rtab-map.
We also compared zed's pose with the result from TF tree in order to varify if our method is right.

We hope you can tell us if something wrong in above steps. Waiting for you response.
Thanks again!

matlabbe, 2019-06-04:
If /map to /odom is not published, look if there are warnings on the terminal (or rqt_console) about missing input topics for rtabmap node.

closed 2019-06-05 by JoeyLeeNPU