introlab/rtabmap_ros #315: rtabmapviz segmentation fault

ehong-tl, 2019-06-06:
I encounter segmentation fault when trying to run `rosrun rtabmap_ros rtabmapviz`.

`tl-mems-workstation@tlmemsworkstation-HP-Z420-Workstation` ~ $ rosrun rtabmap_ros rtabmapviz
[ INFO] [1559786243.701028676]: Starting node...
[ INFO] [1559786243.804252221]: rtabmapviz: Using configuration from "/home/tl-mems-workstation/.ros/rtabmapGUI.ini"
Segmentation fault`

I try to run with `gdb` and the output is as follows.

`tl-mems-workstation@tlmemsworkstation-HP-Z420-Workstation ~ $ rosrun --prefix \'gdb -ex run --args\' rtabmap_ros rtabmapviz 
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/tl-mems-workstation/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz...(no debugging symbols found)...done.
Starting program: /home/tl-mems-workstation/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffa0ba5700 (LWP 6909)]
[New Thread 0x7fffa03a4700 (LWP 6910)]
[New Thread 0x7fff9fba3700 (LWP 6911)]
[New Thread 0x7fff9f3a2700 (LWP 6912)]
[New Thread 0x7fff9eba1700 (LWP 6913)]
[New Thread 0x7fff9e3a0700 (LWP 6914)]
[New Thread 0x7fff9db9f700 (LWP 6915)]
[New Thread 0x7fff9d39e700 (LWP 6916)]
[ INFO] [1559786519.213278576]: Starting node...
[New Thread 0x7fff9a14e700 (LWP 6917)]
[New Thread 0x7fff8de8b700 (LWP 6919)]
[New Thread 0x7fff8d68a700 (LWP 6920)]
[New Thread 0x7fff8ce89700 (LWP 6921)]
[New Thread 0x7fff7ffff700 (LWP 6922)]
[New Thread 0x7fff7f7fe700 (LWP 6924)]
[New Thread 0x7fff7effd700 (LWP 6925)]
[New Thread 0x7fff7e7fc700 (LWP 6926)]
[New Thread 0x7fff7dffb700 (LWP 6931)]
[New Thread 0x7fff7d7fa700 (LWP 6944)]
[ INFO] [1559786519.578819641]: rtabmapviz: Using configuration from "/home/tl-mems-workstation/.ros/rtabmapGUI.ini"
[Thread 0x7fff8ce89700 (LWP 6921) exited]
[New Thread 0x7fff8ce89700 (LWP 6963)]
[New Thread 0x7fff654ea700 (LWP 6964)]
[New Thread 0x7fff64ce9700 (LWP 6965)]
[New Thread 0x7fff5a263700 (LWP 6966)]
[New Thread 0x7fff57a62700 (LWP 6967)]
[New Thread 0x7fff55261700 (LWP 6968)]
[New Thread 0x7fff52a60700 (LWP 6969)]
[New Thread 0x7fff5025f700 (LWP 6970)]

Thread 1 "rtabmapviz" received signal SIGSEGV, Segmentation fault.
0x00007fffcf7ff477 in vtkSmartPointerBase::~vtkSmartPointerBase() () from /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.2.so.6.2`

It does not occur every time, some time `rtabmapviz` can open successfully, but most of the time I get segmentation fault. However, I have no problem running standalone `rtabmap`.

Need help to fix this issue. Thanks.

matlabbe, 2019-06-06:
Can you do "`bt`" (backtrace) when in crash in gdb?

ehong-tl, 2019-06-07:
@matlabbe 

Thsi is the output from `bt` after crashed in `gdb`.

```
(gdb) bt
#0  0x00007fffcf7ff47b in vtkSmartPointerBase::~vtkSmartPointerBase() ()
   from /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.2.so.6.2
#1  0x00007fffcf7ff5b7 in vtkSmartPointerBase::operator=(vtkSmartPointerBase const&) () from /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.2.so.6.2
#2  0x00007fffeceb51ba in pcl::visualization::PCLVisualizerInteractorStyle::Initialize() () from /usr/lib/x86_64-linux-gnu/libpcl_visualization.so.1.7
#3  0x00007fffeceec9ba in pcl::visualization::PCLVisualizer::PCLVisualizer(int&, char**, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pcl::visualization::PCLVisualizerInteractorStyle*, bool) ()
   from /usr/lib/x86_64-linux-gnu/libpcl_visualization.so.1.7
#4  0x00007fffed4af849 in rtabmap::CloudViewer::CloudViewer(QWidget*, rtabmap::CloudViewerInteractorStyle*) () from /usr/local/lib/librtabmap_gui.so.0.19
#5  0x00007fffed2f6abd in rtabmap::MainWindow::MainWindow(rtabmap::PreferencesDialog*, QWidget*, bool) () from /usr/local/lib/librtabmap_gui.so.0.19
#6  0x0000000000528545 in rtabmap_ros::GuiWrapper::GuiWrapper(int&, char**) ()
#7  0x00000000005226d4 in main ()
```

matlabbe, 2019-06-07:
Hi,
This bug is related to PCL visualizer when initializing the Interactor. Not sure if it is related, but you can try to uncomment the first line and comment the second line here:
https://github.com/introlab/rtabmap/blob/b5f5623af4fe23580d5d9077c454b615d81f0977/guilib/src/CloudViewer.cpp#L161-L162
```cpp
//_visualizer->setupInteractor(this->GetInteractor(), this->GetRenderWindow());
this->GetInteractor()->SetInteractorStyle (_visualizer->getInteractorStyle());
```
then rebuild/install rtabmap library.

cheers,
Mathieu

ehong-tl, 2019-06-10:
Hi @matlabbe 

I tried your method, still doesn't fix the issue,

```
Thread 1 "rtabmapviz" received signal SIGSEGV, Segmentation fault.
0x00007fffcf7ff477 in vtkSmartPointerBase::~vtkSmartPointerBase() ()
   from /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.2.so.6.2
(gdb) bt
#0  0x00007fffcf7ff477 in vtkSmartPointerBase::~vtkSmartPointerBase() ()
   from /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.2.so.6.2
#1  0x00007fffcf7ff5b7 in vtkSmartPointerBase::operator=(vtkSmartPointerBase const&) () from /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.2.so.6.2
#2  0x00007fffeceb51ba in pcl::visualization::PCLVisualizerInteractorStyle::Initialize() () from /usr/lib/x86_64-linux-gnu/libpcl_visualization.so.1.7
#3  0x00007fffeceec9ba in pcl::visualization::PCLVisualizer::PCLVisualizer(int&, char**, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pcl::visualization::PCLVisualizerInteractorStyle*, bool) ()
   from /usr/lib/x86_64-linux-gnu/libpcl_visualization.so.1.7
#4  0x00007fffed4af919 in rtabmap::CloudViewer::CloudViewer(QWidget*, rtabmap::CloudViewerInteractorStyle*) () from /usr/local/lib/librtabmap_gui.so.0.19
#5  0x00007fffed2f6b8d in rtabmap::MainWindow::MainWindow(rtabmap::PreferencesDialog*, QWidget*, bool) () from /usr/local/lib/librtabmap_gui.so.0.19
#6  0x0000000000528545 in rtabmap_ros::GuiWrapper::GuiWrapper(int&, char**) ()
#7  0x00000000005226d4 in main ()
```

matlabbe, 2019-06-11:
There is something in `pcl::visualization::PCLVisualizerInteractorStyle::Initialize()` that goes wrong. In rtabmap side, it seems crashing on PCLVisualizer constructor [here](https://github.com/introlab/rtabmap/blob/b5f5623af4fe23580d5d9077c454b615d81f0977/guilib/src/CloudViewer.cpp#L131-L136):
```
_visualizer = new pcl::visualization::PCLVisualizer(
    argc, 
    0, 
    "PCLVisualizer", 
    style,
    false);
```

Based on the [constructor declaration](http://docs.pointclouds.org/1.8.1/classpcl_1_1visualization_1_1_p_c_l_visualizer.html):
```cpp
PCLVisualizer (
    int &argc, 
    char **argv, 
    const std::string &name="", 
    PCLVisualizerInteractorStyle *style=PCLVisualizerInteractorStyle::New(), 
    const bool create_interactor=true)
 ```
I don't know if it is related to last param set to false or the style object. You may try the default constructor.

cheers,
Mathieu

ehong-tl, 2019-06-12:
Hi @matlabbe 

I've tried using the default constructor as you suggested, it made matter worse. Now I got `Segmentation Fault `when trying to run standalone `rtabmap` too.

```
Thread 1 "rtabmap" received signal SIGSEGV, Segmentation fault.
0x00007ffff79b2af0 in rtabmap::CloudViewer::CloudViewer(QWidget*, rtabmap::CloudViewerInteractorStyle*) () from /usr/local/lib/librtabmap_gui.so.0.19
(gdb) bt
#0  0x00007ffff79b2af0 in rtabmap::CloudViewer::CloudViewer(QWidget*, rtabmap::CloudViewerInteractorStyle*) () from /usr/local/lib/librtabmap_gui.so.0.19
#1  0x00007ffff77f9b3d in rtabmap::MainWindow::MainWindow(rtabmap::PreferencesDialog*, QWidget*, bool) () from /usr/local/lib/librtabmap_gui.so.0.19
#2  0x0000000000406f18 in main ()
```

matlabbe, 2019-06-12:
The standalone was working... not sure why rtabmapviz cannot work too (maybe caused by some ROS dependencies linking on different version of the same library). A workaround is to use RVIZ for visualization instead (you can add rtabmap_ros/MapCloud display to see the 3D map).

ehong-tl, 2019-06-13:
Hi @matlabbe ,

Ok thanks, I will use RVIZ instead for visualization.