introlab/rtabmap_ros #316: the package version is 0.19.2 and the library files installed at step 2 are version 0.17.6

AyushJhalani, 2019-06-06:
getting this error even after sequentially following the steps:

CMake Error at rtabmap_ros/CMakeLists.txt:34 (find_package):
  Could not find a configuration file for package "RTABMap" that is
  compatible with requested version "0.19.3".

  The following configuration files were considered but not accepted:

    /opt/ros/kinetic/lib/x86_64-linux-gnu/rtabmap-0.17/RTABMapConfig.cmake, version: 0.17.6

-- Configuring incomplete, errors occurred!
See also "/home/tstone10/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/tstone10/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

matlabbe, 2019-06-06:
Following the [build from source instructions](https://github.com/introlab/rtabmap_ros#build-from-source), don't forget to uninstall the binaries:
```bash
$ sudo apt-get remove ros-kinetic-rtabmap ros-kinetic-rtabmap-ros
```

AyushJhalani, 2019-06-06:
Worked. In addition to this I was not compiling the package outside catkin that install the required libraries.

closed 2019-06-06 by AyushJhalani