introlab/rtabmap_ros #318: For Realsense D435

Garand0o0, 2019-06-14:
Hi,
`roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" depth_topic:=/camera/aligned_depth_to_color/image_raw rgb_topic:=/camera/color/image_raw camera_info_topic:=/camera/color/camera_info approx_sync:=false`
Is the `image_raw` pre-calibration of the d435?

`roslaunch realsense2_camera rs_camera.launch align_depth:=true`
Does the d435 only send pre-calibration data?

how can I use the calibrated date of d435?

thanks

matlabbe, 2019-06-16:
It depends on what means "`image_raw`" for the realsense driver. Is there an `image_rect` topic? If so, make sure the registered depth image matches the `image_rect` topic, and you can use it. What are the topics available when you launch realsense2 with:
```bash
$roslaunch realsense2_camera rs_camera.launch align_depth:=true
```
?
```bash
$ rostopic list
```
In this example, we assume that raw RGB image has low distortions. RTAB-Map requires that input images are already rectified.

cheers,
Mathieu

Garand0o0, 2019-06-17:
> It depends on what means "`image_raw`" for the realsense driver. Is there an `image_rect` topic? If so, make sure the registered depth image matches the `image_rect` topic, and you can use it. What are the topics available when you launch realsense2 with:
> 
> ```shell
> $roslaunch realsense2_camera rs_camera.launch align_depth:=true
> ```
> 
> ?
> 
> ```shell
> $ rostopic list
> ```
> 
> In this example, we assume that raw RGB image has low distortions. RTAB-Map requires that input images are already rectified.
> 
> cheers,
> Mathieu

Thank you for your reply.

Angit16, 2021-08-25:
Hi Matheu,
I have a doubt related to this issue. I am using D435 Model in a Gazebo simulation. The list of topics published in the beginning after the camera is up, are given below.

`/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_raw
/camera/depth/image_raw/compressed
/camera/depth/image_raw/compressed/parameter_descriptions
/camera/depth/image_raw/compressed/parameter_updates
/camera/depth/image_raw/compressedDepth
/camera/depth/image_raw/compressedDepth/parameter_descriptions
/camera/depth/image_raw/compressedDepth/parameter_updates
/camera/depth/image_raw/theora
/camera/depth/image_raw/theora/parameter_descriptions
/camera/depth/image_raw/theora/parameter_updates
/camera/infra1/camera_info
/camera/infra1/image_raw
/camera/infra1/image_raw/compressed
/camera/infra1/image_raw/compressed/parameter_descriptions
/camera/infra1/image_raw/compressed/parameter_updates
/camera/infra1/image_raw/compressedDepth
/camera/infra1/image_raw/compressedDepth/parameter_descriptions
/camera/infra1/image_raw/compressedDepth/parameter_updates
/camera/infra1/image_raw/theora
/camera/infra1/image_raw/theora/parameter_descriptions
/camera/infra1/image_raw/theora/parameter_updates
/camera/infra2/camera_info
/camera/infra2/image_raw
/camera/infra2/image_raw/compressed
/camera/infra2/image_raw/compressed/parameter_descriptions
/camera/infra2/image_raw/compressed/parameter_updates
/camera/infra2/image_raw/compressedDepth
/camera/infra2/image_raw/compressedDepth/parameter_descriptions
/camera/infra2/image_raw/compressedDepth/parameter_updates
/camera/infra2/image_raw/theora
/camera/infra2/image_raw/theora/parameter_descriptions
/camera/infra2/image_raw/theora/parameter_updates
`

matlabbe, 2021-09-04:
You may have to use `image_raw` instead of `image_rect_raw`. With the simulator, the `_raw` topics are generally already rectified.

yuan-0816, 2023-11-23:
I've encountered a similar issue.
I want to simulate the RealSense D435i in Gazebo and use RTAB-Map for localization. 
Here are my xacro and launch files.

`<?xml version="1.0"?>

<robot xmlns:xacro="http://ros.org/wiki/xacro">
    <xacro:macro name="gazebo_d435i" params=" reference_link
                                            camera_name:=camera
                                            topics_ns:=camera

                                            depth_optical_frame
                                            color_optical_frame
                                            infrared1_optical_frame
                                            infrared2_optical_frame
                                            accel_optical_frame
                                            gyro_optical_frame

                                            visualize:=false
                                            align_depth:=false
                                            enable_pointCloud:=false

                                            unite_imu_method:=false
                                            accel_fps:=250
                                            gyro_fps:=400

                                            clip_distance:=-1.0
                                            depth_width:=1280
                                            depth_height:=720
                                            depth_fps:=30

                                            infra_width:=640
                                            infra_height:=480
                                            infra_fps:=30

                                            color_width:=1920
                                            color_height:=1080
                                            color_fps:=30
                                            ">
        
        <xacro:property name="deg_to_rad" value="0.01745329251994329577"/>
        <gazebo reference="${reference_link}">
            <self_collide>0</self_collide>
            <enable_wind>0</enable_wind>
            <kinematic>0</kinematic>
            
            
            <mu2>1</mu2>
            <fdir1>0 0 0</fdir1>
            
            <kp>1e+13</kp>
            <kd>1</kd>
            
            <sensor name="${camera_name}color" type="camera">
                <camera name="${camera_name}">
                    <horizontal_fov>${69.4*deg_to_rad}</horizontal_fov>
                    <image>
                        <width>${color_width}</width>
                        <height>${color_height}</height>
                        <format>RGB_INT8</format>
                    </image>
                    <clip>
                        <near>0.1</near>
                        <far>100</far>
                    </clip>
                    <noise>
                        <type>gaussian</type>
                        <mean>0.0</mean>
                        <stddev>0.007</stddev>
                    </noise>
[... 461 lines cut ...]
[rviz2-3] [INFO] [1700752195.882848604] [rclcpp]: signal_handler(signum=2)
[imu_filter_madgwick_node-5] [INFO] [1700752195.884572765] [rclcpp]: signal_handler(signum=2)
[rgbd_odometry-1] [WARN] [1700752195.888917444] [rgbd_odometry]: rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. 
[rgbd_odometry-1] rgbd_odometry subscribed to (exact sync):
[rgbd_odometry-1]    /camera/infra1/image_rect_raw,
[rgbd_odometry-1]    /camera/depth/image_rect_raw,
[rgbd_odometry-1]    /camera/infra1/camera_info
[rtabmap-2] [INFO] [1700752195.932880938] [rtabmap]: Parameters are not saved (No configuration file provided...)
[INFO] [imu_filter_madgwick_node-5]: process has finished cleanly [pid 48029]
[INFO] [point_cloud_xyzrgb-4]: process has finished cleanly [pid 48027]
[INFO] [rviz2-3]: process has finished cleanly [pid 48025]
[ERROR] [rtabmap-2]: process[rtabmap-2] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[ERROR] [rgbd_odometry-1]: process[rgbd_odometry-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [rtabmap-2]: sending signal 'SIGTERM' to process[rtabmap-2]
[INFO] [rgbd_odometry-1]: sending signal 'SIGTERM' to process[rgbd_odometry-1]
[rtabmap-2] [INFO] [1700752200.887038270] [rclcpp]: signal_handler(signum=15)
[rgbd_odometry-1] [INFO] [1700752200.887479670] [rclcpp]: signal_handler(signum=15)
[rtabmap-2] rtabmap: Saving database/long-term memory... (located at /home/yuan/.ros/rtabmap.db)
[rtabmap-2] rtabmap: Saving database/long-term memory...done! (located at /home/yuan/.ros/rtabmap.db, 0 MB)
[INFO] [rtabmap-2]: process has finished cleanly [pid 48023]
[INFO] [rgbd_odometry-1]: process has finished cleanly [pid 48021]
`
[image]

"I'm using ROS2 Humble on Ubuntu 22.04."

matlabbe, 2023-11-24:
I suggest to do `rostopic hz` instead of `rostopic list` to acutally know that the topics are published.
```
rostopic hz /camera/infra1/image_rect_raw
rostopic hz /camera/depth/image_rect_raw
rostopic hz /camera/infra1/camera_info
```
I see you are also using `approx_sync` set to false (exact sync), are the image topics published with same stamp?

matlabbe, 2023-11-24:
> I've encountered a similar issue. I want to simulate the RealSense D435i in Gazebo and use RTAB-Map for localization. Here are my xacro and launch files.
> 
> ```
> <?xml version="1.0"?>
> 
> <robot xmlns:xacro="http://ros.org/wiki/xacro">
>     <xacro:macro name="gazebo_d435i" params=" reference_link
>                                             camera_name:=camera
>                                             topics_ns:=camera
> 
>                                             depth_optical_frame
>                                             color_optical_frame
>                                             infrared1_optical_frame
>                                             infrared2_optical_frame
>                                             accel_optical_frame
>                                             gyro_optical_frame
> 
>                                             visualize:=false
>                                             align_depth:=false
>                                             enable_pointCloud:=false
> 
>                                             unite_imu_method:=false
>                                             accel_fps:=250
>                                             gyro_fps:=400
> 
>                                             clip_distance:=-1.0
>                                             depth_width:=1280
>                                             depth_height:=720
>                                             depth_fps:=30
> 
>                                             infra_width:=640
>                                             infra_height:=480
>                                             infra_fps:=30
> 
>                                             color_width:=1920
>                                             color_height:=1080
>                                             color_fps:=30
>                                             ">
>         
>         <xacro:property name="deg_to_rad" value="0.01745329251994329577"/>
>         <gazebo reference="${reference_link}">
>             <self_collide>0</self_collide>
>             <enable_wind>0</enable_wind>
>             <kinematic>0</kinematic>
>             
>             
>             <mu2>1</mu2>
>             <fdir1>0 0 0</fdir1>
>             
>             <kp>1e+13</kp>
>             <kd>1</kd>
>             
>             <sensor name="${camera_name}color" type="camera">
>                 <camera name="${camera_name}">
>                     <horizontal_fov>${69.4*deg_to_rad}</horizontal_fov>
>                     <image>
>                         <width>${color_width}</width>
>                         <height>${color_height}</height>
>                         <format>RGB_INT8</format>
>                     </image>
>                     <clip>
>                         <near>0.1</near>
>                         <far>100</far>
>                     </clip>
>                     <noise>
>                         <type>gaussian</type>
>                         <mean>0.0</mean>
>                         <stddev>0.007</stddev>
>                     </noise>
>                 </camera>
[... 477 lines cut ...]
> [rgbd_odometry-1] rgbd_odometry subscribed to (exact sync):
> [rgbd_odometry-1]    /camera/infra1/image_rect_raw,
> [rgbd_odometry-1]    /camera/depth/image_rect_raw,
> [rgbd_odometry-1]    /camera/infra1/camera_info
> [rtabmap-2] [INFO] [1700752195.932880938] [rtabmap]: Parameters are not saved (No configuration file provided...)
> [INFO] [imu_filter_madgwick_node-5]: process has finished cleanly [pid 48029]
> [INFO] [point_cloud_xyzrgb-4]: process has finished cleanly [pid 48027]
> [INFO] [rviz2-3]: process has finished cleanly [pid 48025]
> [ERROR] [rtabmap-2]: process[rtabmap-2] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
> [ERROR] [rgbd_odometry-1]: process[rgbd_odometry-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
> [INFO] [rtabmap-2]: sending signal 'SIGTERM' to process[rtabmap-2]
> [INFO] [rgbd_odometry-1]: sending signal 'SIGTERM' to process[rgbd_odometry-1]
> [rtabmap-2] [INFO] [1700752200.887038270] [rclcpp]: signal_handler(signum=15)
> [rgbd_odometry-1] [INFO] [1700752200.887479670] [rclcpp]: signal_handler(signum=15)
> [rtabmap-2] rtabmap: Saving database/long-term memory... (located at /home/yuan/.ros/rtabmap.db)
> [rtabmap-2] rtabmap: Saving database/long-term memory...done! (located at /home/yuan/.ros/rtabmap.db, 0 MB)
> [INFO] [rtabmap-2]: process has finished cleanly [pid 48023]
> [INFO] [rgbd_odometry-1]: process has finished cleanly [pid 48021]
> ```
> 
> [image]
> 
> "I'm using ROS2 Humble on Ubuntu 22.04."

Discussion moved to https://github.com/introlab/rtabmap_ros/issues/1070

closed 2023-11-24 by matlabbe

matlabbe, 2025-08-03:
It looks like you mean "at high speed odometry gets lost". I suggest to use IR-Depth mode with realsense D435i, for larger field of view, less motion blur and perfect IR-depth synchronization. See this example: https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_examples/launch/realsense_d435i_infra.launch.py

Then if the camera gets lost again, read "Lost Odometry" section here: https://github.com/introlab/rtabmap/wiki/Kinect-mapping#lost-odometry-red-screens