introlab/rtabmap_ros #320: Synchronization problem with lidar and camera

Lily-XueCY, 2019-06-22:
Hi,
I run RTAB-MAP on Ubuntu 16.04 (ROS: kinetic 1.12.14) using Ouster OS1-64 and Realsense D435i. Lidar and camera work well when I use them separately, but I never get the map if I use them together and it always comes with the warning:

> /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom,
   /rtabmap/rgbd_image,
   /os1_cloud_node/points

I checked all these topics and found all of them were published with timestamps (screenshots are attached), so I think it may because of the synchronization problem.
[image]
[image]
Here is my launch file:
<launch>

  
  <node pkg="tf2_ros" type="static_transform_publisher" name="base_link_to_camera_link" args="0 0 5 0 0 0 base_link camera_link"/>
  <node pkg="tf2_ros" type="static_transform_publisher" name="base_link_to_os1_sensor" args="0 0 0 0 0 0 base_link os1_sensor"/>
  
  <group ns="rtabmap">

    <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="standalone rtabmap_ros/rgbd_sync" output="screen">
      <remap from="rgb/image"       to="/camera/color/image_raw"/>
      <remap from="depth/image"     to="/camera/aligned_depth_to_color/image_raw"/>
      <remap from="rgb/camera_info" to="/camera/color/camera_info"/>   
      <remap from="rgbd_image"       to="rgbd_image"/> 
      
      
      <param name="approx_sync"       value="false"/> 
    </node>

    
    <node pkg="rtabmap_ros" type="rgbd_odometry" name="rgbd_odometry" output="screen">
      <param name="subscribe_rgbd" type="bool"   value="true"/>
      <param name="frame_id"       type="string" value="base_link"/>
      <remap from="rgbd_image" to="rgbd_image"/>
    </node>

    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start">
          <param name="frame_id"        type="string" value="base_link"/>
          <param name="subscribe_depth" type="bool"   value="false"/>
          <param name="subscribe_rgbd"  type="bool"   value="true"/>
          <param name="subscribe_scan_cloud"  type="bool"   value="true"/>

          <remap from="odom"       to="odom"/>
          <remap from="rgbd_image" to="rgbd_image"/>
          <remap from="scan_cloud" to="/os1_cloud_node/points"/>

          <param name="queue_size"  type="int"  value="10"/>
          <param name="approx_sync" type="bool" value="true"/>

          
          <param name="RGBD/AngularUpdate"        type="string" value="0.01"/>
          <param name="RGBD/LinearUpdate"         type="string" value="0.01"/>
          <param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/>
    </node>
  </group>
</launch>
Do you have any idea how to fix this problem? 
Thank you.

matlabbe, 2019-06-24:
The problem is that os1 cloud node publishes clouds with its own clock (starting at zero), make it impossible to synchronize with other sensors afterwards. A quick fix could be to add the following before [this](https://github.com/ouster-lidar/ouster_example/blob/e890d32608ca058da3e2915813e168c93c6f9985/ouster_ros/src/os1_cloud_node.cpp#L62):
```cpp
msg.header.stamp = ros::Time::now();
```
Note that timestamps accuracy may be lost with this approach. We could also find the time offset between the first cloud received and the current ROS time, then apply the offset to all subsequents node, that way the delay will still be consistent between the scans.

If you want to use the os1 IMU, you would have to do something similar too (offsetting the stamps to match current ROS time).

cheers,
Mathieu

Lily-XueCY, 2019-07-01:
Hi @matlabbe, thank you, it works now.

closed 2019-07-01 by Lily-XueCY

Samehibrahim51, 2023-04-14:
can you share the launch file for fusion between ouster and d435i? Thank you.

Samehibrahim51, 2023-04-20:
> Hi @matlabbe, @Lily-XueCY , @Samehibrahim51 
I tried your launch file with ouster lidar 16 beam and D435i as follow:

Launching the ouster lidar: roslaunch ouster_ros os1.launch sensor_hostname:=192.168.11.108

Launching the D435i:
roslaunch realsense2_camera rs_camera.launch     align_depth:=true     unite_imu_method:="linear_interpolation"     enable_gyro:=true      enable_accel:=true

Launching D435i imu filter:
rosrun imu_filter_madgwick imu_filter_node     _use_mag:=false     _publish_tf:=false     _world_frame:="enu"     /imu/data_raw:=/camera/imu     /imu/data:=/rtabmap/imu

Then, launching your lanch file but I changed the scan topic to be /ouster/points as it is my point cloud as:
roslaunch rtabmap_ros test_ouster_gen2_d435i.launch

I got a .db file with high size but there is no map on the map viewer when using:
rtabmap rtabmao.db

and the errors or the warnings are: 
warning: [ WARN] [1681939077.650004804]: Could not get transform from base_link to camera_color_optical_frame after 0.100000 seconds (for stamp=1681939077.334201)! Error="canTransform: target_frame base_link does not exist.. canTransform returned after 0.100378 timeout was 0.1.".
[ WARN] [1681937643.038526058]: /rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap subscribed to (approx sync):
/odom
/rgbd_image
/ouster/points

matlabbe, 2023-04-21:
`Could not get transform from base_link to camera_color_optical_frame after 0.100000 seconds (for stamp=1681939077.334201)! Error="canTransform: target_frame base_link does not exist.. canTransform returned after 0.100378 timeout was 0.1.".`
Looks like your TF tree is not setup correctly...

Samehibrahim51, 2023-04-21:
> `Could not get transform from base_link to camera_color_optical_frame after 0.100000 seconds (for stamp=1681939077.334201)! Error="canTransform: target_frame base_link does not exist.. canTransform returned after 0.100378 timeout was 0.1.".` Looks like your TF tree is not setup correctly...

I can try to see the tree again, but is this launch file is good to obtain a .db file map for both ouster lidar and D435i? and is there any rtabmap launch file already exist for both of them?