introlab/rtabmap_ros #324: what we should do about the global/local loop closure detection in rtabmap_ros??

ZuDame, 2019-07-17:
Hello,
I recently researched the rtabmap_ros code packages, including some parameter adjustments which learn from RTABMAP. But when I use rtabmap_ros to do real-time map construction, there is a serious problem. I don't know whether my parameters are wrong or where the algorithm itself is.
  The problem is: I am building a map in real time, such as an office-sized environment and have built a large part of the area (until no problems), but when I return to the previous location, the algorithm will perform closed-loop detection, which is fine. However,  the other maps that have already been created will have some gaps and loopholes on the ground. I am really confused about that I can't sovle this, Is it a bug?or a closed-loop detection problem?

ZuDame, 2019-07-19:
like this picture had show, The global map is complete and continuous at first, but this happens when a closed loop or dynamic obstacle update is detected, and the place that should not be removed is also removed (the area drawn by the circle)

ZuDame, 2019-07-19:
[image]

ZuDame, 2019-07-19:
[image]

ZuDame, 2019-07-19:
I find where the problem it is, if I hold the whole map complete when gets the closed loop, on the same time I can't remove the dynamic obstracles. And if I want to remove the dynamic obstracles, I won't get the complete map!!.  Is it a bug or not?

matlabbe, 2019-07-19:
`map_filter_radius` parameter doesn't remove dynamic objects, it just hide some close nodes. Can you show the map if you remove that parameter?

ZuDame, 2019-07-20:
Sure, When I remove the parameter, it's just like this. 
 [image]

ZuDame, 2019-07-20:
> `map_filter_radius` parameter doesn't remove dynamic objects, it just hides some close nodes for **visualization-only**. Can you show the map if you remove that parameter?

[image]

ZuDame, 2019-07-20:
Sure, When I remove the parameter, It just like this:
[image]

ZuDame, 2019-07-20:
By the way, I use the Depth Camera to do VSLAM. And I hope that while acquiring the global map, I can refresh the dynamic obstacles encountered on the way and eliminate its movement track. Below are the parameters I set, I hope that can help you to analysis the problem.

`<node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start">
              <param name="frame_id" type="string" value="$(arg frame_id)"/>
              <param name="odom_frame_id" type="string" value="T265_odom_frame"/>

              <param name="Grid/CellSize" value="$(arg Grid/CellSize)"/>
              <param name="subscribe_depth" type="bool" value="false"/>
              <param name="subscribe_rgbd"  type="bool" value="true"/>

              <remap from="odom"                    to="/T265/odom/sample"/>
              <remap from="rgbd_image"              to="rgbd_image"/>
              <param name="queue_size"              type="int"      value="10"/>
              <param name="Rtabmap/DetectionRate"   type="string"   value="5"/>
 
              <param name="Grid/RangeMax"        value="3" />
              <param name="Grid/RangeMin"        value="0" />
     
              <param name="cloud_output_voxelized"    type="string"    value="true" />   
              <param name="decimation"                value="4" />

              
              <param name="map_filter_radius"   value="2.0" />
              <param name="map_filter_angle"   value="30" />

              
              <param name="Optimizer/Strategy"      type="string"   value="1"/> 
              <param name="Optimizer/Robust"        type="string"   value="false"/>
              <param name="RGBD/OptimizeMaxError"   type="string"   value="0.1"/> 

              <param name="Icp/PointToPlane"                              value="false" />  
              <param name="Icp/MaxCorrespondenceDistance"                 value="0.05" />
              <param name="RGBD/NeighborLinkRefining"                     value="false" />

              <param name="GridGlobal/FullUpdate"           value="false"   />
              <param name="GridGlobal/FootprintRadius"      value="1.0"     />
              
        </node>`

matlabbe, 2019-07-23:
Hi,

Normally to clear dynamic obstacles without a laser scan input is to set `Grid/RayTracing` parameter to `true`. If you need only a 2D occupancy grid map, another optimization is to set `Grid/3D` to `false`. See Figures 14-15-16 of this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf) for examples of different approaches to create the map.

What `map_filter_radius` and `map_filter_angle` are doing is if the robot traverses again the same area, so that multiple nodes are overlapping, only most recent node (in that radius and orientation) will be use to assemble the map. It will indirectly remove obstacles of a previous node when navigating at the same location again (with obstacle gone on second traversal). It will use indeed a lot less computing ressources than doing ray tracing, but if the robot is looking at the area where there was an obstacle from another point of view (so the nodes in the map are not overlapping), the obstacle won't be cleared. 

cheers,
Mathieu

ZuDame, 2019-07-24:
Hi, Thanks for your reply,
I have tried your parameter: change `Grid/RayTracing` into `true`, and comment the parameter: `map_filter_radius`, `map_filter_angle`.  However the 3D map won't remove the dynamic obstacles.
[image]

ZuDame, 2019-07-24:
When the person leaves, the map still retains the shape of the person.
[image]

matlabbe, 2019-07-24:
By default, rtabmap updates the map only if we move. To make always update the map, try setting `map_always_update` to true (for `rtabmap` node). See also https://github.com/introlab/rtabmap_ros/issues/159#issuecomment-288797743, it is more related to OctoMap, but similar issue can happen if the camera cannot see a wall behind after the obstacle disappeared.

Also, make sure you are looking at the `/rtabmap/cloud_map` topic, not `/rtabmap/mapData` with MapCloud plugin. MapCloud will always show everything. 

cheers,
Mathieu

ZuDame, 2019-07-24:
it Still not working, after I done what you have said.....

[image]

matlabbe, 2019-07-24:
Can you share this database?

matlabbe, 2019-07-24:
A no sorry, you should use `octomap_cloud` if you are in 3D. You may also compare with 2D `grid_map`.

ZuDame, 2019-07-24:
The `octomap_cloud` Can't solve the problem, this is my database:
https://drive.google.com/open?id=1kylLD5ibB1Et0ZmB2xmu_LFoH_u4S74e

matlabbe, 2019-07-24:
Thx for the database. There are some problems that have solutions :) You are using version 0.17.6, there was a [commit](https://github.com/introlab/rtabmap/commit/de32e53868e66c1aacd4e15ec9594c6a44c0ee3e) at the end of last year about an issue of too sparse empty space using the OctoMap. With latest rtabmap version, the empty space is a lot more dense, so obstacles are more easily cleared. Here is the octomap_cloud when regenerating all local occupancy grids with latest version (the legs are cleared):
[image]

After increasing `GridGlobal/OccupancyThr` parameter from 0.5 to 0.85, the black box can be also cleared:
[image]

So I suggest at least that you get the latest rtabmap version (>=0.19.3), either by using ROS melodic binaries or rebuilding rtabmap [from source](https://github.com/introlab/rtabmap_ros#build-from-source) (kinetic binaries will be upgraded to 0.19.3 soon).

ZuDame, 2019-07-25:
Thanks you for your patiences, I am trying what you have said that rebuilding rtabmap from source, But I don't want to upgrade ROS version( my ROS version ROS-kinetic ), if that, What version `rtabmap_ros` should I use? Because when I keep the `rtabmap_ros` version (0.17.6-kinetic), there is some erros about "has no member named... ", Just like this 

`
/home/ian/catkin_ws/src/rtabmap_ros/src/CoreWrapper.cpp: In member function ‘void rtabmap_ros::CoreWrapper::publishStats(const ros::Time&)’:
/home/ian/catkin_ws/src/rtabmap_ros/src/CoreWrapper.cpp:2604:10: error: ‘const class rtabmap::Statistics’ has no member named ‘getSignatures’
    stats.getSignatures(),
          ^
/home/ian/catkin_ws/src/rtabmap_ros/src/CoreWrapper.cpp:2630:36: error: ‘const class rtabmap::Statistics’ has no member named ‘getSignatures’
   if(stats.poses().size() && stats.getSignatures().size())
                                    ^
/home/ian/catkin_ws/src/rtabmap_ros/src/CoreWrapper.cpp:2639:60: error: ‘const class rtabmap::Statistics’ has no member named ‘getSignatures’
    for(std::map<int, Signature>::const_iterator iter=stats.getSignatures().begin();
                                                            ^
/home/ian/catkin_ws/src/rtabmap_ros/src/CoreWrapper.cpp:2640:17: error: ‘const class rtabmap::Statistics’ has no member named ‘getSignatures’
     iter!=stats.getSignatures().end();
                 ^

`

matlabbe, 2019-07-25:
Like explained [here](https://github.com/introlab/rtabmap_ros#build-from-source), `ros-kinetic-rtabmap` and `ros-kinetic-rtabmap-ros` should be uninstalled: 
```bash
$ sudo apt-get remove ros-kinetic-rtabmap ros-kinetic-rtabmap-ros
```

If you already built/install last version of `rtabmap`, you only need to to clone `rtabmap_ros` in your `catkin_ws` and do `catkin_make`. If you use default master branch from rtabmap, use also default master branch from rtabmap_ros repo.

cheers,
Mathieu

ZuDame, 2019-07-26:
Thx very much, I have upgrade my `rtabmap` library to 0.19.3 from resource, and keep the `rtabmap_ros` version is the latest version. Finally, It solved my problem !  I hope that your `rtabmap library ` will be better and better. After researched, I found the rtabmap is really really worthy to study.

ZuDame, 2019-08-08:
Hello,sir,  I have been learning the `rtabmap_ros` packages recently, When I run and return in a large-scale scene, the map can't be closed, and the first and the last can't be connected together. How can I solve this problem?

ZuDame, 2019-08-08:
@matlabbe

ZuDame, 2019-08-08:
Just Like this:

[image]
[image]

matlabbe, 2019-08-16:
A loop closure hasn't being found. It is difficult to see why with only screenshots. It could be that the loop closure hasn't being detected (is the robot navigating the same area with camera in the same direction? it won't detect loop closures visually if the camera is not looking at the same direction), or it was rejected for some reasons (that would appear on the terminal).