introlab/rtabmap_ros #326: /usr/bin/ld: cannot find -lrealsense2

ankitgc1, 2019-07-17:
error in make

[image]

matlabbe, 2019-07-19:
Where realsense2 library is installed?

Example:
```bash
$ ldd rtabmap | grep realsense2
	librealsense2.so.2.22 => /usr/local/lib/librealsense2.so.2.22 (0x00007f971a791000)
```

matlabbe, 2019-07-26:
Should be fixed now with the commit above. The bug appeared only with realsense2 binaries, from source it was ok.

closed 2019-07-26 by matlabbe

themancalledjakob, 2020-02-26:
should you encounter this issue and you are on linux, look for broken symlinks.
`ls -la /usr/lib/x86_64-linux-gnu/`
the broken link would be for example `/usr/lib/x86_64-linux-gnu/librealsense2.so -> librealsense2.so.2.33` in red
just remove the symlink librealsense2.so like this: `sudo rm /usr/lib/x86_64-linux-gnu/librealsense2.so`
then:
`sudo ln -s /usr/lib/x86_64-linux-gnu/librealsense2.so.2.32 /usr/lib/x86_64-linux-gnu/librealsense2.so`

Harimus, 2020-02-28:
@themancalledjakob  
This solved it for me, thanks! Do you know why this broken symlinks could've happened? on my PC the only thing I can think of is different cmake version.

themancalledjakob, 2020-05-27:
@Harimus 
My guess is that a realsense developer pushed a repo with a wrong symlink by accident ;-)
This stuff easily happens if you link to a not-yet-published version (in this case `librealsense2.so.2.33`) on your disk, and therefore don't notice that it's broken for everyone else. This is only an assumption though, I actually have no idea.