introlab/rtabmap_ros #327: D435 cannot create map in cloud_map topic, but mapData is working well

guitarlovelove, 2019-07-19:
Hello.
I have used RTABMAP and RTABMAP_ROS for 3 monthes.
And this problem suddenly happened without any source code changes.

I am using RealSenseD435 and reading cloud_map topic.
Visualization(mapData topic) is surely working well, however, cloud_map is colored red and green, and the shape of point cloud is different from mapData or real world completely.
The cloud_map is consisted by some planes. Please look below screenshot.

[image]

Does someone know what is happening in my RTabMap?
I really need your help...
Thank you for reading.

matlabbe, 2019-07-19:
Is `cloud_map` appears on a plane? If so, the local occupancy grid may be 2D, did you have set `Grid/3D` parameter to false?

guitarlovelove, 2019-07-22:
Thank you very much for replying!!

The detail of this symptom is,  the PointCloud is compressed into a camera's x-y plane in each map update.
So there are various orientation planes after capturing surroundings.

Actually, the parameter "Grid/3D" seems to be true.

The problem is rtabmap_ros/src/MapsManagere.cpp around l:990~l:998 and l:1036~l:1044, especially "util3d::laserScanToPointCloudRGB()" functions, I detected.
I don't understand the reason, but the RealSenseD435 is recognized as LiDAR and above bifurcations must be used. 

The only change when I met this problem was RealSense Driver version update to ver.2.24 by apt.

Because I found the MapsManager.cpp's problem, I will continue to trying to solve this problem on this week. 
If you tell me something about this problem again, I will be glad.
Thank you for reading my poor English.

matlabbe, 2019-07-23:
Can you share the database?

guitarlovelove, 2019-09-02:
Sorry for my late reply, but I eventually fixed this problem by myself!!
I don't know why, but the RealSense2 driver seemed bad.
I fixed this problem by re-install RealSense2 driver with "apt purge ===some packages related to realsense2== " and "apt install them".
Thank you for helping, and sorry for being late to notice it.