introlab/rtabmap_ros #328: What's wrong when I'm using localization mode?

WLwind, 2019-07-20:
I set Mem/IncrementalMemory to false, and Mem/InitWMWithAllNodes to localization, and did not use --delete_db_on_start.
And it come to a lot of this:
`[ERROR] (2019-07-21 00:05:08.969) VWDictionary.cpp:616::addWordRef() Not found word 319476 (dict size=4355)
`
Do anyone know how to solve this?

WLwind, 2019-07-20:
I think maybe the reason is that my rtabmap.db is too large(496MB). I have tried a smaller one (220MB) later and it worked well.☺

matlabbe, 2019-07-23:
It is not because the database is too large, I got some databases over 1-4 GB easily. The database may be corrupted somehow. Try to do a recovery:
```bash
$ rtabmap-recovery my_map.db
```

WLwind, 2019-07-26:
> It is not because the database is too large, I got some databases over 1-4 GB easily. The database may be corrupted somehow. Try to do a recovery:
> 
> ```shell
> $ rtabmap-recovery my_map.db
> ```

Thanks a lot! It worked! It fixed my map well.

closed 2019-07-26 by WLwind