introlab/rtabmap_ros #329: Problem of adding buttons under rtabmap-ros gui to realize functions

TomLuosy, 2019-07-24:
I want to add some buttons under the rtabmap-ros interface to simplify the process of repeating the input commands from the terminal.
For example, instructions for transferring data：roslaunch realsense2_camera rs_camera.launch
I write the function realization of the button in rtabmap independent library, and open the terminal input instructions through Qprocess. But under the rtabmap-ros gui, it doesn't work.
Here is my slot function code. How can I solve this problem? Thank you.
`void MainWindow::startTest()
{
        UDEBUG("");
        QProcess * po = new QProcess(this);
        QStringList strCMD;
        strCMD.append("/opt/ros/kinetic/bin/roslaunch");
        strCMD.append("realsense2_camera");
        strCMD.append("rs_camera.launch");
        po->start("/usr/bin/python",strCMD);
		po->waitForStarted();

}`

matlabbe, 2019-07-24:
I am not used to do that kind of stuff (starting an external process from Qt), maybe this post could help you: https://answers.ros.org/question/209090/how-to-launch-a-roslaunch-in-qt-with-a-terminal-gui/

TomLuosy, 2019-07-25:
> I am not used to do that kind of stuff (starting an external process from Qt), maybe this post could help you: https://answers.ros.org/question/209090/how-to-launch-a-roslaunch-in-qt-with-a-terminal-gui/

First of all, thank you very much for your reply.
I tried the code in the link but it didn't work.So I tried a lot of ways to find out why.Ultimately, I think it's probably because the namespaces of nodes are different when using Qprocess.

Here is my node list for data transmission when using terminal command：
henry@henry-ThinkPad-S5:~$ rosnode list
/camera/realsense2_camera
/camera/realsense2_camera_manager
/rosout
/rtabmap/rgbd_odometry
/rtabmap/rtabmap
/rtabmap/rtabmapviz

Here I use Qprocess in the button to transmit data：
henry@henry-ThinkPad-S5:~$ rosnode list
/rosout
**/rtabmap**/camera/realsense2_camera
**/rtabmap**/camera/realsense2_camera_manager
/rtabmap/rgbd_odometry
/rtabmap/rtabmap
/rtabmap/rtabmapviz

I think that's why Qprocess doesn't work properly.So where to find the namespace address of the node in the file to solve this kind of problem? Thank you.

matlabbe, 2019-07-26:
If you use [rtabmap.launch](https://github.com/introlab/rtabmap_ros/blob/ca42a35709782504eeeca9b4a6b2269dd72c9097/launch/rtabmap.launch#L41), you could launch it with `namespace:=""` to clear the namespace argument.

cheers,
Mathieu

TomLuosy, 2019-07-27:
Thank you very much for your help. My problem has been solved!Thanks again!

po->start("gnome-terminal -x sh -c \"export ROS_NAMESPACE=\"\"\nroslaunch 
realsense2_camera rs_camera.launch\n; bash\"");

closed 2019-07-27 by TomLuosy