introlab/rtabmap_ros #331: how to do robot navigation with RTAB-Map and vins-fusion

teddyluo, 2019-07-30:
Hi @matlabbe, 
   I found that RTAB-Map has been supported vins-fusion Odometry since the branch: c0a7c3a, as the screenshot in the following:

[image]
 
  This is an exciting job. Now my problem is straightforward: I want to use vins-fusion as the odometry, and let the robot autonomously navigate in a room. The effect is similar to the one you wrote:

http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot#If_you_don.27t_have_the_robot%EF%BC%89

In a word, what I want to do is "Autonomous Navigation with Turtlebot 2e and a stereo camera", with the odometry being "VINS-FUSION". 

Can you give some examples/guidelines on how to achieve this goal?

Many thanks.

matlabbe, 2019-07-30:
For 3D stereo SLAM with VINS-Fusion in RTAB-Map, see EuRoC example here (we assume with this example RTAB-Map library has been built with VINS-Fusion support, the cmake info should show `With VINS-Fusion = YES`): https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/euroc_datasets.launch

There is an [example](https://github.com/introlab/rtabmap_ros/blob/ca42a35709782504eeeca9b4a6b2269dd72c9097/launch/tests/euroc_datasets.launch#L25-L31) of VINS-Fusion stereo-only and also with IMU.

Another approach is to use VINS-Fusion ros node to provide odometry input to rtabmap node, though not sure if the original package can transform VIO in `base_link` frame instead of camera frame (like the integration in RTAB-Map does). Not sure also if we can force 2D mapping like in RTAB-Map (for applications like Turtlebot).

If you have left/right images, left/right camera_info and imu topic names, I could try to make an example based on the turtlebot launch file.

teddyluo, 2019-07-30:
hi,
   many thanks for the reply. in fact, we do have left and right images, along with left/right camera_info and imu topic. can you make one example for us as a reference? thanks.

matlabbe, 2019-07-30:
There are a couple of things that are different. On turtlebot, the depth image from the kinect is converted to a laser scan for obstacle avoidance in move_base. With stereo, we cannot configure that, unless [stereo_image_proc](http://wiki.ros.org/stereo_image_proc) is used to compute the disparity image (can your camera output a disparity image?), then convert it to a depth image ([rtabmap_ros/disparity_to_depth](http://wiki.ros.org/rtabmap_ros/#rtabmap_ros.2BAC8-disparity_to_depth) nodelet). You may try at least if the [hand-held stereo mapping example](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping) is working (adjust topics and `frame_id`):

```bash
$ roslaunch rtabmap_ros rtabmap.launch \
   args:="-d" \
   left_image_topic:=/camera/left/image_rect \
   right_image_topic:=/camera/right/image_rect \
   left_camera_info_topic:=/camera/left/camera_info \
   right_camera_info_topic:=/camera/right/camera_info \
   stereo:=true \
   frame_id:=base_link
```

Adding IMU and using VINS-Fusion as odometry:
```bash
$ roslaunch rtabmap_ros rtabmap.launch \
   args:="-d --Odom/Strategy 9 --OdomVINS/ConfigPath ~/my_camera_vins_config.yaml" \
   left_image_topic:=/camera/left/image_rect \
   right_image_topic:=/camera/right/image_rect \
   left_camera_info_topic:=/camera/left/camera_info \
   right_camera_info_topic:=/camera/right/camera_info \
   stereo:=true \
   frame_id:=base_link \
   imu_topic:="/imu/data" \
   wait_imu_to_init:="true"
```

For the navigation part, you may refer to this tutorial: http://wiki.ros.org/rtabmap_ros/Tutorials/StereoOutdoorNavigation, the main thing to do is to connect `/rtabmap/grid_map` output topic from rtabmap to `move_base` map input topic (on its global costmap static layer).

cheers,
Mathieu

paulyang1990, 2019-12-09:
Hi I would like to ask how to build rtabmap with VINS support. I enabled cmake with VINS flag by sourcing my ros workspace contains VINS, but during building of rtabp map. the complier kept telling me it cannot locate header files of VINS. Should any other paths or flags need to be set?

matlabbe, 2020-01-07:
What are the errors? Maybe VINS headers have changed...

getupcurry, 2020-05-05:
Hi, @matlabbe , as you said

> For 3D stereo SLAM with VINS-Fusion in RTAB-Map, see EuRoC example here (we assume with this example RTAB-Map library has been built with VINS-Fusion support, the cmake info should show With VINS-Fusion = YES)

Is it ok to use vio to do 2d navigation with a turtlebot by setting paramter "Reg/Force3DoF" to true?

cheers,
matlabbe

matlabbe, 2020-05-25:
With VIO, the `yaw` would be always aligned with gravity (along z axis pointing up), so it could work with `Reg/Force3DoF` to true.

getupcurry, 2020-06-27:
Hi, @matlabbe ,
According to your demo of test_d435i_vio.launch, I wrote a launch file of mynteye, the file as follows:

<launch>
<arg name="database_path"     default="rtabmap.db"/>
<arg name="rtabmapviz"        default="false"/>
<arg name="localization"      default="false"/>
<arg     if="$(arg localization)" name="args"  default=""/>
<arg unless="$(arg localization)" name="args"  default="--delete_db_on_start"/>
<arg name="depth_mode" default="true"/>

<arg name="left_image_topic"        value="/mynteye/left_rect/image_rect"/>
<arg name="right_image_topic"       value="/mynteye/right_rect/image_rect"/>
<arg name="left_camera_info_topic"  value="/mynteye/left_rect/camera_info"/>
<arg name="right_camera_info_topic" value="/mynteye/right_rect/camera_info"/>

<arg name="wait_for_transform"  default="0.2"/> 

<node pkg="imu_filter_madgwick" type="imu_filter_node" name="imu_filter_node">
   <param name="use_mag"       value="false"/>
   <param name="publish_tf"    value="false"/>
   <param name="world_frame"   value="enu"/>
   <remap from="/imu/data_raw" to="/mynteye/imu/data_raw"/>
   <remap from="/imu/data"     to="/rtabmap/imu"/>
</node>

<node pkg="topic_tools" type="throttle" name="imu_throttle" args="messages /rtabmap/imu 100.0"/>

<group ns="rtabmap">
<node if="$(arg depth_mode)" pkg="rtabmap_ros" type="stereo_odometry" name="stereo_odometry" args="--Optimizer/GravitySigma 0.3 --Reg/Force3DoF true --Odom/Strategy 9 --OdomVINS/ConfigPath /home/yu/vins-fusion-mynt/src/MYNT-EYE-VINS-FUSION-Samples/config/mynteye-s/mynt_stereo_imu_config.yaml --Rtabmap/ImagesAlreadyRectified false --Mem/UseOdomGravity true" output="screen">
   <remap from="left/image_rect"   to="/mynteye/left_rect/image_rect"/>
   <remap from="right/image_rect"  to="/mynteye/right_rect/image_rect"/>
   <remap from="left/camera_info"  to="/mynteye/left_rect/camera_info"/>
   <remap from="right/camera_info" to="/mynteye/right_rect/camera_info"/>
   <remap from="imu"               to="/rtabmap/imu_throttle"/>
   <param name="frame_id"          value="base_link"/>
   <param name="wait_imu_to_init"  value="true"/>
</node>
</group>

<include if="$(arg depth_mode)" file="$(find rtabmap_ros)/launch/rtabmap.launch">
   <arg name="rtabmap_args"      value="--delete_db_on_start --Optimizer/GravitySigma 0.3"/>
   <arg name="rgb_topic"         value="/mynteye/left_rect/image_rect"/>
   <arg name="depth_topic"       value="/mynteye/depth/image_raw"/>
   <arg name="camera_info_topic" value="/mynteye/left_rect/camera_info"/>
   <arg name="visual_odometry"   value="false"/>
   <arg name="frame_id"          value="base_link"/>
   <arg name="imu_topic"         value="/rtabmap/imu_throttle"/>
   <arg name="rtabmapviz"        value="$(arg rtabmapviz)"/>
   <arg name="rviz"              value="true"/>
</include>
</launch>

In this launch file, I add "--Reg/Force3DoF true", but the odometry is still 6DoF.  What should I do to set VIO as 3DoF?

Thanks.

getupcurry, 2020-06-27:
Hi, @matlabbe , According to your demo of test_d435i_vio.launch, I wrote a launch file of mynteye, the file as follows:

```
<launch>
<arg name="database_path"     default="rtabmap.db"/>
<arg name="rtabmapviz"        default="false"/>
<arg name="localization"      default="false"/>
<arg     if="$(arg localization)" name="args"  default=""/>
<arg unless="$(arg localization)" name="args"  default="--delete_db_on_start"/>
<arg name="depth_mode" default="true"/>

<arg name="left_image_topic"        value="/mynteye/left_rect/image_rect"/>
<arg name="right_image_topic"       value="/mynteye/right_rect/image_rect"/>
<arg name="left_camera_info_topic"  value="/mynteye/left_rect/camera_info"/>
<arg name="right_camera_info_topic" value="/mynteye/right_rect/camera_info"/>

<arg name="wait_for_transform"  default="0.2"/> 

<node pkg="imu_filter_madgwick" type="imu_filter_node" name="imu_filter_node">
   <param name="use_mag"       value="false"/>
   <param name="publish_tf"    value="false"/>
   <param name="world_frame"   value="enu"/>
   <remap from="/imu/data_raw" to="/mynteye/imu/data_raw"/>
   <remap from="/imu/data"     to="/rtabmap/imu"/>
</node>

<node pkg="topic_tools" type="throttle" name="imu_throttle" args="messages /rtabmap/imu 100.0"/>

<!-- RTAB-Map: depth mode -->
<!-- We have to launch stereo_odometry externally from rtabmap.launch so that rtabmap can use RGB-D input -->

<group ns="rtabmap">
<node if="$(arg depth_mode)" pkg="rtabmap_ros" type="stereo_odometry" name="stereo_odometry" args="--Optimizer/GravitySigma 0.3 --Reg/Force3DoF true --Odom/Strategy 9 --OdomVINS/ConfigPath /home/yu/vins-fusion-mynt/src/MYNT-EYE-VINS-FUSION-Samples/config/mynteye-s/mynt_stereo_imu_config.yaml --Rtabmap/ImagesAlreadyRectified false --Mem/UseOdomGravity true" output="screen">
   <remap from="left/image_rect"   to="/mynteye/left_rect/image_rect"/>
   <remap from="right/image_rect"  to="/mynteye/right_rect/image_rect"/>
   <remap from="left/camera_info"  to="/mynteye/left_rect/camera_info"/>
   <remap from="right/camera_info" to="/mynteye/right_rect/camera_info"/>
   <remap from="imu"               to="/rtabmap/imu_throttle"/>
   <param name="frame_id"          value="base_link"/>
   <param name="wait_imu_to_init"  value="true"/>
</node>
</group>

<include if="$(arg depth_mode)" file="$(find rtabmap_ros)/launch/rtabmap.launch">
   <arg name="rtabmap_args"      value="--delete_db_on_start --Optimizer/GravitySigma 0.3"/>
   <arg name="rgb_topic"         value="/mynteye/left_rect/image_rect"/>
   <arg name="depth_topic"       value="/mynteye/depth/image_raw"/>
   <arg name="camera_info_topic" value="/mynteye/left_rect/camera_info"/>
   <arg name="visual_odometry"   value="false"/>
   <arg name="frame_id"          value="base_link"/>
   <arg name="imu_topic"         value="/rtabmap/imu_throttle"/>
   <arg name="rtabmapviz"        value="$(arg rtabmapviz)"/>
   <arg name="rviz"              value="true"/>
</include>
</launch>
```

In this launch file, I add "--Reg/Force3DoF true", but the odometry is still 6DoF.  What should I do to set VIO as 3DoF?

Thanks.

matlabbe, 2020-07-09:
This has been fixed in https://github.com/introlab/rtabmap/commit/4a09c4bdcf68491cd7941fa28d5c8db0bfcf729d

getupcurry, 2020-07-18:
@matlabbe  Hi, thank you so much to help me. 
I have tested the `euroc-datasets.launch`  demo with VINS(VIO), and I found that the odometry is stable at beginning, and after a few seconds, it drift so much. Can you explain it?
Thanks.

matlabbe, 2020-07-30:
There are some synchronization issues with the bags, sometimes image and imu messages are not arriving correctly based on their stamps. I did some refactoring on rtabmap side to better re-order imu and image topics. See the commit above. There are still some drifts I cannot explain on MH_01_easy.bag at the beginning, but if you start after 25 seconds, the mapping seems ok. V1_01_easy.bag should be ok from the start, minor some lags caused by synchronization of ros topics.

```bash
$ roslaunch rtabmap_ros euroc_datasets.launch args:="Odom/Strategy 9 OdomVINS/ConfigPath ~/catkin_ws/src/VINS-Fusion/config/euroc/euroc_stereo_imu_config.yaml" MH_seq:=true raw_images_for_odom:=true
$ rosbag play --clock -s 25 MH_01_easy.bag
```

You can also try with `raw_images_for_odom:=false`.

sssolitude, 2020-08-06:
> Hi I would like to ask how to build rtabmap with VINS support. I enabled cmake with VINS flag by sourcing my ros workspace contains VINS, but during building of rtabp map. the complier kept telling me it cannot locate header files of VINS. Should any other paths or flags need to be set?

https://gist.github.com/matlabbe/795ab37067367dca58bbadd8201d986c this patch might can solve your problem.

sssolitude, 2020-08-06:
@matlabbe Could you please explain what the arg raw_image_for_odom means. I am trying to  use a third-party camera module which can output stereo images and imu message to run  rtabmap with vins-fusion odometry.

getupcurry, 2020-08-31:
@matlabbe  Thank you very much. But I have to bother you again, there are still some issues:
I have updated the rtabmap and rtabmap_ros to the latest version, and there is no problem when I run the demo `euroc-datasets.launch` with VINS(VIO),  but when I test with my MYNTEYE camera, it still has a drift (note: when I only run VINS-Fusion with MYNTEYE, the odometry is stable and it does not drift), do you know why? And my launch file as follows:
```
<launch>
   
   <!-- Choose visualization -->
   <arg name="rviz" default="true" />
   <arg name="rtabmapviz" default="false" />
   <arg name="stereo_sync" default="false" />
   <arg name="localization"      default="false"/>
   <arg     if="$(arg localization)" name="args"  default=""/>
   <arg unless="$(arg localization)" name="args"  default="--delete_db_on_start"/>
  
   <include file="$(find turtlebot_navigation)/launch/includes/move_base_orb.launch.xml"/>

   <node pkg="imu_filter_madgwick" type="imu_filter_node" name="imu_filter_node">
      <remap from="/imu/data_raw" to="/mynteye/imu/data_raw"/>
      <remap from="/imu/data"     to="/rtabmap/imu"/>
      <param name="use_mag" value="false"/>
      <param name="world_frame" value="enu"/>
      <param name="publish_tf" value="false"/>
   </node> 
       
   <!--node pkg="topic_tools" type="throttle" name="imu_throttle" args="messages /rtabmap/imu 100.0"/-->
    
   <group ns="rtabmap">   
   
      <!-- Stereo Odometry -->   
      <node pkg="rtabmap_ros" type="stereo_odometry" name="stereo_odometry" output="screen">
         <remap from="left/image_rect"       to="/mynteye/left_rect/image_rect"/>
         <remap from="right/image_rect"      to="/mynteye/right_rect/image_rect"/>
         <remap from="left/camera_info"      to="/mynteye/left_rect/camera_info"/>
         <remap from="right/camera_info"     to="/mynteye/right_rect/camera_info"/>
         <remap from="odom"                  to="/odom"/>

         <param name="subscribe_rgbd"  type="bool" value="$(arg stereo_sync)"/>
         <param name="frame_id"        type="string" value="base_footprint"/>
         <!--param name="odom_frame_id"   type="string" value="odom"/-->

         <param name="Odom/Strategy"      type="string" value="9"/> <!-- 0=Frame-to-Map, 1=Frame=to=Frame -->
         <param name="OdomVINS/ConfigPath"        type="string" value="/home/yu/vins_fusion/src/VINS-Fusion/config/mynteye-s/test.yaml"/>
	 <param name="Reg/Force3DoF"                type="string" value="true"/>
         <param name="Vis/EstimationType" type="string" value="1"/> <!-- 0=3D->3D 1=3D->2D (PnP) -->
         <param name="Vis/MaxDepth"       type="string" value="0"/>
         <param name="Odom/GuessMotion" type="string" value="true"/>
         <param name="Vis/MinInliers"     type="string" value="15"/>
         <param name="GFTT/MinDistance"   type="string" value="10"/>
         <param name="GFTT/QualityLevel"  type="string" value="0.00001"/>
         <param name="Vis/FeatureType"   type="string" value="6"/>
         <param name="Kp/DetectorStrategy"  type="string" value="6"/>
         <param name="imu_topic"          type="string" value="/rtabmap/imu"/>
         <param name="wait_imu_to_init"   type="string" value="true"/>

      </node>
   
      <!-- Visual SLAM: args: "delete_db_on_start" and "udebug" -->
      <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)">
         <param name="frame_id"         type="string" value="base_footprint"/>
         <param unless="$(arg stereo_sync)" name="subscribe_stereo" type="bool" value="true"/>
         <param name="subscribe_depth"  type="bool" value="false"/>
         <param name="subscribe_rgbd"      type="bool" value="$(arg stereo_sync)"/>
	 <param name="Reg/Force3DoF"                type="string" value="true"/>

         <remap from="left/image_rect"   to="/mynteye/left_rect/image_rect"/>
         <remap from="right/image_rect"  to="/mynteye/right_rect/image_rect"/>
         <remap from="left/camera_info"  to="/mynteye/left_rect/camera_info"/>
         <remap from="right/camera_info" to="/mynteye/right_rect/camera_info"/>

         <remap from="odom" to="/odom"/>
	 <param name="use_action_for_goal" type="bool" value="true"/>
[... 64 lines cut ...]
      <node if="$(arg rtabmapviz)" pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz" args="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" output="screen">
         <param unless="$(arg stereo_sync)" name="subscribe_stereo"    type="bool" value="true"/>
         <param name="subscribe_odom_info" type="bool" value="true"/>
         <param name="subscribe_rgbd"      type="bool" value="$(arg stereo_sync)"/>
         <param name="queue_size"          type="int" value="10"/>
         <param name="frame_id"            type="string" value="base_footprint"/>
         
         <remap from="left/image_rect"  to="/mynteye/left_rect/image_rect"/>
         <remap from="right/image_rect"  to="/mynteye/right_rect/image_rect"/>
         <remap from="left/camera_info"  to="/mynteye/left_rect/camera_info"/>
         <remap from="right/camera_info" to="/mynteye/right_rect/camera_info"/>
         <remap from="odom_info"         to="odom_info"/>
         <remap from="odom"              to="/odom"/>
         <remap from="mapData"           to="mapData"/>
      </node>
         
   </group>
  
   <!-- Visualisation RVIZ --> 
   <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(find rtabmap_ros)/launch/config/demo_stereo_outdoor.rviz"/>

   <node pkg="tf" type="static_transform_publisher" name="base_link_to_mynteye" args="0 0 1 -1.5707963267948966 0 -1.5707963267948966  base_link mynteye_link 100" />

</launch>
```

matlabbe, 2020-08-31:
Can you share the config file for mynteye you are using? and a launch file for VINS-Fusion version? I could try to reproduce the setup here.

getupcurry, 2020-08-31:
Thanks for your quick answer. Do you also have a mynteye camera in your hand？ If so，I think you can replace the yaml file with yours，and check whether it also has a drift.  And the version of VINS-Fusion is the latest, I built it with rtabmap successfully according to your patch.

matlabbe, 2020-08-31:
I don't have a yaml file for vins with the mynteye camera, I never tested vins with it yet...

matlabbe, 2020-08-31:
The VINS-Fusion example doesn't work (drift a lot!) with my Mynteye S camera:
```
roslaunch mynt_eye_ros_wrapper mynteye.launch
roslaunch vins vins_rviz.launch 
rosrun vins vins_node ~/catkin_ws/src/VINS-Fusion/config/mynteye/mynteye_stereo_imu_config.yaml
```
If you have another commands that work, post it here. Which MyntEye camera do you have?

getupcurry, 2020-09-01:
@matlabbe  Thank you so much. My config file as follows:
**test.yaml:**
```
%YAML:1.1

#common parameters
#support: 1 imu 1 cam; 1 imu 2 cam: 2 cam; 
imu: 1         
num_of_cam: 2  

imu_topic: "/mynteye/imu/data_raw"
image0_topic: "/mynteye/left/image_raw"
image1_topic: "/mynteye/right/image_raw"
output_path: "/home/yu/vins_fusion"

cam0_calib: "test_left.yaml"
cam1_calib: "test_right.yaml"
image_width: 752
image_height: 480

# Extrinsic parameter between IMU and Camera.
estimate_extrinsic: 1   # 0  Have an accurate extrinsic parameters. We will trust the following imu^R_cam, imu^T_cam, don't change it.
                        # 1  Have an initial guess about extrinsic parameters. We will optimize around your initial guess.

body_T_cam0: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ -0.00053157, -0.99999379, -0.00348416, 0.00100499,
       0.99999968, -0.00053366, 0.00059814, -0.04646605,
       -0.0006, -0.00348384, 0.99999375, 0.05583352,
       0., 0., 0., 1. ]

body_T_cam1: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ -0.00243568, -0.99999702, -0.00018658, 0.00151293,
       0.9999429, -0.00243749, 0.01040466, 0.07403454,
       -0.01040508, -0.00016123, 0.99994585, 0.05529258,
       0., 0., 0., 1. ]

#Multiple thread support
multiple_thread: 1

#feature traker paprameters
max_cnt: 150            # max feature number in feature tracking
min_dist: 30            # min distance between two features 
freq: 10                # frequence (Hz) of publish tracking result. At least 10Hz for good estimation. If set 0, the frequence will be same as raw image 
F_threshold: 1.0        # ransac threshold (pixel)
show_track: 1           # publish tracking image as topic
flow_back: 1            # perform forward and backward optical flow to improve feature tracking accuracy

#optimization parameters
max_solver_time: 0.04  # max solver itration time (ms), to guarantee real time
max_num_iterations: 8   # max solver itrations, to guarantee real time
keyframe_parallax: 10.0 # keyframe selection threshold (pixel)

#imu parameters       The more accurate parameters you provide, the better performance
acc_n: 0.1          # accelerometer measurement noise standard deviation. 
gyr_n: 0.01         # gyroscope measurement noise standard deviation.     
acc_w: 0.001        # accelerometer bias random work noise standard deviation.  
gyr_w: 0.0001       # gyroscope bias random work noise standard deviation.     
g_norm: 9.81007     # gravity magnitude

#unsynchronization parameters
estimate_td: 0                      # online estimate time offset between camera and imu
td: 0                            # initial value of time offset. unit: s. readed image clock + td = real image clock (IMU clock)

#loop closure parameters
load_previous_pose_graph: 0        # load and reuse previous pose graph; load from 'pose_graph_save_path'
pose_graph_save_path: "~/output/pose_graph/" # save and load path
save_image: 1                   # save image in pose graph for visualization prupose; you can close this function by setting 0 
```
**test_left.yaml:**
```
%YAML:1.0
---
model_type: KANNALA_BRANDT
camera_name: kannala-brandt
image_width: 752
image_height: 480
projection_parameters:
   k2: -0.01422430939734004
   k3: -0.04305255589162182
   k4: 0.06924549154191210
   k5: -0.03755892784575057
   mu: 367.8182499999811057
   mv: 367.06783779055649575
   u0: 369.36111802984351016
   v0: 230.87283041165201780
```
**test_right.yaml:**
```
%YAML:1.0
---
model_type: KANNALA_BRANDT
camera_name: kannala-brandt
image_width: 752
image_height: 480
projection_parameters:
   k2: -0.02328698589036330
   k3: -0.00660646718883807
   k4: 0.01652182785268044
   k5: -0.01258362241437259
   mu: 368.69136666717554363
   mv: 368.13824142858101141
   u0: 365.16934053227640788
   v0: 233.08230102309912013
```
These are my config files of VINS-Fusion. I do not know whether it works well in your camera.  If it drift a lot, maybe you should recalibrate your MYNTEYE.  Thank you so much for solving so many problems for me.

getupcurry, 2020-09-01:
And my MYNTEYE is S1030-IR.

matlabbe, 2020-09-01:
We have the same camera. One difference I see is that you enabled `estimate_extrinsic`. I think the biggest issue is coming from imu/camera synchronization, there is always a lot of `wait for imu...` warnings when I launch VINS fusion: 
```
rosrun vins vins_node ~/catkin_ws/src/VINS-Fusion/config/mynteye/mynteye_stereo_imu_config.yaml
[ INFO] [1598978007.569093781]: init begins
config_file: /home/mathieu/catkin_ws/src/VINS-Fusion/config/mynteye/mynteye_stereo_imu_config.yaml
USE_IMU: 1
IMU_TOPIC: /mynteye/imu/data_raw
result path /home/tong/output//vio.csv
[ WARN] [1598978007.573738225]:  Optimize extrinsic param around initial guess!
camera number 2
[ INFO] [1598978007.706211535]: Synchronized sensors, fix time offset: 0
[ INFO] [1598978007.706234998]: ROW: 480 COL: 752 
 exitrinsic cam 0
 0.00428124    -0.99997  -0.0064529
   0.999969  0.00423843  0.00663446
-0.00660691  -0.0064811    0.999957
5.25834e-05  -0.0421747   0.0192387
 exitrinsic cam 1
 0.00165252   -0.999985 -0.00520307
   0.999949  0.00160076  0.00993615
-0.00992768 -0.00521922    0.999937
-0.00292519   0.0743791   0.0193027
set g   0   0 9.8
[ INFO] [1598978007.706408891]: reading paramerter of camera /home/mathieu/catkin_ws/src/VINS-Fusion/config/mynteye/left_mei.yaml
[ INFO] [1598978007.706540475]: reading paramerter of camera /home/mathieu/catkin_ws/src/VINS-Fusion/config/mynteye/right_mei.yaml
MULTIPLE_THREAD is 1
[ WARN] [1598978007.706631145]: waiting for image and imu...
init first imu pose
averge acc 9.505009 -1.207654 -1.477698
init R0 
 0.196851  0.620425  0.759159
        0  -0.77431  0.632807
 0.980433 -0.124568 -0.152423
wait for imu ... 
[ WARN] [1598978009.177034355]: gyroscope bias initial calibration -0.000124802 -2.62956e-05 -0.000217221
[ INFO] [1598978009.224925882]: Initialization finish!
time: 1598978009.169869, t: -0.000665 -0.000257 -0.002229 q: -0.260500 0.728589 0.213583 0.596390 
time: 1598978009.249849, t: -0.001597 -0.000752 -0.004902 q: -0.260719 0.728493 0.213739 0.596355 
time: 1598978009.329839, t: -0.001694 -0.000810 -0.005014 q: -0.260702 0.728537 0.213725 0.596313 
time: 1598978009.409819, t: -0.001755 -0.000876 -0.005074 q: -0.260691 0.728578 0.213717 0.596271 
time: 1598978009.489799, t: -0.001771 -0.000903 -0.005098 q: -0.260679 0.728619 0.213709 0.596230 
time: 1598978009.569779, t: -0.002233 0.001027 -0.006803 q: -0.261969 0.728236 0.214826 0.595730 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
time: 1598978009.649769, t: 0.000186 -0.000272 -0.003639 q: -0.264328 0.726812 0.217166 0.595578 
time: 1598978009.729749, t: 0.000196 -0.001327 -0.002751 q: -0.266952 0.725523 0.219574 0.595097 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
time: 1598978009.809729, t: -0.003321 -0.005946 -0.003968 q: -0.255608 0.731011 0.209377 0.597033 
time: 1598978009.889709, t: 0.009192 0.011846 -0.002178 q: -0.264213 0.726654 0.217948 0.595536 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
time: 1598978009.969699, t: 0.009783 0.012216 0.004889 q: -0.264305 0.726750 0.217985 0.595365 
time: 1598978010.049679, t: 0.008858 0.014150 0.011544 q: -0.264964 0.726927 0.218214 0.594772 
time: 1598978010.129659, t: 0.010310 0.013431 0.015126 q: -0.263210 0.727664 0.216651 0.595220 
time: 1598978010.209639, t: 0.012167 0.016279 0.016885 q: -0.263577 0.727451 0.216909 0.595225 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
```
Do you also have `wait for imu..`. about 4-5 times each time an image is processed? Maybe there is something wrong with my unit.

getupcurry, 2020-09-01:
Yes. I also have the problem. Do you think this may be caused by the camera？Have you tried with Real-sense D435i  to do the demo  d435i_vio？

matlabbe, 2020-09-01:
With the D435i, I see the same instability at the beggining when wait for imu... warnings are shown, but after a while it seems the synchronization works then the tracking is good. Here is a log (beggining it was jumping a lot, then when the warnings disappeared it was ok):
```bash
$ roslaunch realsense2_camera rs_camera.launch unite_imu_method:="copy" enable_gyro:=true enable_accel:=true enable_infra1:=true enable_infra2:=true
$ roslaunch vins vins_rviz.launch
$ rosrun vins vins_node ~/catkin_ws/src/VINS-Fusion/config/realsense_d435i/realsense_stereo_imu_config.yaml
```

```
[ INFO] [1598979579.884373548]: td 0.012851
time: 1598979579.859350, t: -0.240756 0.031692 0.319759 q: 0.602575 -0.797228 0.020766 0.029979 
wait for imu ... 
wait for imu ... 
[ INFO] [1598979579.936548971]: td 0.013630
time: 1598979579.926644, t: -0.246816 0.045388 0.375601 q: 0.602524 -0.797315 0.019188 0.029772 
wait for imu ... 
wait for imu ... 
wait for imu ... 
[ INFO] [1598979580.007545151]: td 0.007179
time: 1598979579.993945, t: -0.257573 0.026005 0.391033 q: 0.602038 -0.797657 0.018853 0.030621 
wait for imu ... 
wait for imu ... 
wait for imu ... 
[ INFO] [1598979580.073174348]: td 0.005863
time: 1598979580.061251, t: -0.280261 0.008987 0.391729 q: 0.600777 -0.798542 0.019698 0.031763 
wait for imu ... 
wait for imu ... 
[ INFO] [1598979580.143400054]: td 0.004925
time: 1598979580.128511, t: -0.276449 -0.006705 0.389935 q: 0.600183 -0.798814 0.021294 0.034970 
wait for imu ... 
[ INFO] [1598979580.196190308]: td 0.007665
time: 1598979580.195143, t: -0.275516 -0.026345 0.394509 q: 0.599086 -0.799560 0.021958 0.036302 
wait for imu ... 
wait for imu ... 
[ INFO] [1598979580.270814849]: td 0.010270
time: 1598979580.261523, t: -0.289793 -0.053224 0.392474 q: 0.597183 -0.800970 0.022793 0.036060 
wait for imu ... 
wait for imu ... 
wait for imu ... 
[ INFO] [1598979580.337557648]: td 0.006875
time: 1598979580.327897, t: -0.283215 -0.075301 0.394301 q: 0.596886 -0.801131 0.022618 0.037471 
wait for imu ... 
wait for imu ... 
[ INFO] [1598979580.397692621]: td 0.003049
time: 1598979580.394265, t: -0.277330 -0.087880 0.403611 q: 0.596508 -0.801293 0.022884 0.039799 
[ INFO] [1598979580.488543970]: td 0.003806
time: 1598979580.460629, t: -0.266317 -0.117865 0.311166 q: 0.592672 -0.803764 0.026229 0.044895 
[ INFO] [1598979580.542871782]: td 0.003806
time: 1598979580.526987, t: -0.262514 -0.124371 0.331001 q: 0.593085 -0.803362 0.027015 0.046148 
wait for imu ... 
[ INFO] [1598979580.596528498]: td 0.003806
time: 1598979580.593340, t: -0.276444 -0.142661 0.348938 q: 0.592734 -0.803641 0.026864 0.045881 
wait for imu ... 
[ INFO] [1598979580.660778316]: td 0.003806
time: 1598979580.659687, t: -0.266668 -0.145524 0.336145 q: 0.591510 -0.804344 0.027553 0.048867 
[ INFO] [1598979580.757746875]: td 0.003806
time: 1598979580.726030, t: -0.245839 -0.139447 0.318977 q: 0.588785 -0.806118 0.028994 0.051628 
wait for imu ... 
[ INFO] [1598979580.792658118]: td 0.003806
time: 1598979580.792367, t: -0.279200 -0.154376 0.355802 q: 0.587229 -0.807299 0.027369 0.051784 
[ INFO] [1598979580.865781867]: td 0.003806
time: 1598979580.858700, t: -0.278309 -0.152305 0.315351 q: 0.584851 -0.808639 0.028584 0.056878 
wait for imu ... 
[ INFO] [1598979580.930720230]: td 0.003806
time: 1598979580.925027, t: -0.263666 -0.182109 0.335296 q: 0.582047 -0.810655 0.028276 0.057100 
[ INFO] [1598979580.994393616]: td 0.003806
time: 1598979580.991349, t: -0.287057 -0.192899 0.344279 q: 0.577555 -0.813802 0.029914 0.057112 
[ INFO] [1598979581.067586496]: td 0.003806
time: 1598979581.057666, t: -0.278663 -0.198798 0.269724 q: 0.573000 -0.816598 0.034362 0.060479 
[ INFO] [1598979581.131664278]: td 0.003806
time: 1598979581.123976, t: -0.304107 -0.205903 0.302754 q: 0.571997 -0.817257 0.035088 0.060652 
[ INFO] [1598979581.205557235]: td 0.003806
time: 1598979581.190311, t: -0.310208 -0.226132 0.288600 q: 0.570005 -0.818626 0.032926 0.062142 
[ INFO] [1598979581.267280715]: td 0.003806
time: 1598979581.256691, t: -0.315937 -0.244264 0.313172 q: 0.569305 -0.819117 0.031833 0.062662 
[ INFO] [1598979581.335372437]: td 0.003806
time: 1598979581.323065, t: -0.341946 -0.275166 0.315069 q: 0.568127 -0.819898 0.029548 0.064237 
[ INFO] [1598979581.433136067]: td 0.003806
time: 1598979581.389435, t: -0.367537 -0.323557 0.276167 q: 0.565298 -0.821896 0.026987 0.064778 
[ INFO] [1598979581.469361960]: td 0.003806
time: 1598979581.455799, t: -0.395057 -0.360113 0.283889 q: 0.562996 -0.823480 0.023217 0.066160 
[ INFO] [1598979581.542316158]: td 0.003806
time: 1598979581.522157, t: -0.407549 -0.393594 0.276799 q: 0.559738 -0.825584 0.020127 0.068550 
[ INFO] [1598979581.614222613]: td 0.003806
time: 1598979581.588510, t: -0.416887 -0.411349 0.217214 q: 0.553916 -0.829428 0.015964 0.070509 
[ INFO] [1598979581.672849031]: td 0.003806
time: 1598979581.654859, t: -0.440953 -0.436139 0.239089 q: 0.553924 -0.829161 0.012569 0.074177 
[ INFO] [1598979581.738306027]: td 0.003806
time: 1598979581.721202, t: -0.456337 -0.446909 0.204146 q: 0.550008 -0.831872 0.007294 0.073671 
[ INFO] [1598979581.805108457]: td 0.003806
time: 1598979581.787540, t: -0.469295 -0.462219 0.224178 q: 0.547468 -0.833355 0.002796 0.076093 
[ INFO] [1598979581.883032684]: td 0.003806
time: 1598979581.853873, t: -0.484203 -0.482002 0.198478 q: 0.542034 -0.836747 -0.001956 0.077780 
[ INFO] [1598979581.934899618]: td 0.003806

```

With rtabmap, there is no error log about imu not received since last image received and the tracking is good.

getupcurry, 2020-09-02:
That means imu/camera synchronization of MYNTEYE is not well?  But as a commercial camera, I think imu and camera should be well synchronized. Do you think it has something to do with the calibration parameters? What is your idea of  this problem?  Thanks.

matlabbe, 2020-09-03:
I am not sure if it is a problem with the camera or the mynteye ros wrapper. From stereo_odometry point of view, it detects that between two images received, it didn't receive imu msgs. Assuming that imu msgs are published faster than images, I would expect having some imu msgs between images, or a least in the future before receiving the images. Here is my observation:

image received with stamp 10
image received with stamp 13
imu received with stamp 8
imu received with stamp 9
imu received with stamp 10
imu received with stamp 11
imu received with stamp 12
image received with stamp 16
imu received with stamp 13
image received with stamp 19imu received with stamp 12

Expected:
imu received with stamp 8
imu received with stamp 9
imu received with stamp 10
imu received with stamp 11
image received with stamp 10
image received with stamp 20

But I think you could create a simple code showing this problem to mynteye repo.

getupcurry, 2020-09-11:
@matlabbe  Hi, sorry for the delay. 
I have a request : could you test the `test_d435i_vio.launch` with Real-sense D435i？ If it performs well (tracks well and no drift), I will buy one.  Thank you from the bottom of my heart.

getupcurry, 2020-09-21:
@matlabbe ，Hi.  Looking forward to your reply!

matlabbe, 2020-09-27:
Yes, `test_d435i_vio.launch` works ok with my D435i. Make sure to disable the IR emitter or put a tape on the IR emitter. VINS started to drift a lot when seeing only IR points from the IR emitter. 

cheers,
Mathieu

getupcurry, 2020-09-28:
Thanks a lot.

wz8530, 2023-02-17:
> > Hi I would like to ask how to build rtabmap with VINS support. I enabled cmake with VINS flag by sourcing my ros workspace contains VINS, but during building of rtabp map. the complier kept telling me it cannot locate header files of VINS. Should any other paths or flags need to be set?
> 
> https://gist.github.com/matlabbe/795ab37067367dca58bbadd8201d986c this patch might be able to solve your problem.

I download the vins-fusion_be55a93_pull136.patch file and run the following command:
`
$ git apply --ignore-space-change --ignore-whitespace vins-fusion_be55a93_pull136.patch
$ cd ~/vins_fusion_ws/
$ catkin_make      (successfully)
$ source ~/vins_fusion_ws/devel/setup.bash
$ cd ~/rtabmap/build
$ source ~/vins_fusion_ws/devel/setup.bash
$ cmake -DWITH_VINS=ON ..
`
However, when compiling rtabmap, I also found that the vins header file could not be located.

Compilation errors are as follows:
**/project/rtabmap/corelib/src/odometry/OdometryVINS.cpp:39:10: fatal error: estimator/estimator.h: No such file or directory
 #include <estimator/estimator.h>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

@sssolitude @matlabbe How to solve this problem?