introlab/rtabmap_ros #332: Another issue concerns the using of rtabmap independent library buttons under ROS

TomLuosy, 2019-07-31:
When I display the button in the rtabmap stand-alone Library under ros, it doesn't work properly.

Here is the error message：
[ERROR] (2019-07-31 10:55:24.625) 
MainWindow.cpp:4745::openDatabase() Database can only be opened in IDLE state.

matlabbe, 2019-07-31:
That button is hidden on `rtabmapviz` because we cannot switch the database used by `rtabmap` node on runtime. `rtabmapviz` is also meant to be used purely for monitoring (the interface could be running on a remote computer than `rtabmap` node). Actually the MainWindow state under ROS is `Monitoring` (set [here](https://github.com/introlab/rtabmap_ros/blob/ca42a35709782504eeeca9b4a6b2269dd72c9097/src/GuiWrapper.cpp#L103)), and cannot be `Idle`.

TomLuosy, 2019-08-03:
> 该按钮被隐藏，`rtabmapviz`因为我们无法`rtabmap`在运行时切换节点使用的数据库。`rtabmapviz`也意味着纯粹用于监视（接口可以在远程计算机上运行而不是`rtabmap`节点）。实际上ROS下的MainWindow状态是`Monitoring`（[在这里](https://github.com/introlab/rtabmap_ros/blob/ca42a35709782504eeeca9b4a6b2269dd72c9097/src/GuiWrapper.cpp#L103)设置），而不是`Idle`。

Thank you very much for your reply. This gives me a better understanding of ROS.

closed 2019-08-03 by TomLuosy