introlab/rtabmap_ros #334: rtabmap VIO example doesn't work with Azure Kinect

RigerLee, 2019-08-03:
In euroc_datasets.launch, the tf from base_link to imu is fixed:
`<node pkg="tf" type="static_transform_publisher" name="imu_base_link" args="0 0 0 3.1415926 -1.570796 0 base_link imu4 5"/>`

And the tf from imu to left camera:
`<node pkg="tf" type="static_transform_publisher" name="cam0_imu_link" args="-0.021640 -0.064677 0.009811 1.555925 0.025777 0.003757 imu4 cam0 50"/>`
is derived from euroc_stereo_imu_config.yaml:
`[0.0148655429818, -0.999880929698, 0.00414029679422, -0.0216401454975]`
`[0.999557249008, 0.0149672133247, 0.025715529948,  -0.064676986768]`
`[-0.0257744366974, 0.00375618835797, 0.999660727178, 0.00981073058949]`
`[0, 0, 0, 1]`

In my case, I have test modified vins-fusion(take RGBD and IMU as input) with two sensors: kinect2+external IMU and azure kinect.

For kinect2+external IMU, I keep the tf from base_link to imu as it is in euroc_datasets.launch:
`<node pkg="tf" type="static_transform_publisher" name="imu_base_link" args="0 0 0 3.1415926 -1.570796 0 base_link imu 5"/>`
And the tf from imu to rgb/depth camera:
`<node pkg="tf" type="static_transform_publisher" name="cam0_imu_link" args="-0.09503034 -0.00460962 -0.06292056 3.132753 -0.0068463 -0.0345971 imu kinect2_link 50" />`
is derived from this:
`[-0.9999375, -0.00907104, 0.00653615, -0.09503034]`
`[0.00883932, -0.99936044, -0.03464936, -0.00460962]`
`[0.00684628, -0.03458942, 0.99937816, -0.06292056]`
`[0., 0., 0., 1.]`
And it works fine.

For the new azure kinect, I also keep  the tf from base_link to imu as it is in euroc_datasets.launch and calculate the tf from imu to rgb/depth camera:
`<node pkg="tf" type="static_transform_publisher" name="cam0_imu_link" args="0.0470500859 -0.0289010918 -0.000932946137 -1.5758591 -0.0015905 -1.6021107 imu_link rgb_camera_link 50"/>`
which is derived from this:
`[-5.06267863e-03, -3.13168855e-02,  9.99496668e-01, 4.70500859e-02]`
`[-9.99985939e-01, -1.43119039e-03, -5.11002113e-03,-2.89010918e-02]`
`[1.59050901e-03, -9.99508462e-01, -3.13092122e-02,-9.32946137e-04]`
`[0.00000000e+00,  0.00000000e+00,  0.00000000e+00, 1.00000000e+00]`
However, rtabmap rgbd mapping doesn't work this time.
When I test my modified vins-fusion with azure, it works fine. So I'm wondering if the tf from base_link to imu is wrong this time.
Could you please explain why base_link to imu is rotated by a fixed angle and how can I adjust it to make it work on a new sensor?

RigerLee, 2019-08-06:
Now I found out that it is due to the initialization strategy of odometryVINS (Only init estimator after receiving rgbd data. Several imu_msgs will be throw out, which is essential for VINS to init). I solved this by init estimator in the constructor.

But still, I'd like to know  why base_link to imu is rotated by a fixed angle.

matlabbe, 2019-08-09:
Hi,

If imu in [VINS config](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/e9d9911a99a38d68bbc0eb4f9a27277f7d2c6d4f/config/euroc/euroc_stereo_imu_config.yaml#L5) is enabled, rtabmap should wait for imu to initialize: 
https://github.com/introlab/rtabmap/blob/3cafdf911b6cd9183afe0eaf544124583958acd2/corelib/src/odometry/OdometryVINS.cpp#L361

Not sure if there should be a minimum of IMU topics that should be received before initializing the VinsEstimator. With RTAB-Map, the general approach is to use TF and `camera_info` to setup camera parameters in VINS dynamically, instead of using hard-coded config files. If a number of IMU topics would be needed to initialize the VinsEstimator, we could buffer the received imu topics up to X topics and wait to fill that buffer before doing the initialization. I could add a parameter for that buffer size, so we would throw all images until the imu buffer is full.

For the fixed angle between base_link and imu, it is to transform base frame in ROS standard (which is x forward, y left and z up). From the visensor setup: https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets

[image]

cheers,
Mathieu

RigerLee, 2019-08-11:
Thank you for your reply! A buffer for IMU before first RGBD image also works for Azure Kinect.

closed 2019-08-11 by RigerLee