introlab/rtabmap_ros #335: Export Lidar and RGB-D

schmiran, 2019-08-05:
Hello everyone, 

I use a Lidar sensor, a T265 and two D435 for the mapping. The mapping works well so far. Unfortunately I can't get the map out, as I see it in rtabmapviz. I can only export either the Lidar data or the RGB-D data to PLY by using the standalone RTAB and open my *.db file. Is there any way to export the data together?

Here a screenshot of my data:
[image]

And here my launchfile:
```
<launch>

  <!-- EDIT FROM HERE -->
  <!-- Set which sensors u want to use (for now just 2 D series Sensors are supported) -->
  <arg name="use_t265"                default="true"/>

  <arg name="rgbd_cam_count"          default="2"/>
  <arg name="use_d435_1"              default="true"/>
  <arg name="use_d435_2"              default="true"/>
  <arg name="use_d415"                default="false"/>

  <arg name="use_vlp16"               default="true"/>

<!-- DO NOT EDIT BELOW HERE -->

<!-- Include sensor nodes -->
  <include if="$(arg use_d435_1)" file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="camera"        value="d435_1" />
    <arg name="serial_no"     value="*************" />
    <arg name="align_depth"   value="true" />
    <arg name="enable_infra1" value="false" />
    <arg name="enable_infra2" value="false" />
  </include>

  <include if="$(arg use_d435_2)" file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="camera"        value="d435_2" />
    <arg name="serial_no"     value="*************" />
    <arg name="align_depth"   value="true" />
    <arg name="enable_infra1" value="false" />
    <arg name="enable_infra2" value="false" />
  </include>

  <include if="$(arg use_d415)" file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="camera"        value="d415" />
    <arg name="serial_no"     value="*************" />
    <arg name="align_depth"   value="true" />
    <arg name="enable_infra1" value="false" />
    <arg name="enable_infra2" value="false" />
  </include>

  <include if="$(arg use_t265)" file="$(find realsense2_camera)/launch/rs_t265.launch">
    <arg name="camera"            value="t265" />
    <arg name="serial_no"         value="*************" />
    <arg name="unite_imu_method"  value="linear_interpolation" />
    <arg name="enable_fisheye1"     value="false"/>
    <arg name="enable_fisheye2"     value="false"/>
  </include>

  <include if="$(arg use_vlp16)" file="$(find velodyne_pointcloud)/launch/VLP16_points.launch"/>

  <!-- TF: x y z z° y° x° -->
  <node if="$(arg use_d435_1)" pkg="tf" type="static_transform_publisher" name="base_to_d435_1_tf"
      args="0.025 -0.09 0.113 -0.52 0.0 0.0 /base_link /d435_1_link 200" />
  <node if="$(arg use_d435_2)" pkg="tf" type="static_transform_publisher" name="base_to_d435_2_tf"
      args="0 0.125 0.113 0.52 0.0 0.0 /base_link /d435_2_link 200" />
  <node if="$(arg use_d415)" pkg="tf" type="static_transform_publisher" name="base_to_d415_tf"
      args="0.0 0.02 0.183 0.0 0.0 0.0 /base_link /d415_link 200" />
  <node if="$(arg use_t265)" pkg="tf" type="static_transform_publisher" name="base_to_t265_tf"
      args="0 0.0 0.21 0.0 0.0 0.0 /base_link /t265_link 200" />
  <node if="$(arg use_t265)" pkg="tf" type="static_transform_publisher" name="pose_to_base_tf"
      args="0 0.0 0.21 0.0 0.0 0.0 /t265_pose_frame /base_link 200" />
[... 85 lines cut ...]
      <remap if="$(arg use_t265)" from="odom" to="/t265/odom/sample" />
      <remap if="$(arg use_vlp16)" from="scan_cloud" to="/velodyne_points"/>

      <param name="subscribe_rgb"           type="bool"   value="false"/>
      <param name="subscribe_depth"     type="bool"   value="false"/>
      <param name="subscribe_rgbd"      type="bool"   value="true"/>

      <!-- if using t265 for odometry -->
      <param if="$(arg use_t265)" name="visual_odometry" value="false"/>
      <param if="$(arg use_t265)" name="odom_topic" value="/t265/odom/sample"/>

      <param if="$(arg use_vlp16)" name="subscribe_scan_cloud" type="bool" value="true"/>
      <param if="$(arg use_vlp16)" name="scan_cloud_topic"        value="/velodyne_points" />

      <param name="frame_id"            type="string" value="base_link"/>

      <param name="rgbd_cameras"       type="int"    value="$(arg rgbd_cam_count)"/>

    </node>
    </group>
</launch>

```

Many thanks in advance

matlabbe, 2019-08-09:
Hi,

There is no way to export online both data (RGBD cloud and Lidar cloud) on the same output. However with the standalone or rtabmapviz, if you export the PLY for cameras, then the PLY for lidar, they should be in the same coordinate frame. With a software like CloudCompare or MeshLab, you could open both and re-export the final merged cloud.

cheers,
Mathieu

schmiran, 2019-08-14:
Hi, 

thank you. That's fine, of course. 

Greetings

closed 2019-08-14 by schmiran