introlab/rtabmap_ros #336: How to modify the default save name of 3d-map?

o0ozz, 2019-08-06:
Hello, I'd like to ask an issue. 

After running rtabmap_ros, the 3D-map named rtabmao.db will be saved in the hidden folder (/.ros). Now, I can modify the save path in the rtabmap.launch, but the name is the only one. If I save it many times without changing the name, the previous map will be overwritten. So I want to know if I can automatically save different names, such as time stamps, as in the rtabmap .

This is my modification of the path.
`<arg name="database_path"           default="~/Documents/rtabmap.db"/>`

matlabbe, 2019-08-09:
Hi,

The name of the database is always the same (default `rtabmap.db`). To record on different names, we would have to manually launch `rtabmap.launch` with a different database paths. Or for convenience, set the stamp directly when launching `rtabmap.launch`:
```bash
$ roslaunch rtabmap_ros rtabmap.launch \
   database_path:=~/Documents/rtabmap_`date +'%s'`.db \
   ...
```

You should see log:
```bash
...
[ INFO] [1565383096.133123562]: rtabmap: Using database 
from "/home/mathieu/Documents/rtabmap_1565383093.db" (0 MB).
...
```

For convenience, I think you could also add the date in the launch file like in this post: https://answers.ros.org/question/261616/get-timestamp-during-roslaunch/

cheers,
Mathieu

o0ozz, 2019-08-14:
Hi, Mathieu.
I have solved this problem with your help. Thank you very much.

closed 2019-08-14 by matlabbe