introlab/rtabmap_ros #337: find_object implementation for localization

jonpol01, 2019-08-06:
Hi,

I am working with find_object_2d used for mapping.. tried the saveobjects example and i just want to confirm/ask if setting the `localization:=true` after mapping, will the objects detected helps with localization??

If Yes, may i ask for where to look first to confirm it, tried database-viewer and it seems its not on Landmark category.

And if No, is there a workaround for a simple implementation for it.?

matlabbe, 2019-08-10:
Hi,

For the first question no. However, it is a good suggestion. I updated [find_object_2d](https://github.com/introlab/find-object) to get more accurate position of the objects in the map (there was a time sync problem). I also updated `rtabmap_ros` with a new script [objects_to_tags.py](https://github.com/introlab/rtabmap_ros/blob/master/scripts/objects_to_tags.py) to convert output of [find_object_2d](https://github.com/introlab/find-object) into [apriltag_ros](http://wiki.ros.org/apriltag_ros)'s AprilTagDetectionArray topic (format used by rtabmap_ros for landmarks). I then updated demo_find_object.launch
https://github.com/introlab/rtabmap_ros/blob/f5a51aa87f0a872aed1abfb4ecd6c6c609b02caa/launch/demo/demo_find_object.launch#L120-L124
to add the option to use objects as landmarks. Here is an example:
```bash
$ roslaunch rtabmap_ros demo_find_object.launch save_objects_as_landmarks:=true rtabmapviz:=true
$ rosbag play --clock demo_find_object.bag
```
You should see green links representing the landmarks when detected in Graph view of rtabmapviz. Landmarks are used like loop closures, so they add constraints to graph. In localization mode, landmarks are used for localization. You may however adjust the parameters of find_object_2d to accept transform only with a higher number of inliers, otherwise the orientation estimation may be not very accurate, causing large jumps on localization.

So to test this example, you have to build latest [find_object_2d](https://github.com/introlab/find-object), [rtabmap](https://github.com/introlab/rtabmap) and [rtabmap_ros](https://github.com/introlab/rtabmap_ros) from source (make sure to install [apriltag_ros](https://github.com/AprilRobotics/apriltag_ros) package too to have AprilTagDetectionArray msg).

cheers,
Mathieu

jonpol01, 2019-09-02:
Thank you very much for this solution..

However, after i updated the find_object package, i cannot use or the "Add object from scene" has a blank window.
[image]

matlabbe, 2019-09-03:
Thx for reporting, I fixed the bug in the commit above. Feel free to reopen if the original question is still problematic.

closed 2019-09-03 by matlabbe

takahiko-hasegawa-hdjp, 2024-03-07:
Im tryinsg to use find_objects, object_to_tags.py and rtabmap apriltags msg interface for landmark based localization but i encountered some issues.

in object_to_tags.py provided by rtabmap, there is this following code:
 (trans,quat) = listener.lookupTransform(data.header.frame_id, objFramePrefix_+'_'+str(int(objId)), data.header.stamp) in the callback function.

but there is a crash. the findobject node publishes objectsStamped topics but it doesnt have the frameID which causes a crash. how can i resolve this issue? thanks!

matlabbe, 2024-03-09:
That [ObjectsStamped](https://github.com/introlab/find-object/blob/master/msg/ObjectsStamped.msg) msg should have a header with `frame_id`, if not than the topic received is not a `ObjectsStamped`.