introlab/rtabmap_ros #338: Where is the data entry?

o0ozz, 2019-08-08:
Hi,

The sensor I use is D435i which supports a minimum FPS of 6. During the experiment, the sensor moves very slowly and I do not want so many photos. At the same time, I want to use the attitude angle of IMU output, and only use the fixed angle photos to model. So I wonder if it can be done by modifying the code at the data entry. Where is the data entry if possible? What should I do if I can't?

 Thanks!

matlabbe, 2019-08-13:
Hi, not sure to understand "only use the fixed angle photos to model". We can input IMU to rgbd_odometry node, which will work better if RTAB-Map is built with VIO approaches though. However, from the minimal RTAB-Map config, the default odometry approach (F2M) can support IMU loosely. I updated the [hand-held tutorial on ros](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping) for the D435i:
```bash
$ roslaunch realsense2_camera rs_camera.launch \
   align_depth:=true \
   unite_imu_method:="linear_interpolation"
$ rosrun imu_filter_madgwick imu_filter_node \
   _use_mag:=false \
   _publish_tf:=false \
   _world_frame:="enu" \
   /imu/data_raw:=/camera/imu \
   /imu/data:=/rtabmap/imu
$ roslaunch rtabmap_ros rtabmap.launch \
   rtabmap_args:="--delete_db_on_start --Optimizer/GravitySigma 0.3" \
   depth_topic:=/camera/aligned_depth_to_color/image_raw \
   rgb_topic:=/camera/color/image_raw \
   camera_info_topic:=/camera/color/camera_info \
   approx_sync:=false \
   wait_imu_to_init:=true \
   imu_topic:=/rtabmap/imu
```

o0ozz, 2019-08-14:
Hi, 

I have implemented the use of IMU, but I still have a few questions to ask.

The first is about the data entry mentioned earlier. My sensor is rotating sloely. The image from sensor corresponds to an elevation angle. I want the overlap angle between every two adjacent image used in 3D-mapping to be between 10 and 20 degrees, instead of using sensors to transmit all images. so I want to find the data entry. 

Secondly, I want to display the attitude angle on the terminal or rtabmap_ros gui. I have added a formula for calculating Euler angles from quaternions, and it is displayed in the form of warning on the terminal. Is there a statement in rtabmap_ros that calculates and displays the angle?
The procedure I added to CoreWrapper.cpp is as follows:
```
float q0,q1,q2,q3;
q0=msg->orientation.x;
q1=msg->orientation.y;
q2=msg->orientation.z;
q3=msg->orientation.w;

float roll,pitch,yaw;
pitch = atan2(2 * (q1*q2 + q3*q0), q3*q3 - q0*q0 - q1*q1 + q2*q2);
roll = asin(-2 * (q0*q2 - q3*q1));
yaw = atan2(2 * (q0*q1 + q3*q2), q3*q3 + q0*q0 - q1*q1 - q2*q2);
ROS_WARN("the pitch=%f   roll=%f   yaw=%f",pitch,roll,yaw);
```
The terminal is shown below:
```
[ WARN] [1565771824.360967864]: the pitch=-84.623055   roll=0.169400   yaw=-0.190396
[ WARN] [1565771824.363328002]: the pitch=-84.620918   roll=0.168446   yaw=-0.190163
[ WARN] [1565771824.363527769]: the pitch=-84.635788   roll=0.168364   yaw=-0.190440
[ WARN] [1565771824.366081104]: the pitch=-84.621643   roll=0.168797   yaw=-0.190449
[ WARN] [1565771824.367816747]: the pitch=-84.632751   roll=0.169965   yaw=-0.190471
[ WARN] [1565771824.369002892]: the pitch=-84.624825   roll=0.171458   yaw=-0.190485
[ WARN] [1565771824.370950857]: the pitch=-84.622238   roll=0.172839   yaw=-0.190502
```

matlabbe, 2019-08-17:
Depending on what you are looking for, you may echo the [TF](http://wiki.ros.org/tf) directly:
```
$ rosrun tf tf_echo odom camera_link
```

Do you have a video or a picture of the setup? If you only want to publish an image each 10/20 degrees of rotation, I think you could indeed create a middle node to subscribe to IMU topic and camera topic, evaluate the quaternion as you did and republish the image on another name only if the sensor rotated enough.

TomLuosy, 2019-12-15:
Hi,I encountered similar problems in the development process. How can I modify the publisher's code so that it can subscribe to IMU data and transfer image data under the set conditions?

TomLuosy, 2019-12-16:
I found the variable name of IMU data（sensor_msgs::Imu）, but I don't know where to add logical judgment to control whether to subscribe to camera information.

matlabbe, 2019-12-19:
You can use [message_filters ](http://wiki.ros.org/message_filters) to subscribe to make a callback with more than one topic.