introlab/rtabmap_ros #341: Errors with RViz and RTABmapviz

D-aggie, 2019-08-09:
Hello, I am currently experiencing an issue with launching either visual software applications. Playing around with different settings I have found after running catkin_make_isolated the sourcing the terminal changes the outcome. "source ./devel/setup.bash" I receive this error:

*ERROR: cannot launch node of type [rtabmap_ros/rgbd_odometry]: can't locate node [rgbd_odometry] in package [rtabmap_ros]
ERROR: cannot launch node of type [rtabmap_ros/rtabmap]: can't locate node [rtabmap] in package [rtabmap_ros]
ERROR: cannot launch node of type [rtabmap_ros/rtabmapviz]: can't locate node [rtabmapviz] in package [rtabmap_ros]
process[rviz-4]: started with pid [18188]
process[points_xyzrgb-5]: started with pid [18189]
type is rtabmap_ros/point_cloud_xyzrgb
[ERROR] [1565364381.943131056]: Failed to load nodelet [/points_xyzrgb] of type [rtabmap_ros/point_cloud_xyzrgb] even after refreshing the cache: Could not find library corresponding to plugin rtabmap_ros/point_cloud_xyzrgb. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1565364381.943198983]: The error before refreshing the cache was: Could not find library corresponding to plugin rtabmap_ros/point_cloud_xyzrgb. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[points_xyzrgb-5] process has died [pid 18189, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet standalone rtabmap_ros/point_cloud_xyzrgb rgb/image:=/camera/color/image_raw depth/image:=/camera/aligned_depth_to_color/image_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image_relay cloud:=voxel_cloud __name:=points_xyzrgb __log:=/home/david/.ros/log/19f3fa22-bab7-11e9-9745-3c9509e8b4fd/points_xyzrgb-5.log].
log file: /home/david/.ros/log/19f3fa22-bab7-11e9-9745-3c9509e8b4fd/points_xyzrgb-5*.log
[ERROR] [1565364382.721294738]: PluginlibFactory: The plugin for class 'rtabmap_ros/MapCloud' failed to load.  Error: Could not find library corresponding to plugin rtabmap_ros/MapCloud. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1565364382.721554903]: PluginlibFactory: The plugin for class 'rtabmap_ros/Info' failed to load.  Error: Could not find library corresponding to plugin rtabmap_ros/Info. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1565364382.721662729]: PluginlibFactory: The plugin for class 'rtabmap_ros/MapGraph' failed to load.  Error: Could not find library corresponding to plugin rtabmap_ros/MapGraph. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.*

But if I run "source /opt/ros/kinetic/setup.bash" only RTABmapviz opens, no RViz.
Any ideas on how to solve this problem are greatly appreciated

matlabbe, 2019-08-16:
Hi,

Is rtabmap built from source? I generally never use `catkin_make_isolated`. For [rtabmap](https://github.com/introlab/rtabmap) package, it should be built outside the `catkin_ws`, like in the instructions [here (step 2)](https://github.com/introlab/rtabmap_ros#build-from-source).

Note that when you do only "`source /opt/ros/kinetic/setup.bash`", it overrides the paths set with "`source ~/catkin_ws/devel/setup.bash`", so if rtabmap packages are built in `catkin_ws`, it won't be found. Make sure to call "`source ~/catkin_ws/devel/setup.bash`" after "`source /opt/ros/kinetic/setup.bash`".

cheers,
Mathieu

farhangnaderi, 2021-09-24:
I have the same issue and have followed as @matlabbe has mentioned. But no map coming.

matlabbe, 2021-09-28:
@farhangnaderi "But no map coming", do you mean you have the same errors? or that no map is shown in rviz/rtabmapviz? This issues is about errors when launching nodes, where ROS cannot find them.

farhangnaderi, 2021-09-28:
@matlabbe I got it wrong then. my bad. My problem is going on [here](https://github.com/introlab/rtabmap/issues/770)