introlab/rtabmap_ros #342: rtabmap slows down camera framerate, does not update quickly

pringithub, 2019-08-14:
Hello,

I'm trying to stream video from my ODROID-XU4/RealSense D435 setup to a remote Ubuntu machine running rtabmap. Attempting to stream the D435 at the default 30fps, I get an actual fps of 4-9 Hz on both robot and remote computers (>2 Hz is OK). When I run rtabmap, I notice the framerate (through `rostopic hz []`) drop to 2-5 Hz. The frames are correctly synced on the output of the robot, however the two computers are not synced (can't ntpdate the ODROID due to proxy issues) - this might be a problem? The update time shown in the rtabmap [INFO] logs is 40-60ms. The problem is that, even though the frames are coming in at a decent framerate, rtabmap takes much longer to process them and update the map/pose.

eliabntt, 2020-01-28:
any news about this?

matlabbe, 2020-02-05:
If you have more details or update, feel free to comment. I cannot tell if it is an hardware problem or not, but not having both computer clocks synchronized will generate false lags on the client computer when TF is involved.